turbokongen/hass-AMS #1: Memory leak

Hellowlol, 2019-12-31:
The memory seems to increase over time. I suspect a memory leak

turbokongen, 2020-01-02:
I have not notice this. I have a logging of the memory used on my server running HA. Those did not change, but hey, other systems might do other things. Happy to include your findings.

turbokongen, 2020-02-29:
@Hellowlol Could you please test https://github.com/turbokongen/hass-AMS/tree/memory_leak
to see if that takes care of the memory leak?

turbokongen, 2020-03-15:
So, I've been debugging this, and here are my findings:
Tried casting data variable in read_bytes()
Tried emptying variables after use
Tried flushing serial buffer
Tried using .inWaiting() to read buffer, but this leads to high CPU usage again.

Whatever of the measures I've applied, seperately and alltogether don't work.

I've found where it comes from:
```
Filename: /home/pi/.homeassistant/custom_components/ams/__init__.py

Line #    Mem usage    Increment   Line Contents
================================================
    88  62.832031 MiB  62.832031 MiB       @profile(precision=6)
    89                                 def read_bytes(self):
    90                                     """Read the raw data from serial port."""
    91  62.832031 MiB   0.000000 MiB           byte_counter = 0
    92  62.832031 MiB   0.000000 MiB           bytelist = []
    93  62.832031 MiB   0.000000 MiB           buffer = bytes()
    94  62.843750 MiB   0.000000 MiB           while self._running:
    95  62.843750 MiB   0.011719 MiB               buffer = self._ser.read()
    96  62.843750 MiB   0.000000 MiB               if buffer:
    97  62.812500 MiB   0.000000 MiB                   bytelist.extend(buffer)
    98  62.812500 MiB   0.000000 MiB                   if buffer == FRAME_FLAG and byte_counter > 1:
    99  62.812500 MiB   0.000000 MiB                       self._ser.flushInput()
   100  62.812500 MiB   0.000000 MiB                       buffer = bytes()
   101  62.812500 MiB   0.000000 MiB                       _LOGGER.debug('buffer: %s', self._ser.inWaiting())
   102  62.812500 MiB   0.000000 MiB                       return bytelist
   103  62.812500 MiB   0.000000 MiB                   buffer = bytes()
   104  62.812500 MiB   0.000000 MiB                   byte_counter = byte_counter + 1
   105                                         else:
   106  62.843750 MiB   0.000000 MiB                   buffer = bytes()
   107  62.843750 MiB   0.000000 MiB                   continue
```
By the look of it, the issue seems to come from the pyserial lib.

Hellowlol, 2020-03-16:
I have tried to debug the same, previously. I never figure out why the memory usage increased.

closed 2020-03-22 by turbokongen