turbokongen/hass-AMS #2: Cpu usage

Hellowlol, 2019-12-31:
The cpu usage is very high (I have a solution for this pr is coming)

turbokongen, 2020-01-02:
Yes, I did notice that. Initially I wanted to use the async serial, but I could not get it to work. Looking forward for your PR.

thomasja27, 2020-02-24:
The same thing happens to me, the load on the CPU increases to 30% when the plugin is active. usually I am at 1-2%

turbokongen, 2020-02-29:
could you please try the https://github.com/turbokongen/hass-AMS/tree/cpu_ branch to see if that changes anything?

Hellowlol, 2020-02-29:
I noticed the same issue on my rpi3 and nuc i5

turbokongen, 2020-02-29:
Does the cpu_ branch change the behaviour? I got 5-10% less Cpu usage on my server, and 1-4% less on the rpi3+

Hellowlol, 2020-02-29:
Dunno, im using the stuff branch. Ill try the cpu_ later.

thomasja27, 2020-02-29:
With the cpu_ branch load is around 2-3% so it did make a huge difference :) The load was around 30% before.

My system:
Lenovo M92P
Intel Core I3-2120T @ 2.6 GHz
6 GB Ram
256GB SSD

turbokongen, 2020-02-29:
Perfect! thanks for the feedback.

PR #10 (2020-02-29): Fix cpu usage
Fixes #2 Cpu usage high.

Move constants to file.
closed 2020-02-29 by turbokongen