turbokongen/hass-AMS #8: Kaifa meter

thomasja27, 2020-02-03:
I have opened the HAN-port on my meter (KAIFA) but do not get any data into Home Assistant. Would think it needs to be rewritten in order to make this work since there is a difference in the data from Kamstrup and Kaifa. 

I've logged some data from the meter. This is list 1 (comes every 2 sec) and list 2 (comes every 10 sec). list 3 comes every hour, so I'll try to get it logged as well.

`
List 2
7e a0 79 01 02 01 10 80 93 e6 e7 00 0f 40 00 00
00 09 0c 07 e4 02 03 01 0f 1b 0a ff 80 00 00 02
0d 09 07 4b 46 4d 5f 30 30 31 09 10 36 39 37 30
36 33 31 34 30 35 38 30 38 34 36 39 09 08 4d 41
33 30 34 48 33 45 06 00 00 00 37 06 00 00 00 00
06 00 00 00 00 06 00 00 01 17 06 00 00 01 f3 06
00 00 03 61 06 00 00 03 84 06 00 00 09 29 06 00
00 00 00 06 00 00 09 23 d1 f7 7e 

List1
7e a0 27 01 02
01 10 5a 87 e6 e7 00 0f 40 00 00 00 09 0c 07 e4
02 03 01 0f 1b 0c ff 80 00 00 02 01 06 00 00 00
37 a2 2e 7e 

List 1
7e a0 27 01 02 01 10 5a 87 e6 e7 00
0f 40 00 00 00 09 0c 07 e4 02 03 01 0f 1b 0e ff
80 00 00 02 01 06 00 00 00 37 19 2c 7e 

List 1
7e a0 27
01 02 01 10 5a 87 e6 e7 00 0f 40 00 00 00 09 0c
07 e4 02 03 01 0f 1b 10 ff 80 00 00 02 01 06 00
00 00 38 17 cc 7e 

List 1
7e a0 27 01 02 01 10 5a 87 e6
e7 00 0f 40 00 00 00 09 0c 07 e4 02 03 01 0f 1b
12 ff 80 00 00 02 01 06 00 00 00 37 5b 36 7e 

List 2
7e
a0 79 01 02 01 10 80 93 e6 e7 00 0f 40 00 00 00
09 0c 07 e4 02 03 01 0f 1b 14 ff 80 00 00 02 0d
09 07 4b 46 4d 5f 30 30 31 09 10 36 39 37 30 36
33 31 34 30 35 38 30 38 34 36 39 09 08 4d 41 33
30 34 48 33 45 06 00 00 00 36 06 00 00 00 00 06
00 00 00 00 06 00 00 01 18 06 00 00 01 ef 06 00
00 03 62 06 00 00 03 85 06 00 00 09 30 06 00 00
00 00 06 00 00 09 25 cf a2 7e
`

thomasja27, 2020-02-03:
Found this at https://github.com/roarfred/AmsToMqttBridge/blob/master/Code/Arduino/AmsToMqttBridge/AmsToMqttBridge.ino

Is this possible to use?

void readHanPort_Kaifa(int listSize) 
{
	// Only care for the ACtive Power Imported, which is found in the first list
	if (listSize == (int)Kaifa::List1 || listSize == (int)Kaifa::List2 || listSize == (int)Kaifa::List3)
	{
		if (listSize == (int)Kaifa::List1)
		{
			if (debugger) debugger->println(" (list #1 has no ID)");
		}
		else
		{
			String id = hanReader.getString((int)Kaifa_List2::ListVersionIdentifier);
			if (debugger) debugger->println(id);
		}

		// Get the timestamp (as unix time) from the package
		time_t time = hanReader.getPackageTime();
		if (debugger) debugger->print("Time of the package is: ");
		if (debugger) debugger->println(time);

		// Define a json object to keep the data
		//StaticJsonBuffer<500> jsonBuffer;
		DynamicJsonBuffer jsonBuffer;
		JsonObject& root = jsonBuffer.createObject();

		// Any generic useful info here
		root["id"] = WiFi.macAddress();
		root["up"] = millis();
		root["t"] = time;

		// Add a sub-structure to the json object, 
		// to keep the data from the meter itself
		JsonObject& data = root.createNestedObject("data");

		// Get the temperature too
		tempSensor.requestTemperatures();
		float temperature = tempSensor.getTempCByIndex(0);
		data["temp"] = String(temperature);

		// Based on the list number, get all details 
		// according to OBIS specifications for the meter
		if (listSize == (int)Kaifa::List1)
		{
			data["P"] = hanReader.getInt((int)Kaifa_List1::ActivePowerImported);
		}
		else if (listSize == (int)Kaifa::List2)
		{
			data["lv"] = hanReader.getString((int)Kaifa_List2::ListVersionIdentifier);
			data["id"] = hanReader.getString((int)Kaifa_List2::MeterID);
			data["type"] = hanReader.getString((int)Kaifa_List2::MeterType);
			data["P"] = hanReader.getInt((int)Kaifa_List2::ActiveImportPower);
			data["Q"] = hanReader.getInt((int)Kaifa_List2::ReactiveImportPower);
			data["I1"] = hanReader.getInt((int)Kaifa_List2::CurrentL1);
			data["I2"] = hanReader.getInt((int)Kaifa_List2::CurrentL2);
			data["I3"] = hanReader.getInt((int)Kaifa_List2::CurrentL3);
			data["U1"] = hanReader.getInt((int)Kaifa_List2::VoltageL1);
			data["U2"] = hanReader.getInt((int)Kaifa_List2::VoltageL2);
			data["U3"] = hanReader.getInt((int)Kaifa_List2::VoltageL3);
		}
		else if (listSize == (int)Kaifa::List3)
		{
			data["lv"] = hanReader.getString((int)Kaifa_List3::ListVersionIdentifier);;
			data["id"] = hanReader.getString((int)Kaifa_List3::MeterID);
			data["type"] = hanReader.getString((int)Kaifa_List3::MeterType);
			data["P"] = hanReader.getInt((int)Kaifa_List3::ActiveImportPower);
			data["Q"] = hanReader.getInt((int)Kaifa_List3::ReactiveImportPower);
			data["I1"] = hanReader.getInt((int)Kaifa_List3::CurrentL1);
			data["I2"] = hanReader.getInt((int)Kaifa_List3::CurrentL2);
			data["I3"] = hanReader.getInt((int)Kaifa_List3::CurrentL3);
			data["U1"] = hanReader.getInt((int)Kaifa_List3::VoltageL1);
			data["U2"] = hanReader.getInt((int)Kaifa_List3::VoltageL2);
			data["U3"] = hanReader.getInt((int)Kaifa_List3::VoltageL3);
			data["tPI"] = hanReader.getInt((int)Kaifa_List3::CumulativeActiveImportEnergy);
			data["tPO"] = hanReader.getInt((int)Kaifa_List3::CumulativeActiveExportEnergy);
			data["tQI"] = hanReader.getInt((int)Kaifa_List3::CumulativeReactiveImportEnergy);
			data["tQO"] = hanReader.getInt((int)Kaifa_List3::CumulativeReactiveExportEnergy);
		}

		// Write the json to the debug port
		if (debugger) {
			debugger->print("Sending data to MQTT: ");
			root.printTo(*debugger);
			debugger->println();
		}

		// Make sure we have configured a publish topic
		if (ap.config.mqttPublishTopic == 0 || strlen(ap.config.mqttPublishTopic) == 0)
			return;

		// Publish the json to the MQTT server
		char msg[1024];
		root.printTo(msg, 1024);
		mqtt.publish(ap.config.mqttPublishTopic, msg);
	}
}

thomasja27, 2020-02-03:
Here is List 3 from my meter: (comes every hour)

`7e a0 9b 01 02 01 10 ee ae e6 e7 00 0f 40 00 00
00 09 0c 07 e4 02 03 01 10 00 0a ff 80 00 00 02
12 09 07 4b 46 4d 5f 30 30 31 09 10 36 39 37 30
36 33 31 34 30 35 38 30 38 34 36 39 09 08 4d 41
33 30 34 48 33 45 06 00 00 00 77 06 00 00 00 00
06 00 00 00 00 06 00 00 01 21 06 00 00 02 a6 06
00 00 03 28 06 00 00 03 fb 06 00 00 08 ef 06 00
00 00 00 06 00 00 08 fa 09 0c 07 e4 02 03 01 10
00 0a ff 80 00 00 06 00 a0 24 0f 06 00 00 00 00
06 00 01 35 81 06 00 1f 6e af 2b 02 7e
`

turbokongen, 2020-02-16:
I will try to implement this, I'll make a new branch you can test.

turbokongen, 2020-02-16:
Ok, I've coded a parser based on what you have sent, and compared it to info I found.
Could you please try the kaifa branch?
You must remove your original install of hass-AMS and install this.
Type kaifa in the meter manufacturer field.
Enable debug logging for the module.
Please post any errors if it doesn't work :)
https://github.com/turbokongen/hass-AMS/tree/kaifa

thomasja27, 2020-02-17:
Thank you so much for looking in to this!
Looks like there is something wrong.
Here is my log: 
2020-02-17 07:36:32 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 07:36:32 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2020-02-17 07:36:34 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:34 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 34, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 111, 167, 79, 126]
2020-02-17 07:36:35 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 07:36:35 DEBUG (MainThread) [custom_components.ams.sensor] HUB in async_stup_entry= {}
2020-02-17 07:36:35 DEBUG (MainThread) [custom_components.ams.sensor] AMS_SENSORS in async_setup_entry= {}
2020-02-17 07:36:35 DEBUG (MainThread) [custom_components.ams.sensor] async_add_devices in end of async_setup_entry
2020-02-17 07:36:36 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:36 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 36, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 111, 106, 72, 126]
2020-02-17 07:36:38 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:38 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 38, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 108, 74, 120, 126]
2020-02-17 07:36:40 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:40 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 121, 1, 2, 1, 16, 128, 147, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 40, 255, 128, 0, 0, 2, 13, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 53, 56, 48, 56, 52, 54, 57, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 1, 107, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 66, 6, 0, 0, 7, 65, 6, 0, 0, 3, 144, 6, 0, 0, 3, 229, 6, 0, 0, 8, 233, 6, 0, 0, 0, 0, 6, 0, 0, 9, 0, 94, 127, 126]
2020-02-17 07:36:42 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:42 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 42, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 110, 194, 84, 126]
2020-02-17 07:36:44 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:44 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 44, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 110, 15, 83, 126]
2020-02-17 07:36:46 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:46 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 46, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 110, 180, 81, 126]
2020-02-17 07:36:48 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:48 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 48, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 110, 77, 73, 126]
2020-02-17 07:36:50 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:50 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 121, 1, 2, 1, 16, 128, 147, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 50, 255, 128, 0, 0, 2, 13, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 53, 56, 48, 56, 52, 54, 57, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 1, 109, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 69, 6, 0, 0, 7, 73, 6, 0, 0, 3, 149, 6, 0, 0, 3, 236, 6, 0, 0, 8, 236, 6, 0, 0, 0, 0, 6, 0, 0, 9, 22, 152, 247, 126]
2020-02-17 07:36:52 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:52 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 52, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 108, 41, 111, 126]
2020-02-17 07:36:54 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:54 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 54, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 108, 146, 109, 126]
2020-02-17 07:36:56 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
2020-02-17 07:36:56 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 36, 56, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 108, 179, 96, 126]
2020-02-17 07:36:58 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check
[... cut ...]
2020-02-17 07:38:10 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 121, 1, 2, 1, 16, 128, 147, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 7, 38, 10, 255, 128, 0, 0, 2, 13, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 53, 56, 48, 56, 52, 54, 57, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 1, 105, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 70, 6, 0, 0, 7, 52, 6, 0, 0, 3, 124, 6, 0, 0, 3, 237, 6, 0, 0, 8, 236, 6, 0, 0, 0, 0, 6, 0, 0, 9, 23, 78, 71, 126]
2020-02-17 07:38:12 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid header CRC check

thomasja27, 2020-02-17:
No sensordata yet.

Error message:
Loggdetaljer (ERROR)
Mon Feb 17 2020 08:45:27 GMT+0100 (sentraleuropeisk normaltid)
Error while setting up ams platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 159, in _async_setup_platform
    await asyncio.wait_for(asyncio.shield(task), SLOW_SETUP_MAX_WAIT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/config/custom_components/ams/sensor.py", line 45, in async_setup_entry
    'state': data[sensor_name].get('state'),
AttributeError: 'int' object has no attribute 'get'

Log:
2020-02-17 08:45:24 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 08:45:24 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2020-02-17 08:45:24 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid packet size 17
2020-02-17 08:45:24 DEBUG (Thread-2) [custom_components.ams] failed package: [135, 45, 24, 255, 128, 0, 0, 2, 1, 6, 0, 0, 0, 129, 17, 217, 126]
2020-02-17 08:45:26 DEBUG (Thread-2) [custom_components.ams] [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 8, 45, 26, 255, 128, 0, 0, 2, 1, 6, 0, 0, 0, 130, 49, 233, 126]
2020-02-17 08:45:26 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-17 08:45:26 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= []
2020-02-17 08:45:27 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 08:45:27 DEBUG (MainThread) [custom_components.ams.sensor] HUB in async_stup_entry= {'packet_size': 41, 'day_of_week': 'Monday', 'date_time': '2020-228-17 08:45:26', 'list_type': 1}
2020-02-17 08:45:27 DEBUG (MainThread) [custom_components.ams.sensor] AMS_SENSORS in async_setup_entry= {'packet_size': 41, 'day_of_week': 'Monday', 'date_time': '2020-228-17 08:45:26', 'list_type': 1}

turbokongen, 2020-02-17:
Ok, found another issue, please try again

thomasja27, 2020-02-17:
Getting data in the log now, but there is still an error when setting up the sensor:

Loggdetaljer (ERROR)
Mon Feb 17 2020 10:27:02 GMT+0100 (sentraleuropeisk normaltid)
Error while setting up ams platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 159, in _async_setup_platform
    await asyncio.wait_for(asyncio.shield(task), SLOW_SETUP_MAX_WAIT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/config/custom_components/ams/sensor.py", line 45, in async_setup_entry
    'state': data[sensor_name].get('state'),
AttributeError: 'int' object has no attribute 'get'

Log:
2020-02-17 10:09:25 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 10:09:25 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2020-02-17 10:09:26 DEBUG (Thread-2) [custom_components.ams] [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 10, 9, 26, 255, 128, 0, 0, 2, 1, 6, 0, 0, 0, 96, 237, 115, 126]
2020-02-17 10:09:26 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-17 10:09:26 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 96, 237]
2020-02-17 10:09:27 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 10:09:27 DEBUG (MainThread) [custom_components.ams.sensor] HUB in async_stup_entry= {'packet_size': 41, 'day_of_week': 'Monday', 'date_time': '2020-2-17 10:09:26', 'list_type': 1, 'active_power_n': 24813}
2020-02-17 10:09:27 DEBUG (MainThread) [custom_components.ams.sensor] AMS_SENSORS in async_setup_entry= {'packet_size': 41, 'day_of_week': 'Monday', 'date_time': '2020-2-17 10:09:26', 'list_type': 1, 'active_power_n': 24813}

turbokongen, 2020-02-17:
Ok, looks wierd, but made another change

thomasja27, 2020-02-17:
Now its working :) 
But I think there are some decimals missing in "active_power_import" I have consumption now of 2-300w now, but it shows 20000-30000w. And sometimes in between it shows only 90w?

I know that the value out of the meter comes without a comma, so when the value is 10000 it is really 100,00w 

[image]
[image]

thomasja27, 2020-02-17:
The Amps are also wrong, L1 should be 0,7 amps

turbokongen, 2020-02-17:
I've updated the decimals. please try again.

turbokongen, 2020-02-17:
Does the attributes look ok for all the sensors?
Does the hourly sensor give proper readings?

thomasja27, 2020-02-17:
Now the values ​​look more correct :)
The name on active power import changes between "ams_active_power_import_None" and "ams_active_power_import_6970631405808469" 
[image]
[image]
Here is the hourly consumption, but I haven't got any values ​​yet on the latest version you uploaded. Looks like there is a comma missing here too
[image]

thomasja27, 2020-02-17:
I think it something with te 10 sec telegram thats wrong, the name on active import power changes from none to the meternumber every 10 sec. and the value is wrong every 10 sec.

turbokongen, 2020-02-17:
Looking at the previously posted logs at the 10sec raw telegrams, they all contain 0 for the imported power field. That explains why the value are "wrong", but it's the meter data.
I fixed the name issue, and changed the decimals on hourly energy readings.

turbokongen, 2020-02-17:
Ah, found the issue, new code uploaded.
Kaifa does not provide active power data i 10sec telegrams.

thomasja27, 2020-02-17:
Now it works! Have to wait for the hourly telegram, then we will check the value here as well.

thomasja27, 2020-02-17:
Desiamals must be moved four places on the hourly values. Should maybe have been rounded too?
[image]

turbokongen, 2020-02-17:
Are you sure? It's the cummulative kWh reading of the meter. You surely must have used more than 1kwh since you got the meter Installed?

thomasja27, 2020-02-17:
Ahh that is total measure! Thought there was energy consumption in an hour. Sorry!

Then it seems that all values ​​are correct according to what I can see. The meter is in my cabin, so I'll check all the values ​​in the display against Home Assistant. Going out for the weekend so look at it then.

Checked the log now, does this look okay?

2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams] [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 13, 12, 14, 255, 128, 0, 0, 2, 1, 6, 0, 0, 0, 96, 221, 0, 126]
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 96, 221]
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams] sensor_list= ['ams_active_power_import']
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams] AMS_DEVICES= ['ams_active_power_import', 'ams_voltage_l2', 'ams_current_l3', 'ams_current_l2', 'ams_current_l1', 'ams_voltage_l1', 'ams_active_power_import', 'ams_voltage_l3', 'ams_reactive_power_import', 'ams_reactive_power_export', 'ams_active_power_export', 'ams_reactive_energy_import', 'ams_active_power_import', 'ams_active_energy_import', 'ams_reactive_energy_export', 'ams_active_energy_export']
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams] sensors are the same, updating states
2020-02-17 13:12:14 DEBUG (Thread-2) [custom_components.ams] hass.data[AMS_SENSORS] = {'ams_active_power_import': {'state': 247.97, 'attributes': {'timestamp': '2020-2-17 13:12:14', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}}
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] updating sensor ams_active_power_import
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:14 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-17 13:12:15 DEBUG (MainThread) [custom_components.ams.sensor] Sensor not in hass.data
2020-02-17 13:12:16 DEBUG (Thread-2) [custom_components.ams] [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 17, 1, 13, 12, 16, 255, 128, 0, 0, 2, 1, 6, 0, 0, 0, 95, 80, 209, 126]
2020-02-17 13:12:16 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-17 13:12:16 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 95, 80]
2020-02-17 13:12:16 DEBUG (Thread-2) [custom_components.ams] sensor_list= ['ams_active_power_import']
2020-02-17 13:12:16 DEBUG (Thread-2) [custom_components.ams] AMS_DEVICES= ['ams_active_power_import', 'ams_voltage_l2', 'ams_current_l3', 'ams_current_l2', 'ams_current_l1', 'ams_voltage_l1', 'ams_active_power_import', 'ams_voltage_l3', 'ams_reactive_power_import', 'ams_reactive_power_export', 'ams_active_power_export', 'ams_reactive_energy_import', 'ams_active_power_import', 'ams_active_energy_import', 'ams_reactive_energy_export', 'ams_active_energy_export']

turbokongen, 2020-02-17:
Perfect!

thomasja27, 2020-02-17:
You need to add a button to donate, would love to support your work :)

turbokongen, 2020-02-20:
Could you please try this branch: https://github.com/turbokongen/hass-AMS/tree/kaifa
It contains some other optimlaizations.

thomasja27, 2020-02-21:
Its working:
[image]

But when I look at the data here, something is not right. Amp on 2 of the phases draws around 16A, and almost nothing on the last one. This should be 3-4000w, but it shows 9000w. Have tested with another OBIS decoder in terminal, and there it looks correct. Must check this when I go to the cabin today.

turbokongen, 2020-02-21:
Oki, I have put in some wrong data, I will fix.

turbokongen, 2020-02-21:
New fix up: https://github.com/turbokongen/hass-AMS/tree/aidon

thomasja27, 2020-02-21:
[image]
Move the decimals 2 places, then it ok :) sould be 4502W

turbokongen, 2020-02-21:
Done

thomasja27, 2020-02-21:
[image]

Now I think everything is right, you get a report when I come to the cabin and read directly on the meter :)

thomasja27, 2020-02-21:
The total energy messure is wrong, shows 100706.984kWh but the actual messurement is around 111XX kWh (checked with OBIS decoder in terminal)
[image]

turbokongen, 2020-02-21:
Do you have the latest code installed? I made some unit changes earlier today.

thomasja27, 2020-02-21:
Yes i have installed the whole package.

turbokongen, 2020-02-21:
Ok, can you paste me the log of the hour data package?

thomasja27, 2020-02-21:
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams] [126, 160, 155, 1, 2, 1, 16, 238, 174, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 21, 5, 11, 0, 10, 255, 128, 0, 0, 2, 18, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 53, 56, 48, 56, 52, 54, 57, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 18, 77, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 77, 6, 0, 0, 72, 131, 6, 0, 0, 25, 158, 6, 0, 0, 55, 89, 6, 0, 0, 8, 240, 6, 0, 0, 0, 0, 6, 0, 0, 9, 19, 9, 12, 7, 228, 2, 21, 5, 11, 0, 10, 255, 128, 0, 0, 6, 0, 170, 168, 131, 6, 0, 0, 0, 0, 6, 0, 1, 147, 95, 6, 0, 32, 207, 204, 65, 18, 126]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 0, 0]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 0, 0]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 0, 77]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 72, 131]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 25, 158]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 55, 89]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 8, 240]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 0, 0]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 9, 19]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [7, 228]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [6, 0, 170, 168, 131]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 0, 0, 0]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 1, 147, 95]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] fields= [0, 32, 207, 204]
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams] sensor_list= ['ams_active_power_export', 'ams_reactive_power_import', 'ams_reactive_power_export', 'ams_current_l1', 'ams_current_l2', 'ams_current_l3', 'ams_voltage_l1', 'ams_voltage_l2', 'ams_voltage_l3', 'ams_active_energy_import', 'ams_active_energy_export', 'ams_reactive_energy_import', 'ams_reactive_energy_export']
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams] AMS_DEVICES= ['ams_active_power_import', 'ams_active_power_export', 'ams_active_power_import', 'ams_reactive_power_import', 'ams_current_l2', 'ams_voltage_l1', 'ams_current_l1', 'ams_current_l3', 'ams_voltage_l2', 'ams_voltage_l3', 'ams_reactive_power_export']
2020-02-21 11:00:11 DEBUG (Thread-2) [custom_components.ams] new_devices= ['ams_active_energy_import', 'ams_active_energy_export', 'ams_reactive_energy_import', 'ams_active_power_import', 'ams_reactive_energy_export']
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] HUB in async_setup_entry-async_add_sensor= {'ams_active_power_export': {'state': 0.0, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_import': {'state': 0, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_export': {'state': 77, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_current_l1': {'state': 18.563, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l2': {'state': 6.558, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l3': {'state': 14.169, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_voltage_l1': {'state': 228.8, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l2': {'state': 0.0, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l3': {'state': 232.3, 'attributes': {'timestamp': '2020-2-21 11:00:10', 'meter_manufa [... cut ...] ase 230V 3-Wire meter', 'meter_serial': '6970631405808469', 'unit_of_measurement': 'kVArh', 'icon': 'mdi:gauge'} 
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams] sending sensor data
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] updating sensor ams_reactive_energy_export
2020-02-21 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] async_add_sensor in async_setup_entry

turbokongen, 2020-02-21:
Ok, new code uploaded :+1:

thomasja27, 2020-02-21:
Took a restart on the server just before this telegram, possibly that was why it failed

2020-02-21 12:00:11 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid frame CRC check
2020-02-21 12:00:11 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 155, 1, 2, 1, 16, 238, 174, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 21, 5, 12, 0, 10, 255, 128, 0, 0, 2, 18, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 53, 56, 48, 56, 52, 54, 57, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 17, 150, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 1, 44, 6, 0, 0, 67, 153, 6, 0, 0, 22, 196, 6, 0, 0, 55, 28, 6, 0, 0, 8, 225, 6, 0, 0, 0, 0, 6, 0, 0, 9, 12, 9, 12, 7, 228, 2, 21, 5, 12, 0, 10, 255, 128, 0, 0, 6, 0, 170, 188, 196, 6, 0, 0, 0, 0, 6, 0, 1, 147, 233, 6, 0, 32, 208, 90, 126]
2020-02-21 12:00:12 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid packet size 3
2020-02-21 12:00:12 DEBUG (Thread-2) [custom_components.ams] failed package: [120, 126, 126]
2020-02-21 12:00:13 DEBUG (Thread-2) [custom_components.ams.parsers.kaifa] Invalid packet size 40
2020-02-21 12:00:13 DEBUG (Thread-2) [custom_components.ams] failed package: [160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 228, 2, 21, 5, 12, 0, 12, 255, 128, 0, 0, 2, 1, 6, 0, 0, 17, 154, 23, 147, 126]

turbokongen, 2020-02-21:
Ok, just let me know if it is ok after the last change 💯

thomasja27, 2020-02-21:
Doesn't look right yet, the decimals have moved, but as you can see from the previous picture, there has been almost no consumption. 100706.984kWh -> 10070.7022kWh.
Several heaters are on, so consumption is constant at 4kW. Registered total consumption of the meter is over 11xxx so there is something that is not decoded correctly

[image]

turbokongen, 2020-02-21:
ok, found the issue. Missed the active energy byte by one.

thomasja27, 2020-02-21:
[image]
Still the same, but im going to the cabin now, so i will take a check directly on the meterdispaly when i get there.

turbokongen, 2020-02-21:
Ok logs will help, especially the fields loglines and the package data.

turbokongen, 2020-02-21:
I found something that could potentially cause this. New code is available.

thomasja27, 2020-02-21:
Now I have read the display on the meter. Total metering was right after the last change you made. Only the decimals are incorrect.
[image]

turbokongen, 2020-02-21:
Perfect! Pushed the corrections.

thomasja27, 2020-02-21:
Tested, all data is correct now :) Thanks!

commit f6e4943a (2020-02-21): Merge adion into master (#9)

* Add parsers directory and add kaifa parser + logic

* Modify checks

* Modify range

* Wrong name

* Correct decimals

* Remove meter_id from name

* Changed hourly reading decimals

* Revert name, update parser

* Update README.md

Add info about kaifa meters and meter options

* Update aidon.py

active power and time updated plus size and crc.Done

* Update aidon.py

TODO long 3ph

* Update aidon.py

first stab at aidon. TODO verify bytes

* Add aidon + optimizing

* Update README.md

* Update README.md

* Update kaifa.py

Correct power reading

* Correct deci [... cut]
closed 2020-02-21 by turbokongen