turbokongen/hass-AMS #12: little question about the kaifa meter

TheDiverM, 2020-03-21:
Hi
I have just installed your nice little software, but on my kaifa 3 phase meter it say 0 volts on ams_voltage_l2, is this something in you module or is it something wrong in my meter?

im using this mbus module:
https://www.aliexpress.com/item/32719562958.html?spm=a2g0s.9042311.0.0.c8314c4dpbv1pv

turbokongen, 2020-03-22:
Thanks for using my software.
Your issue is not a problem. It's how the meter is programmed. I do not know the reasons to why it is so.
It's noted here in this document on page 3: https://www.nek.no/wp-content/uploads/2018/11/Kaifa-KFM_001.pdf

TheDiverM, 2020-03-22:
another issue, migth be that I use a pi3, but the software runs for about 1 hour, then stops, while measurement comes in it is a load of 1.4 to 1.9, when it stops measure, load drops to 0.1

is the pi 3 just to small or is there something else.

the pi that has the mbus adapter only runs the norwegian han sensor, nothing else.
last log entry

from log on rpi:

2020-03-21 15:21:26 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for ams which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-03-21 15:22:01 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 425, in _async_add_entity
    raise HomeAssistantError(msg)
homeassistant.exceptions.HomeAssistantError: Entity id already exists: sensor.ams_active_power_import_none. Platform ams does not generate unique IDs
2020-03-21 15:43:30 WARNING (MainThread) [asyncio] socket.send() raised exception.
2020-03-21 15:43:31 WARNING (MainThread) [asyncio] socket.send() raised exception.
2020-03-21 15:43:31 WARNING (MainThread) [asyncio] socket.send() raised exception.
2020-03-21 15:43:31 WARNING (MainThread) [asyncio] socket.send() raised exception.
2020-03-21 15:43:31 WARNING (MainThread) [asyncio] socket.send() raised exception.
2020-03-21 15:43:31 WARNING (MainThread) [asyncio] socket.send() raised exception.`

turbokongen, 2020-03-22:
The unique_id error is known, but it will not stop homeassistant from working.
The socket.send() error might though.
I have the software running on a pi3 as well, without problems. (I also get the entity id error on the hourly package)

TheDiverM, 2020-03-22:
for me this halts the entire homeassistant collection of data from the meter, only way to get it running again is to restart the PI.

im open to suggestions for how to fix this issue

TheDiverM, 2020-03-22:
I found this line:

homeassistant.exceptions.HomeAssistantError: Entity id already exists: sensor.ams_active_power_import_none. Platform ams does not generate unique IDs

seems like system try to create the ID: 
sensor.ams_active_power_import_none
but it already exists

turbokongen, 2020-03-22:
I am running: 
`/bin/bash: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=3e5e2847bbc51da2ab313bc53d4bdcff0faf2462, stripped`
Please try the unique_id branch and see if that will fix it for you.

turbokongen, 2020-03-22:
Make sure you use the latest code. Just published another fix for this.
[https://github.com/turbokongen/hass-AMS/tree/unique_id](url)

TheDiverM, 2020-03-22:
now Im running on 32 bit hassio for rpi3 (was running on 64 bit when errors occured)
and now it seems like all runs fine, and no errors in the logs.
so maybe put something in readme about error in 64bit version..., ill give another feedback tomorrow if test runs fine all nigth

PR #15 (2020-03-28): Support Unique_id
Get's rid of the following error:
```
2020-03-21 15:22:01 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 425, in _async_add_entity
raise HomeAssistantError(msg)
homeassistant.exceptions.HomeAssistantError: Entity id already exists: sensor.ams_active_power_import_none. Platform ams does not generate unique IDs
```
Fixes #12
closed 2020-03-28 by turbokongen