turbokongen/hass-AMS #17: Missing attribute when using Kaifa one ohase meter

jorgen666, 2020-03-31:
I'm getting readings on all attributes except ams_active_power_import. This one doesn't show up in the version 1.04 on HA 0.107.7
I have Kaifa one phase meter.

turbokongen, 2020-03-31:
Could you please enable debug, and paste the log for about one minute?

jorgen666, 2020-03-31:
Yes I will, I just making a fresh install just to see if this gone bad in other ways

turbokongen, 2020-03-31:
I suspect mayde the unique_id change will make for a complete reinstall.
Meaning:
- Delete the integration
- Restart homeassistant
- Add the integration
See if that makes a change

jorgen666, 2020-03-31:
Ok, Jeg har reinstallert hele denne HA instansen. Er en av slave instansene. Så får man se

jorgen666, 2020-03-31:
Nå har jeg bygd opp denne maskinen i fra bunnen igjen, og får frem alle attributter untatt ams_active_power_  alle andre variabler blir lest av korregt

jorgen666, 2020-03-31:
HHer er loggen etter litt kjøring på denne 
[log ams.txt](https://github.com/turbokongen/hass-AMS/files/4411420/log.ams.txt)

jorgen666, 2020-03-31:
Hei

Jeg gikk tilbake til versjon 1.02 den fungerer korrekt med den strømmåleren jeg har.

turbokongen, 2020-04-01:
Kan du prøve: https://github.com/turbokongen/hass-AMS/tree/bugfix104
Jeg hadde brukt feil sensornavn for å få alle attributtene.

jorgen666, 2020-04-01:
Denne varianten førte ikke til noen endring i forhold til 1.04. Men kan ta å sammenligne den med den fungerende versjon 1.02

turbokongen, 2020-04-02:
Ok. Da funker det ikke å få alle attributes med kaifa som med aidon.
Pusher v1.05 senere idag.

PR #19 (2020-04-02): Bugfix v104
Fixes #17
closed 2020-04-02 by turbokongen