turbokongen/hass-AMS #21: No sensors after restart

jgallis, 2020-04-05:
Running Home Assistant 0.107.7 on VM,with Aidon 6525 meter and have this USB dongel https://www.aliexpress.com/item/32719562958.html
After first install the intergration works perfect, but after restart there are no sensors showing. Need to remove the intergration, take a new restart, and add the intergration again. 

This is how the intergration looks after restart: 
[image]

It looks like the sensors are there when I open the intergration: 
[image]

But no sensors in lovelace UI:
[image]

This is the setup: 
[image]

Have set the logs to debug and after restart it looks like the data is recived, but its not working. Can send you the logs if you want.

Hellowlol, 2020-04-05:
Seems similar to https://github.com/turbokongen/hass-AMS/issues/18

turbokongen, 2020-04-08:
Fixed in release v1.06

closed 2020-04-08 by turbokongen