turbokongen/hass-AMS #26: Failed package

Hellowlol, 2020-04-09:
Working on https://github.com/turbokongen/hass-AMS/pull/24 I saw some failed package in the log.

```
2020-04-09 12:25:01 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 160, 226, 43, 33, 19, 35, 154, 230, 231, 0, 15, 0, 0, 0, 0, 12, 7, 228, 4, 9, 4, 11, 25, 0, 255, 128, 0, 0, 2, 25, 10, 14, 75, 97, 109, 115, 116, 114, 117, 112, 95, 86, 48, 48, 48, 49, 9, 6, 1, 1, 0, 0, 5, 255, 10, 16, 53, 55, 48, 54, 53, 54, 55, 50, 55, 55, 49, 51, 57, 56, 49, 54, 9, 6, 1, 1, 96, 1, 1, 255, 10, 18, 54, 56, 52, 49, 49, 50, 49, 66, 78, 50, 52, 51, 49, 48, 49, 48, 52, 48, 9, 6, 1, 1, 1, 7, 0, 255, 6, 0, 0, 8, 122, 9, 6, 1, 1, 2, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 1, 3, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 1, 4, 7, 0, 255, 6, 0, 0, 0, 126]
2020-04-09 12:25:02 DEBUG (Thread-2) [custom_components.ams] failed package: [9, 6, 1, 1, 31, 7, 0, 255, 6, 0, 0, 1, 162, 9, 6, 1, 1, 51, 7, 0, 255, 6, 0, 0, 2, 76, 9, 6, 1, 1, 71, 7, 0, 255, 6, 0, 0, 2, 164, 9, 6, 1, 1, 32, 7, 0, 255, 18, 0, 230, 9, 6, 1, 1, 52, 7, 0, 255, 18, 0, 229, 9, 6, 1, 1, 72, 7, 0, 255, 18, 0, 230, 128, 20, 126]
```
Any idea what this might be?

turbokongen, 2020-04-09:
Have a look in the parser.
There are some check run with test_valid_data method.
Header crc
Frame Crc check
Size check
Data flag check
If any of those fail the package is discarded as failed.
That being said, a bit earlier in the log, the reason should be logged.

tor. 9. apr. 2020, 12:32 skrev Steffen Fredriksen <notifications@github.com

Hellowlol, 2020-04-09:
Oh, i disabled alot of logging as it more spammy then i wanted when i was working on another issue.. Ill close this for now. Ill reopen if it still a issue.

closed 2020-04-09 by Hellowlol