turbokongen/hass-AMS #29: Uncaught exception - IndexError: list index out of range

maxgyver87, 2020-07-12:
First of thank you so much for your work on this integration.

I am for the past couple of weeks trying to get this to work but I am out of options now and hope you can point me to what I am doing wrong.

I have a Aidon 6515 Installed.
Got the [MBUS module](https://www.aliexpress.com/item/32719562958.html?spm=a2g0s.9042311.0.0.c8314c4dpbv1pv)  you are using.
HAN-port is open.

Log can be found here:
[AMS_error_1.txt](https://github.com/turbokongen/hass-AMS/files/4908383/AMS_error_1.txt)

Setup with meter_manufacturer: 'auto' reports  [custom_components.ams] No parser detected
Setup with meter_manufacturer: 'auto' and parity: 'N' reports Invalid header CRC check - Uncaught exception - IndexError: list index out of range

Same result with other parity options except it doesn't report the Invalid header CRC check.

Please let me know if I can do anything else to get this sorted out.

turbokongen, 2020-07-12:
I will have a look at it later today. On mobile now. Most probably it's a matter of adjusting the code according to your meter. It was made with only one meter type tested.

turbokongen, 2020-07-12:
Please try the https://github.com/turbokongen/hass-AMS/tree/aidon_listtype_fix branch and see if that fixes the error for you.
The autodetect I will have to adress separately.

maxgyver87, 2020-07-12:
You sir are a hero :D
Both Aidon and auto seem to work now.

The active and reactive_energy entities are not being populated as seen here.
[image]
but so far it looks promising.

I included a minute of both meter_manufacturer: 'aidon' and 'auto' here.
[1min_working_ams_log.txt](https://github.com/turbokongen/hass-AMS/files/4909770/1min_working_ams_log.txt)

If you happen to have a link somewhere or vipps, id be happy to buy you a coffee or øl.

turbokongen, 2020-07-13:
Yes, the energy sensor gets populated at the hour the first time. Good to see it's working 👌🏻Enjoy!

turbokongen, 2020-07-13:
Fixed by #30

closed 2020-07-13 by turbokongen