turbokongen/hass-AMS #31: Aidon 6534 working

mariwing, 2020-07-21:
Hi,

Just wanted you to know that I have an Adion 6534 and it works just great with your component so you should consider moving it from "untested" in the readme.md. 

Thank you for all your work :-)

turbokongen, 2020-07-21:
Please let me know if the hourly values appear. If not, please post the packages posted around the hour mark, and I will inspect. 
You need debug logging enabled for that to appear.

mariwing, 2020-07-21:
No hourly values (or other accumalative values). Here are the log from around the hour mark.

2020-07-21 22:00:05 DEBUG (Thread-2) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 6, 239, 2, 2, 15, 0, 22, 27, 115, 1, 126]
2020-07-21 22:00:05 DEBUG (Thread-2) [custom_components.ams.parsers] fields= [0, 0, 6, 239]
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l1
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l2
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l3
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l2
2020-07-21 22:00:05 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l3
2020-07-21 22:00:07 DEBUG (Thread-2) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 6, 232, 2, 2, 15, 0, 22, 27, 107, 198, 126]
2020-07-21 22:00:07 DEBUG (Thread-2) [custom_components.ams.parsers] fields= [0, 0, 6, 232]
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l1
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l2
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l3
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l2
2020-07-21 22:00:08 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l3
2020-07-21 22:00:12 DEBUG (Thread-2) [custom_components.ams.parsers.aidon] Invalid packet size 396
2020-07-21 22:00:12 DEBUG (Thread-2) [custom_components.ams] failed package: [126, 161, 138, 65, 8, 131, 19, 235, 253, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 18, 2, 2, 9, 6, 1, 1, 0, 2, 129, 255, 10, 11, 65, 73, 68, 79, 78, 95, 86, 48, 48, 48, 49, 2, 2, 9, 6, 0, 0, 96, 1, 0, 255, 10, 16, 55, 51, 53, 57, 57, 57, 50, 56, 57, 53, 57, 49, 51, 49, 57, 53, 2, 2, 9, 6, 0, 0, 96, 1, 7, 255, 10, 4, 54, 53, 51, 52, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 6, 233, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 3, 179, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 46, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 3, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 32, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 141, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 153, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 142, 2, 2, 15, 255, 22, 35, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 228, 7, 21, 2, 21, 0, 0, 255, 0, 0, 0, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 0, 143, 135, 251, 2, 2, 15, 1, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 1, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 0, 2, 239, 2, 2, 15, 1, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 14, 232, 168, 2, 2, 15, 1, 22, 32, 202, 145, 126]
2020-07-21 22:00:12 DEBUG (Thread-2) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 6, 233, 2, 2, 15, 0, 22, 27, 190, 89, 126]
2020-07-21 22:00:12 DEBUG (Thread-2) [custom_components.ams.parsers] fields= [0, 0, 6, 233]
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l1
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l2
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l3
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l2
2020-07-21 22:00:13 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l3

turbokongen, 2020-07-21:
Please try https://github.com/turbokongen/hass-AMS/tree/aidon-bugfix to see if that fixes your problem 👍 
Found two issues: Wrong coded maxsize for large aidon packet, and a typo in the parser.

mariwing, 2020-07-22:
Thanks for looking into this. 

The error is gone but the sensors still have unknown state

2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams] data read from port=[126, 161, 138, 65, 8, 131, 19, 235, 253, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 18, 2, 2, 9, 6, 1, 1, 0, 2, 129, 255, 10, 11, 65, 73, 68, 79, 78, 95, 86, 48, 48, 48, 49, 2, 2, 9, 6, 0, 0, 96, 1, 0, 255, 10, 16, 55, 51, 53, 57, 57, 57, 50, 56, 57, 53, 57, 49, 51, 49, 57, 53, 2, 2, 9, 6, 0, 0, 96, 1, 7, 255, 10, 4, 54, 53, 51, 52, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 7, 35, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 2, 194, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 40, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 3, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 37, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 102, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 114, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 110, 2, 2, 15, 255, 22, 35, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 228, 7, 22, 3, 10, 0, 0, 255, 0, 0, 0, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 0, 143, 144, 177, 2, 2, 15, 1, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 1, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 0, 2, 239, 2, 2, 15, 1, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 14, 234, 122, 2, 2, 15, 1, 22, 32, 150, 34, 126]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 7, 35]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 2, 194]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 40]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [3, 2]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 37]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 102]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 114]
2020-07-22 11:00:10 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 110]
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l1
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l2
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l3
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l2
2020-07-22 11:00:11 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l3

turbokongen, 2020-07-22:
Ok. Thank you. Found another issue, Please download the branch again and test please:+1:

mariwing, 2020-07-23:
Now I get the total kwh from the meter:
 
sensor.ams_active_energy_export_7359992895913195 | 0.0 | meter_timestamp: 2020-7-23 10:00:00 meter_manufacturer: Aidon_V0001 meter_type: 6534 3-phase Meter with CB and Neutral Current Measurement meter_serial: 7359992895913195 obis_code: 1.0.2.8.0.255 unit_of_measurement: kWh icon: mdi:gauge friendly_name: ams_active_energy_export_7359992895913195

sensor.ams_active_energy_import_7359992895913195 | 94150.08 | meter_timestamp: 2020-7-23 10:00:00 meter_manufacturer: Aidon_V0001 meter_type: 6534 3-phase Meter with CB and Neutral Current Measurement meter_serial: 7359992895913195 obis_code: 1.0.1.8.0.255 unit_of_measurement: kWh icon: mdi:gauge friendly_name: ams_active_energy_import_7359992895913195

If this is what to be exptected then you can again rename the issue ;-)

turbokongen, 2020-07-23:
Perfect! Thanks for confirming. New release will be added. 💯

PR #32 (2020-07-23): Correct aidon 6534 packet size checker. Typo in parser.
Wrong size comparison for big packets. (6534)
Typo in parser.(Packet position)
Wrong referenced list_type for comparison in 3phase 6534 meter.
Update README.md
Fixes #31
closed 2020-07-23 by turbokongen