turbokongen/hass-AMS #34: Spike in current

mariwing, 2020-09-28:
Recently I have seen these kinds of spikes in current on my 3-phase meter. So far I have only seen it at l2. Is this something that could be handled in the code (disregard obviously corrupt data) or should I use a filter component in Home Assistant to get rid of them?

[image]

turbokongen, 2020-09-28:
The complete package is evaluated with two crc checksums before being decoded, so the corrupt data is coming from the meter itself.
Probably a firmware bug in the meter that have been introduced in a new firmware version, since this occured recently.
You should notify your grid supplier(nettleverandør) of this bug.
You can go back in your logs and see when this started, and supply that in the bugreport aswell.

mariwing, 2020-09-30:
Thank you for the swift reply. I was not questioning your code 😊 I was thinking more like having a sanity check or filter. I agree that this should be fixed by the meter manufacturer. In the meantime I will use the range filter: https://www.home-assistant.io/integrations/filter/. I have some zwave plugs that also send spiked numbers so I already filter those.

closed 2020-09-30 by mariwing