turbokongen/hass-AMS #37: Missing sensor

polroa, 2020-11-13:
Hi!
The ams_current_l2 sensor is not available in my Home Assistant. as seen in the enclosed screenshot in my sensors list. All the other ams sensors, a total of 13, works fine.
My configuration is :
Raspberry Pi 3B
Home assistant 0.117.6
HassOS 4.16
hass-AMS v 1.13

My AMS is Aidon 6525, and the module I'm using is https://www.ebay.com/itm/USB-To-MBUS-Slave-Module-Master-slave-Communication-Debugging-Bus-Monitoring/174403021186?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2648

My configuration.yaml looks like this:
# AMS Smartmåler Configuration
ams:
  serial_port: '/dev/ttyUSB0' # Required
  parity: 'N'  # Optional, defaults to 'N'
  meter_manufacturer: 'aidon' # Optional, defaults to 'auto'

[image]

Any ideas?

turbokongen, 2020-11-14:
Some aidon meters does not provide l2 data. See https://www.nek.no/wp-content/uploads/2019/02/Aidon-HAN-Interface-Description-v11A-ID-34331.pdf
Could you send me the debug output of the large packets?

polroa, 2020-11-15:
Is this what you asked for?

[home-assistant.log](https://github.com/turbokongen/hass-AMS/files/5542470/home-assistant.log)

turbokongen, 2020-11-15:
Yes, thank you. That is package type 12. That confirms my initial thought. You have a "3W" meter which does not provide data for I2 according to aidon's description: `**missing from 3-phase 3W meters and 1-phase meters` in page 11 of their documentation.
3W is 3-wire i'm guessing??
I will be closing this as it is normal behaviour.

closed 2020-11-15 by turbokongen