turbokongen/hass-AMS #38: aliexpress module has no USB - so how to do that in the yaml?

dkcsn, 2021-01-24:
Hi - and what an awesome plugin.
This module is referered as working
[(https://www.aliexpress.com/item/32834331647.html?spm=a2g0o.detail.1000060.1.74cfdcd4qts4jp)]

But for the yaml config - USB seems to be the only way?
ams:
serial_port: '/dev/ttyUSB0' # Required
And the aliexpress module has no USB - so how to do that in the yaml?
I have one of the supported Kamstrup and have ordered the IR eye reader, so it would be great to know the idea of the non USB module?

turbokongen, 2021-01-24:
Hi, and thank you!
I have not tested this device myself, just added it to the working devices because others have used it.
You can use any "device" in the /dev directory. Ne need for usb. If you have a serial port instead of USB, use that. It is not required to be USB, just the path to the serial port data is received to.

dkcsn, 2021-01-24:
So fast reply - Thank you :-)
I see - the distance from my meter to my Home Assistance, require that I go Wifi, so I was thinking of ESP Home module, connected to the reader. Any thoughts of an enhancement like this? - or can i take the water signal into ESP Home and create the same sensors i HA so your ADDON can do the magic afterwords?

turbokongen, 2021-01-24:
I'm sorry, I have no experience in doing such a thing. My plugin parses the raw data from the meter, so you might be out of luck.

Hellowlol, 2021-01-24:
I had the same issue. I just use a spare rpi with home assistant installed and I just use the mqtt statestream to push data.

turbokongen, 2021-03-08:
Closed by clarifying the readme.

closed 2021-03-08 by turbokongen