turbokongen/hass-AMS #39: Will it be possible to add several instances? I have two AMS in ny house. (House+flat)?

bthovda, 2021-01-27:


TheDiverM, 2021-01-27:
I have a separate pi for each meter and, then use remote homeassistant to collect the data into my central homeassistant.
Also my pi has a possibility for din mount into fuse box.

bthovda, 2021-01-27:
Yes. I have been down that route.
I guess i had 5 pi’s running at some point.
Trying to centralize into one Intel nuc i7 and one pi for my ventilation.

turbokongen, 2021-01-30:
I have no experience with this, but in theory this should work. The sensors are identified by meter_id and data.
I'm not sure about the protocol of Mbus. but it might be possible to connect all the meter cables to one usb stick? 
I have not looked into the electrical spec of Mbus, but if it is a bus, it should work.

TheDiverM, 2021-01-30:
Just my experience but it might have changed.

I have only been able to run this module stable over long times on a 32 bit homeassistant, 

So if someone has experience on running this stable on a 64 bit nuc it would be nice to hear

bthovda, 2021-04-20:
> Just my experience but it might have changed.
> 
> I have only been able to run this module stable over long times on a 32 bit homeassistant,
> 
> So if someone has experience on running this stable on a 64 bit nuc it would be nice to hear

Yes. I have two intel nucs with proxmox running home assistant as virtual machines. 
No problems with this integration. (One Aidon and one kamstrup AMS)

Hellowlol, 2021-04-20:
This should work just fine. I think the only thing that is needed is to disable the check for single instance.

PR #47 (2021-06-01): Allow multiple instances
Fixes #39
closed 2021-06-01 by turbokongen