turbokongen/hass-AMS #40: No Device or Entities is created

CyberSecDog, 2021-02-10:
I installed the component from HACS.
No problem installing but i cant find any new device or entities in HA after configuring it.

My configuration is :
Virtual machine in proxmox 
Core Version core-2021.2.2
Supervisor Version supervisor-2021.02.6
Operating System Home Assistant OS 5.11
hass-AMS v 1.13

  serial_port: '/dev/ttyUSB0'
  parity: 'N'
  meter_manufacturer: 'auto'  or 'aidon'

I'm using : https://www.aliexpress.com/item/32719562958.html?spm=a2g0s.9042311.0.0.c8314c4dpbv1pv
And the meter is a: Aidon 6484SE

Know that HA now have upgraded to Python 3.9

turbokongen, 2021-02-11:
Please check logs for any hint, and enable debug logging for ams component. 
I am running dev branch with the exact same usb adapter, though with kamstrup meter.

turbokongen, 2021-02-22:
Fixed by #43

closed 2021-02-22 by turbokongen