turbokongen/hass-AMS #41: Swedish HAN-port data profile support?

Bo1jo, 2021-02-17:
My meter is an Aidon 6484SE. Any chans the code could be supplemented to support the Swedish HAN-port data profile? The Swedish HAN-protocol is sending 1 list every 10 s (same list every time).

(page 15)
[aidonfd-rj45-han-interface-se-v14a-1.pdf](https://github.com/turbokongen/hass-AMS/files/5998267/aidonfd-rj45-han-interface-se-v14a-1.pdf)

turbokongen, 2021-02-18:
By the look of the documentation you provided, the data transmitted is the same. So in theory it should work. You will get None for meter type. Please eneable debugging when installing the package. The raw packages will be usefull alongside any error messages.

turbokongen, 2021-02-18:
I have created a new branch: https://github.com/turbokongen/hass-AMS/tree/unknown_meter
Only added data for the addon modules to prevent None for meter type. please use that for testing.:100:

Bo1jo, 2021-02-18:
Thanks! Tried the new branch. I'm getting an Uncaught thread exception error:

Logger: root
Source: custom_components/ams/__init__.py:184
First occurred: 2:39:29 PM (1 occurrences)
Last logged: 2:39:29 PM

Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 155, in connect
    self.meter_manufacturer = self._find_parser(pkg)
  File "/config/custom_components/ams/__init__.py", line 189, in _find_parser
    if _test_meter(pkg, AIDON_METER_SEQ):
  File "/config/custom_components/ams/__init__.py", line 184, in _test_meter
    for i in range(len(pkg)):
TypeError: object of type 'NoneType' has no len()

and a No parser detected warning:

Logger: custom_components.ams
Source: custom_components/ams/__init__.py:198
Integration: Norwegian AMS (documentation, issues)
First occurred: 2:37:23 PM (13 occurrences)
Last logged: 2:39:23 PM

No parser detected

turbokongen, 2021-02-18:
Ok, it bombs out detecting meter type. Package is None. 
Can you try to manually set meter manufacturer to Aidon?
Also can you post the "data read from port" debug line from the log?

Bo1jo, 2021-02-18:
Ok, I'm not that experienced with HA, but this is what i find in the log (after manually seting manufacurer to Aidon):

2021-02-18 21:06:24 DEBUG (Thread-3) [custom_components.ams.parsers.aidon] Invalid packet size 581
2021-02-18 21:06:24 DEBUG (Thread-3) [custom_components.ams] failed package: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 2, 18, 4, 21, 6, 20, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 2, 229, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 2, 108, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 11, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 21, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 9, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 33, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 22, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 22, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 0, 133, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 233, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 1, 157, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 1, 25, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 0, 191, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 110, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 80, 16, 31, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 107, 237, 54, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 32, 48, 18, 2, 2, 15, 0, 22, 32, 109, 58, 126]

turbokongen, 2021-02-19:
Ok, I need to modify the parser to recognize these packages. Can you pastebin or upload a text with more failed packages.
I need to establish a way to recognize the format over time.

Bo1jo, 2021-02-20:
[ams_log.txt](https://github.com/turbokongen/hass-AMS/files/6014204/ams_log.txt)

OK! I've uploded a text.
I'm using this usb adapter: https://www.aliexpress.com/item/32719562958.html?spm=a2g0s.9042311.0.0.c8314c4dpbv1pv

turbokongen, 2021-02-20:
Perfect!
I have uploaded a first draft of the code for you to test.
Please test with both 'auto' and 'aidon_se' for meter type.
Please test :100:

Bo1jo, 2021-02-20:
Thanks!

With **'auto'** I get basically nothing. From the log:
2021-02-20 19:21:27 INFO (MainThread) [homeassistant.setup] Setting up ams
2021-02-20 19:21:27 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-02-20 19:21:33 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-02-20 19:21:33 DEBUG (MainThread) [custom_components.ams] Finish init of AMS

Then nothing happens. No data read is registered in the log, no entities created etc.

**But with setting 'aidon_se' magic happens!**
26 entites created. I've pasted example of data read below. However dataread for L1 seams to be off. L2 and L3 seams ok.
These entities gives unreasonably high or low values or zero:
ams_active_energy_export_00
ams_active_energy_import_00
ams_active_power_export_l1_00
ams_active_power_import_l1_00
ams_reactive_energy_export_00
ams_reactive_energy_import_00
ams_reactive_power_export_l1_00
ams_reactive_power_import_l1_00
ams_voltage_l1_00

2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 2, 20, 6, 19, 50, 10, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 9, 232, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 5, 205, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 31, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 58, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 36, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 55, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 29, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 60, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 2, 163, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 1, 36, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 4, 151, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 2, 214, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 2, 190, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 208, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 81, 157, 217, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 108, 159, 190, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 32, 59, 81, 2, 2, 15, 0, 22, 32, 174, 247, 126]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] list_type is 27
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [7, 229]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 9, 232]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 5, 205, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 31, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 58, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 36, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 55, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 29, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [9, 60, 2]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 2, 163]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 1, 36]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 4, 151]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 2, 214]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 2, 190]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 1, 208]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [6, 1, 81, 157]
2021-02-20 19:50:13 DEBUG (Thread-4) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
[... cut ...] e=0.0.0.2.2.15, unit_of_measurement=kVArh, icon=mdi:gauge, friendly_name=ams_reactive_energy_export_00 @ 2021-02-20T19:42:43.465634+01:00>, new_state=<state sensor.ams_reactive_energy_export_00=3709998.11; meter_timestamp=2021-2-20 19:50:10, meter_manufacturer=Aidon_H0001, meter_type=6484 RF2-system module Slave 500mW Integrated HAN, meter_serial=00, obis_code=0.0.0.2.2.15, unit_of_measurement=kVArh, icon=mdi:gauge, friendly_name=ams_reactive_energy_export_00 @ 2021-02-20T19:42:43.465634+01:00>>

turbokongen, 2021-02-20:
I will investigate in the morning. Probably some wrong mapping at my side 🚽

turbokongen, 2021-02-20:
Found the issue BTW. Please try newly uploaded code.
It is expected that the export sensors will be 0. (You are not producing any electricity?)
Some of the values may need to be factored. Please check that they are correct, but maybe off by comma separation.

Bo1jo, 2021-02-20:
I'm not producing electricity, so export sensors will be 0.

Tried the new code. Seems to be correct at a quick glance. I'll have a closer look tomorrow!

Bo1jo, 2021-02-21:
Thanks! I've been taking a closer look on the numbers.

Most values seems to be correct. All the momentary values appears to be correct but the cumulative values seems to be off.
ams_active_energy_export_00 gives 0, which i correct. But since I don't produce energy i can't verify.
ams_active_energy_import_00 value changes very slowly. Value is 1 007 498.78 kWh (which seems like a very big number) and rising very slowly. The slow increase don't seem in line with the power measurements for L1, L2 and L3 (which all seems to be correct).

ams_reactive_energy_export_00 =21142.9 kVArh - value slowly rising
ams_reactive_energy_import_00 =71358.16 kVArh - value slowly rising
I'm not an electrical engineer, so I don't completely get the "reactive energy" theory. But as I understand it should be correct to have som reactive energy export, even if I don't produce energy, since reactive energy is energy not 100 % used by the consuming product and some of it is returned to the power grid. But the values reported for import/export seems odd and not in line with the momentary values for reactive energi L1, L2 and L3.

The aidon meter manual describes this measurements as:
Cumulative hourly active import energy (A+) (Q1+Q4) kWh
Cumulative hourly active export energy (A-) (Q2+Q3) kWh
Cumulative hourly reactive import energy (R+) (Q1+Q2) kVArh
Cumulative hourly reactive export energy (R-) (Q3+Q4) kVArh

To me it sounds like the values should reset to 0 every hour, but they just keep slowly increasing. But that maybe just me misunderstanding the manual.

No entities is created with "auto" meter type, still must use "aidon_se".

turbokongen, 2021-02-21:
The cummulative sensor will never be zeroed. They will also slowly rise. If you check your meter reading at the meter to compare, I will figure out the correct scaler for the cummulative sensors. Thank you for the feedback 👍

Bo1jo, 2021-02-22:
Thank you for your swift replies and great work!

_Values last night:_
ams_active_energy_import_00: 1007499.22 kWh 
Total import from my meter reading: 22176 kWh

_Values this morning:_
ams_active_energy_import_00: 1007499.87 kWh
Total import from my meter reading: 22193 kWh

_Diff:_
ams_active_energy_import_00: 0.65
meter reading: 17

Let me know if i should check the values again later on, to get more measurements.

turbokongen, 2021-02-22:
Perfect.
Found some issues.
Please try newly uploaded code. 💯

Bo1jo, 2021-02-22:
Tried it, and now ams_active_energy_import_00 is the same as the meter reading. 👍
As far as I can tell all the sensors give correct readings now!

Found some more info in the log when using setting auto (have to use aidon_se to make it work):
2021-02-22 09:10:03 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-02-22 09:10:09 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-02-22 09:10:09 DEBUG (MainThread) [custom_components.ams] Finish init of AMS

2021-02-22 09:10:12 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-02-22 09:10:12 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-02-22 09:10:12 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-02-22 09:10:12 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-02-22 09:10:12 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer

turbokongen, 2021-02-22:
Excellent!
I did a small change to the datatype being searched for in the autodetection routine.
Please try latest.

Bo1jo, 2021-02-22:
2021-02-22 11:42:22 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-02-22 11:42:22 INFO (Thread-3) [custom_components.ams] Detected Swedish Aidon meter by list type

Works with setting "auto"! Great job!

turbokongen, 2021-02-22:
Excellent! I will put up a New release later today. Thanks a million 🕺

closed 2021-02-22 by turbokongen