turbokongen/hass-AMS #42: The ams platform for the sensor integration does not support platform setup

TomaszKn, 2021-02-18:
Hi
I'm testing you component.
When I'm install and config it in configuration.yaml home assistant report problem:
`ERROR (MainThread) [homeassistant.components.sensor] The ams platform for the sensor integration does not support platform setup. Please remove it from your config.`

How fix it?

Core: core-2021.2.3
Supervisor: supervisor-2021.02.9
Operating System Raspbian GNU/Linux 10 (buster) (rpi 4)
Docker version 19.03.13

turbokongen, 2021-02-18:
You should _not_  have an entry under ```sensor:``` It is a component config. Please follow the documentation in the readme for yaml setup.:
```yaml
ams:
  serial_port: '/dev/ttyUSB0' # Required
  parity: 'N'  # Optional, defaults to 'N'
  meter_manufacturer: 'auto' # Optional, defaults to 'auto'
```

turbokongen, 2021-02-22:
Did this solve your problem?

TomaszKn, 2021-02-22:
Yes
THX

closed 2021-02-23 by turbokongen