turbokongen/hass-AMS #44: Sensors become unavailable after restart

Bo1jo, 2021-03-25:
I'm having the problem that all my sensors become unavailable after restart.

2021-03-25 09:30:34 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-03-25 09:30:40 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-03-25 09:30:40 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid packet size 41
2021-03-25 09:30:40 DEBUG (Thread-4) [custom_components.ams] failed package: [1, 0, 3, 8, 0, 255, 6, 0, 118, 108, 51, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 33, 79, 55, 2, 2, 15, 0, 22, 32, 142, 124, 126]

If i remove the integration and the folder in custom_components, reboot and reinstall everything works fina again. Until the next restart...

AMS version 1.4.0

Bo1jo, 2021-03-26:
The problem don't seem to occur every reboot. After last reboot (nothing changed in HA regarding hass-AMS):

2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 3, 26, 5, 9, 36, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 42, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 65, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 14, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 21, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 7, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 68, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 60, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 53, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 0, 254, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 253, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 1, 168, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 1, 18, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 0, 145, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 93, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 106, 157, 188, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 118, 155, 226, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 33, 94, 115, 2, 2, 15, 0, 22, 32, 226, 50, 126]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] list_type is 27
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [7, 229]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 3, 42]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 65]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 14]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 21]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 7]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [9, 68]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [9, 60]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [9, 53]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 254]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 253]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 1, 168]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 1, 18]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 145]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 93]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [1, 106, 157, 188]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 0, 0, 0]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 118, 155, 226]
2021-03-26 09:36:31 DEBUG (Thread-3) [custom_components.ams.parsers] fields= [0, 33, 94, 115]

turbokongen, 2021-03-26:
Hi, sorry for not replying sooner.

At first glance, the first log you provided, show that invalid data is received from the meter(serial) so nothing get's decoded.
Easy way to see this is that all packages start with 126 and ends with 126.
At the same log it also shows that it has managed to detect the swedish meter, so at least one good package has been received.
But the first package is not used for decoding, only detecting meter type.
The component will not bomb out because of that, but it might take 10-30-60seconds (if more packages are invalid) to update.
Does data continue to come in to the logs after what you provided?

Bo1jo, 2021-03-26:
Hi! No problem! I don't expect 24/7 tech support :)

I've forgot to mention that I've changed my meter to aidon_se during my attempts to solve the issue. So the logs provided isn't with auto detection of meter type.

After getting the error message in my first log no new log entries occur, even after several minutes. So data don't continue to come.

After another reboot (after the second, successful, log i provided) I get this error:

2021-03-26 09:46:20 INFO (MainThread) [homeassistant.setup] Setting up ams
2021-03-26 09:46:20 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-03-26 09:46:26 ERROR (MainThread) [homeassistant.setup] Error during setup of component ams
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/config/custom_components/ams/__init__.py", line 70, in async_setup
    _setup(hass, config[DOMAIN])
  File "/config/custom_components/ams/__init__.py", line 61, in _setup
    hub = AmsHub(hass, config)
  File "/config/custom_components/ams/__init__.py", line 113, in __init__
    self._ser = serial.Serial(
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
OSError: [Errno 5] I/O error

And then after another reboot everything works fine again...

Bo1jo, 2021-03-26:
**"auto" setting to make no difference. Works after reboot:**

2021-03-26 10:12:36 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-03-26 10:12:41 INFO (Thread-4) [custom_components.ams] Autodetecting meter manufacturer
2021-03-26 10:12:41 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-03-26 10:12:42 DEBUG (Thread-4) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-03-26 10:12:42 INFO (Thread-4) [custom_components.ams] Detected Swedish Aidon meter by list type
2021-03-26 10:12:52 DEBUG (Thread-4) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 3, 26, 5, 10, 12, 50, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 15, 137, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 8, 94, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 64, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 77, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 65, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 38, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 33, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 31, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 5, 84, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 2, 42, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 6, 74, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 3, 28, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 5, 56, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 2, 190, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 106, 161, 4, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 118, 157, 16, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 33, 94, 192, 2, 2, 15, 0, 22, 32, 119, 27, 126]
2021-03-26 10:12:52 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] list_type is 27

**But then after another reboot:**

2021-03-26 10:25:56 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/ttyUSB0
2021-03-26 10:26:01 ERROR (MainThread) [homeassistant.setup] Error during setup of component ams
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/config/custom_components/ams/__init__.py", line 70, in async_setup
    _setup(hass, config[DOMAIN])
  File "/config/custom_components/ams/__init__.py", line 61, in _setup
    hub = AmsHub(hass, config)
  File "/config/custom_components/ams/__init__.py", line 113, in __init__
    self._ser = serial.Serial(
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
OSError: [Errno 5] I/O error

**And after another reboot everything works fint. Seems to be working every other time**

turbokongen, 2021-03-26:
Good info! 
That error is actually something happening with the usb port. The serial component cannot read from it. How are the serial port defined? Via YAML or with Integration page?
Errno 5 is error at OS level, I might be able to catch it and retry the setup. Please check that your serial port does not change after a host reboot.
Does the error appear at host reboot or homeassistant restart?

Bo1jo, 2021-03-26:
The serial port is defined in YAML (I've tried with Integration as well, but the problem occurs then as well)

The error appears at host reboot (homeassistant restart don't seem to be any problem)

turbokongen, 2021-03-26:
Perfect. Solution: Instead of using /dev/ttyUSB find the device in /dev/serial/by-id and use that path instead. If it does not exist, create a new udev rule. 
This articles might help:https://unix.stackexchange.com/questions/394411/persistent-names-for-physical-usb-ports
Please report your result:)

turbokongen, 2021-03-26:
Sorry, posted the wrong article. Edited the original post.

Bo1jo, 2021-03-27:
Thanks! Tried it. Sensors still unavailable after most restarts (works in about 1 restart of 6). Don't matter if setting is "auto" or "aidon_se".

When it don't work this is all i can find in the log:
2021-03-27 16:44:35 INFO (MainThread) [homeassistant.setup] Setting up ams
2021-03-27 16:44:35 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-03-27 16:44:41 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-03-27 16:44:41 INFO (MainThread) [homeassistant.setup] Setup of domain ams took 5.8 seconds
2021-03-27 16:44:44 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.ams

No more entries regarding AMS in the log after that.

When it works it looks similar:
2021-03-27 17:07:27 INFO (MainThread) [homeassistant.setup] Setting up ams
2021-03-27 17:07:27 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-03-27 17:07:27 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-03-27 17:07:27 INFO (MainThread) [homeassistant.setup] Setup of domain ams took 0.1 seconds
2021-03-27 17:07:31 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.ams
2021-03-27 17:07:33 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 3, 27, 6, 17, 7, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 9, 68, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 5, 80, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 36, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 52, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 29, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 31, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 18, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 20, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 3, 7, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 1, 57, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 4, 21, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 2, 110, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 2, 19, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 165, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 107, 97, 115, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 118, 196, 59, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 33, 115, 254, 2, 2, 15, 0, 22, 32, 109, 71, 126]
2021-03-27 17:07:33 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] list_type is 27
and so on. 

So there is never any data read when it don't work, but no error message either. Do you have any clue? Hardware error?

turbokongen, 2021-03-28:
It looks to be a hardware error or OS error. I will try to add a catch for the Errno5 to try and reconnect, maybe also I try a timer for no data. I will notify when code is ready. 👍

turbokongen, 2021-03-28:
Actually, It will not do any good. It will just loop on reconnect.
1. Have you tried restart of Home-assistant after a reboot start? Does that fix it? 
2. If 1 does not fix it, stop homeassistant, unplug and replug the usb adapter. Does that fix it?

Bo1jo, 2021-03-29:
Ok, I've done some testing.

1. I've tried it. Sometimes it works after restart most of the times it don't. Same as after a reboot.
2. Tried that as well. Since I'm running HA in ESXI it isn't as simple as just unplugging the usb adapter and the replug it. But II've done some testing in ESXI and unplugged the adapter physically as well. Anyway, it don't seem to fix it.
3. How do I set the baurate manually? I tried adding baudrate to the yaml config, but i just gives error messages.

turbokongen, 2021-03-29:
Ok, I see. 
Well the good news is that we have isolated where it fails. At OS/Hardware level.
Bad news is I have no experience with dealing with such an issue. First thing that comes to mind is passthough.
Maybe this article will help you out:
It is for PCi, but can also be used for USB
https://kb.vmware.com/s/article/1010789

Bo1jo, 2021-03-30:
Thanks! I'll se if I can figure it out. I will try some different approaches to see if I can find a problem at OS level. I'm also thinking about trying another USB to Mbus-adapter to see if that solves it. Delivery from aliexpress takes a few weeks however. I'll report my findings.

Bo1jo, 2021-08-07:
Havn't made any progress with my current module, so I changed to https://www.aliexpress.com/item/32834331647.html?spm=a2g0o.detail.1000060.1.74cfdcd4qts4jp
Now I'm getting the error below: (meter_manufacturer: 'auto', but I get error when setting meter_manufacturer: 'aidon_se' as well)

2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:31 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:31 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:31 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:31 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:31 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:31 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:31 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:32 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:32 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:32 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:32 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:51 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:51 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:51 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:51 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:51 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:51 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:51 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:40:52 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:40:52 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:40:52 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:40:52 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:40:52 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:40:52 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:40:52 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-07 22:41:02 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-07 22:41:02 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-07 22:41:02 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:41:02 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:41:02 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:41:02 WARNING (Thread-3) [custom_components.ams] No parser detected
[... cut ...]
2021-08-07 22:43:21 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-07 22:43:21 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-07 22:43:21 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-07 22:43:21 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-08-07 22:43:21 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer

Any clues why its not working?

Bo1jo, 2021-08-08:
This is the log entries when meter_manufacturer: 'aidon_se'

2021-08-08 08:01:16 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2021-08-08 08:01:16 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-08-08 08:01:19 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid packet size 13
2021-08-08 08:01:19 DEBUG (Thread-4) [custom_components.ams] failed package: [254, 95, 254, 184, 253, 61, 223, 254, 0, 255, 152, 252, 126]
2021-08-08 08:01:20 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:01:20 DEBUG (Thread-4) [custom_components.ams] failed package: [192, 225, 253, 120, 33, 137, 255, 254, 100, 255, 255, 255, 48, 62, 208, 159, 130, 246, 195, 63, 177, 253, 12, 255, 28, 254, 252, 238, 236, 254, 254, 9, 79, 96, 12, 255, 254, 82, 255, 240, 16, 153, 243, 248, 64, 255, 217, 219, 150, 241, 29, 255, 161, 127, 114, 252, 19, 243, 159, 111, 124, 224, 159, 48, 249, 94, 143, 169, 25, 50, 135, 255, 255, 62, 253, 164, 18, 176, 19, 192, 248, 242, 252, 200, 135, 205, 252, 191, 248, 230, 255, 126]
2021-08-08 08:01:29 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:01:29 DEBUG (Thread-4) [custom_components.ams] failed package: [254, 255, 255, 253, 255, 255, 255, 127, 11, 255, 225, 191, 208, 134, 255, 254, 129, 111, 208, 231, 246, 127, 248, 223, 86, 255, 118, 254, 115, 144, 191, 251, 28, 254, 152, 95, 27, 255, 62, 255, 127, 249, 187, 0, 220, 96, 255, 36, 254, 89, 126]
2021-08-08 08:01:49 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:01:49 DEBUG (Thread-4) [custom_components.ams] failed package: [255, 62, 130, 254, 112, 15, 163, 127, 95, 128, 255, 251, 249, 127, 8, 254, 255, 7, 223, 62, 255, 60, 244, 255, 255, 48, 62, 176, 191, 76, 244, 59, 240, 137, 241, 78, 230, 222, 240, 31, 255, 254, 255, 55, 247, 255, 96, 255, 201, 130, 153, 255, 7, 128, 236, 254, 255, 193, 254, 255, 250, 236, 255, 240, 255, 128, 252, 222, 252, 248, 72, 191, 183, 255, 195, 15, 254, 236, 248, 223, 2, 57, 176, 37, 63, 254, 255, 23, 255, 247, 244, 255, 101, 255, 48, 255, 30, 2, 255, 255, 219, 129, 255, 72, 255, 63, 181, 123, 8, 249, 255, 15, 31, 247, 62, 6, 63, 184, 94, 255, 252, 1, 191, 3, 63, 144, 191, 252, 25, 255, 139, 255, 255, 132, 255, 176, 144, 127, 236, 255, 255, 0, 190, 237, 240, 255, 152, 253, 6, 255, 223, 253, 204, 255, 97, 255, 224, 49, 254, 216, 254, 254, 251, 127, 202, 125, 246, 232, 255, 143, 96, 156, 255, 227, 159, 255, 214, 152, 127, 252, 240, 253, 232, 254, 158, 252, 248, 255, 255, 255, 255, 240, 114, 190, 131, 159, 252, 251, 3, 63, 1, 255, 255, 254, 144, 63, 248, 255, 196, 95, 240, 251, 255, 2, 191, 24, 255, 167, 221, 224, 254, 8, 15, 19, 255, 248, 110, 57, 250, 3, 255, 255, 55, 254, 86, 249, 255, 255, 8, 246, 194, 255, 177, 251, 255, 228, 102, 255, 255, 32, 255, 126]
2021-08-08 08:01:59 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:01:59 DEBUG (Thread-4) [custom_components.ams] failed package: [19, 255, 255, 135, 112, 255, 3, 254, 255, 248, 248, 249, 151, 252, 70, 6, 175, 176, 140, 255, 152, 255, 225, 224, 156, 244, 255, 255, 180, 244, 225, 251, 129, 188, 240, 254, 246, 248, 254, 254, 254, 142, 253, 255, 252, 63, 246, 195, 239, 190, 255, 23, 3, 160, 252, 248, 251, 65, 31, 224, 143, 248, 127, 248, 232, 9, 254, 12, 1, 255, 255, 239, 224, 255, 8, 164, 131, 108, 211, 25, 160, 128, 31, 161, 126]
2021-08-08 08:02:08 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:02:08 DEBUG (Thread-4) [custom_components.ams] failed package: [74, 255, 124, 253, 16, 252, 239, 62, 120, 244, 177, 251, 198, 252, 243, 250, 62, 248, 251, 166, 252, 239, 157, 255, 176, 255, 13, 239, 254, 254, 249, 127, 149, 31, 122, 252, 12, 123, 191, 255, 127, 6, 255, 191, 255, 255, 255, 248, 11, 223, 156, 209, 184, 250, 227, 251, 195, 112, 159, 255, 129, 252, 217, 223, 197, 255, 255, 255, 239, 254, 255, 255, 126]
2021-08-08 08:02:18 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:02:18 DEBUG (Thread-4) [custom_components.ams] failed package: [191, 194, 255, 153, 255, 64, 220, 0, 111, 201, 255, 60, 16, 127, 13, 254, 124, 240, 253, 121, 208, 112, 255, 255, 254, 100, 255, 254, 216, 240, 255, 27, 255, 251, 6, 111, 253, 127, 140, 252, 255, 36, 255, 255, 191, 191, 243, 231, 25, 30, 15, 255, 200, 255, 9, 255, 243, 240, 3, 64, 207, 2, 255, 158, 191, 189, 223, 50, 254, 240, 240, 255, 253, 255, 255, 64, 255, 104, 48, 31, 36, 127, 193, 128, 147, 139, 144, 223, 247, 158, 248, 232, 191, 240, 192, 254, 226, 246, 191, 192, 228, 249, 226, 253, 229, 252, 254, 254, 255, 158, 255, 255, 72, 255, 247, 152, 15, 255, 248, 255, 228, 126]
2021-08-08 08:02:29 DEBUG (Thread-4) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-08 08:02:29 DEBUG (Thread-4) [custom_components.ams] failed package: [19, 255, 230, 0, 255, 222, 252, 237, 253, 136, 255, 248, 255, 127, 232, 127, 240, 223, 48, 255, 35, 39, 27, 214, 255, 254, 10, 255, 207, 7, 255, 65, 240, 254, 246, 254, 96, 255, 191, 48, 254, 255, 247, 224, 64, 254, 4, 254, 223, 35, 241, 64, 249, 25, 23, 167, 253, 223, 128, 255, 59, 201, 188, 254, 236, 12, 231, 123, 255, 255, 96, 251, 6, 252, 223, 255, 224, 255, 27, 6, 240, 159, 242, 255, 208, 255, 2, 194, 127, 236, 198, 195, 255, 231, 194, 236, 242, 252, 220, 253, 224, 195, 254, 190, 252, 130, 254, 254, 239, 255, 173, 86, 255, 128, 255, 204, 240, 36, 246, 55, 238, 191, 152, 47, 175, 124, 158, 56, 254, 146, 255, 94, 255, 95, 0, 254, 129, 250, 139, 123, 103, 240, 54, 0, 254, 251, 2, 255, 200, 254, 207, 15, 129, 14, 255, 64, 252, 219, 244, 215, 127, 255, 95, 190, 244, 207, 255, 225, 64, 4, 63, 255, 208, 254, 65, 189, 129, 145, 175, 128, 255, 200, 254, 22, 191, 254, 250, 0, 208, 27, 254, 126]

turbokongen, 2021-08-08:
From the log you provided:
The start of the packages is missing (no start flag 126). 
It only receives partial data. 
The data received does not make sense at all. There is no readable data in there. 
The package size differs from reception to reception.

I went through the logs from when we added the swedish aidon support. What I am wondering is: what changed since then?

Bo1jo, 2021-08-08:
What changed is the adapter I'm using. I can still get it working with my old adapter (even though i requires numerous reboots), so nothing has changed with the aidon meter. I guess I have to investigate why the new adapter only receives partial data.

Bo1jo, 2021-08-23:
The later versions of the code seems to be a part of the problem. When using version 1.5 with setting "auto" i get:

2021-08-23 20:55:31 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-08-23 20:55:31 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-23 20:55:31 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-08-23 20:55:32 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-23 20:55:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-08-23 20:55:32 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-08-23 20:55:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-08-23 20:55:32 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-08-23 20:55:32 WARNING (Thread-3) [custom_components.ams] No parser detected

and with setting "aidon_se":
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 23, 1, 19, 51, 40, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 7, 119, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 1, 240, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 23, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 10, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 21, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 44, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 21, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 37, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 2, 27, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 110, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 0, 224, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 103, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 1, 168, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 12, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 178, 139, 43, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 131, 10, 235, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 46, 28, 198, 2, 2, 15, 0, 22, 32, 53, 167, 126]
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] list_type is 27
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] meter_date_time, OBIS:[0, 0, 1, 0, 0, 255], Index:(24, 30), Type:9 Double OBIS
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] meter_date_time, [0, 0, 1, 0, 0, 255], 2021, 8, 23, Monday, 19, 51, 40, 2021-8-23-19-51-51-40
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] ams_active_power_import, OBIS:[1, 0, 1, 7, 0, 255], Index:(48, 54), Type:6 Double OBIS
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 1911, Index:(55, 59)
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] ams_active_power_export, OBIS:[1, 0, 2, 7, 0, 255], Index:(69, 75), Type:6 Double OBIS
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 0, Index:(76, 80)
2021-08-23 20:51:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] ams_reactive_power_import, OBIS:[1, 0, 3, 7, 0, 255], Index:(90, 96), Type:6 Double OBIS

But with version 1.3.0:
2021-08-23 21:01:22 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-08-23 21:01:22 INFO (Thread-3) [custom_components.ams] Detected Swedish Aidon meter by list type
2021-08-23 21:01:32 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 23, 1, 20, 1, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 4, 165, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 67, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 22, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 10, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 22, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 31, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 7, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 19, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 2, 12, 2, 2,  [... cut ...] 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 178, 139, 237, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 131, 10, 236, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 46, 28, 219, 2, 2, 15, 0, 22, 32, 88, 106, 126]
2021-08-23 21:01:32 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] list_type is 27

So something changed in the later versions, which doesent work with aidon_se.

turbokongen, 2021-08-23:
Oh! I bumped pyserial from 3.4 to 3.5. Could you try to revert the version.
Edit manifest.json
[image: image.png]

turbokongen, 2021-08-23:
Also try this branch: https://github.com/turbokongen/hass-AMS/tree/baudrate
I see that maybe the swedish meters transfer at a different speed (115200 baud) than the Norwegian ones (2400baud).

Bo1jo, 2021-08-23:
Reverting to 3.4 has no effect:
2021-08-23 21:34:22 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid frame CRC check
2021-08-23 21:34:22 DEBUG (Thread-3) [custom_components.ams] failed package: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 23, 1, 20, 34, 20, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 207, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 17, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 7, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 24, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 53, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 60, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 43, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 1, 108, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 175, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 0, 131, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 115, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 1, 222, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 39, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 178, 142, 126]
2021-08-23 21:34:22 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-23 21:34:22 DEBUG (Thread-3) [custom_components.ams] failed package: [2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 131, 11, 14, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 46, 29, 70, 2, 2, 15, 0, 22, 32, 19, 46, 126]
2021-08-23 21:34:32 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid frame CRC check
2021-08-23 21:34:32 DEBUG (Thread-3) [custom_components.ams] failed package: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 23, 1, 20, 34, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 207, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 17, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 7, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 24, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 56, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 60, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 45, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 1, 108, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 177, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 0, 131, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 115, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 1, 220, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 40, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 178, 142, 126]
2021-08-23 21:34:32 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-23 21:34:32 DEBUG (Thread-3) [custom_components.ams] failed package: [2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 131, 11, 14, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 46, 29, 70, 2, 2, 15, 0, 22, 32, 234, 58, 126]
2021-08-23 21:34:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid frame CRC check
2021-08-23 21:34:42 DEBUG (Thread-3) [custom_components.ams] failed package: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 23, 1, 20, 34, 40, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 207, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 9, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 17, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 7, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255,  [... cut ...]
    _setup(hass, config[const.DOMAIN])
  File "/config/custom_components/ams/__init__.py", line 44, in _setup
    hub = AmsHub(hass, config)
  File "/config/custom_components/ams/__init__.py", line 97, in __init__
    self._ser = serial.Serial(
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 219, in __init__
    self._exclusive = None
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 289, in baudrate
    """
ValueError: Not a valid baudrate: None

turbokongen, 2021-08-23:
Did you make sure that 3.4 is the one Installed? Homeassistant components also bumped to 3.5 so if you use any serial components those needs to be reversed as well. Solution to the baudrate error is to remove and readd the ams integration. It fails because of missing config entry.

Bo1jo, 2021-08-28:
**I've tried the baudrate branch on a separate home assistant installation, without any other integrations than AMS, but i still get the same error:**
2021-08-28 21:01:21 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-08-28 21:01:21 ERROR (MainThread) [homeassistant.setup] Error during setup of component ams
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 287, in baudrate
    b = int(baudrate)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 249, in _async_setup_component
    result = await task
  File "/config/custom_components/ams/__init__.py", line 53, in async_setup
    _setup(hass, config[const.DOMAIN])
  File "/config/custom_components/ams/__init__.py", line 44, in _setup
    hub = AmsHub(hass, config)
  File "/config/custom_components/ams/__init__.py", line 97, in __init__
    self._ser = serial.Serial(
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 219, in __init__
    self.baudrate = baudrate
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 289, in baudrate
    raise ValueError("Not a valid baudrate: {!r}".format(baudrate))
ValueError: Not a valid baudrate: None

**I've changed to 3.4 in manifest.json, but I'm don't know how to check which version is used.**

**I've got it working with 1.4.3 (which also seems to be 3.5). When it works everything seems fine:**

2021-08-28 21:31:28 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-08-28 21:31:34 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-08-28 21:31:40 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-28 21:31:40 DEBUG (Thread-3) [custom_components.ams] failed package: [64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 28, 6, 20, 29, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 4, 47, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 25, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 18, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 8, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 24, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 48, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 22, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 43, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 1, 153, 2, 2, 0, 126]
2021-08-28 21:31:41 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-28 21:31:41 DEBUG (Thread-3) [custom_components.ams] failed package: [162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 229, 8, 28, 6, 20, 31, 40, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 4, 54, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 23, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 18, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 8, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 24, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 44, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 18, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 40, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 1, 159, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 126]
2021-08-28 21:31:42 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid header CRC check
2021-08-28 21:31:42 DEBUG (Thread-3) [custom_components.ams] failed package: [2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 0, 152, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 0, 125, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 1, 252, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 1, 12, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 1, 180, 147, 203, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 131, 70, 39, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 46, 145, 0, 2, 2, 15, 0, 22, 32, 154, 227, 126]
2021-08-28 21:31:52 DEBUG (Thread-3) [custom_components.ams.parsers.aidon_se] Invalid frame CRC check
[... cut ...]
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 166, in connect
    self.sensor_data, _ = parser.parse_data(self.sensor_data,
  File "/config/custom_components/ams/parsers/aidon_se.py", line 103, in parse_data
    sensor_data[key][const.SENSOR_ATTR][
KeyError: 'ams_active_energy_import'

**With 1.3.0 eveything works fine.**

turbokongen, 2021-08-29:
Please try the latest of baudrate branch.
That last error is new, and is fixed in that branch.
You need to delete the integration from home-assistant integrrations page, restart home-assistant and readd the integration from the integrations page.
To check what version is in place while running, list the pip packages (`pip list | grep pyserial`)  from the home-assistant venv while homeassistant is up and running.

Bo1jo, 2021-08-29:
Thanks! 

I've checked my pyserial version, it is 3.5.

I've tried the latest baudrate branch. When setting up from the home assistant integrations page and setting the baudrate to 115200 I get:

_2021-08-29 14:46:56 DEBUG (MainThread) [custom_components.ams.config_flow] ['/dev/ttyS1: ttyS1', '/dev/ttyS0: ttyS0', '/dev/ttyUSB1: USB-Serial Controller', '/dev/ttyUSB0: USB Serial']
2021-08-29 14:47:22 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
2021-08-29 14:47:22 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2021-08-29 14:47:26 DEBUG (MainThread) [custom_components.ams] Finish init of AMS_

then nothing more.

If i try to setup in config.yaml with:
ams:
  serial_port: '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0'
  parity: 'N'
  meter_manufacturer: 'aidon_se'
  baudrate: 115200

I can't restart:
_Failed to call service homeassistant/restart. The system cannot restart because the configuration is not valid: Invalid config for [ams]: [baudrate] is an invalid option for [ams]. Check: ams->ams->baudrate. (See /config/configuration.yaml, line 14)._

I also tried just:
ams:
  serial_port: '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0'
  parity: 'N'
  meter_manufacturer: 'aidon_se'

then i get this error:
_2021-08-29 21:36:39 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-__Prolific_Technology_Inc._USB-Serial_Controller-if00-port0_
_2021-08-29 21:36:39 ERROR (MainThread) [homeassistant.setup] Error during setup of component ams
Traceback (most recent call last):_
  _File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 291, in baudrate
    b = int(baudrate)_
_TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'_

_During handling of the above exception, another exception occurred:_

_Traceback (most recent call last):_
  _File "/usr/src/homeassistant/homeassistant/setup.py", line 249, in _async_setup_component
    result = await task_
  _File "/config/custom_components/ams/__init__.py", line 53, in async_setup
    _setup(hass, config[const.DOMAIN])_
  _File "/config/custom_components/ams/__init__.py", line 44, in _setup
    hub = AmsHub(hass, config)_
  _File "/config/custom_components/ams/__init__.py", line 97, in __init__
    self._ser = serial.Serial(_
  _File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 223, in __init__
    self.baudrate = baudrate_
  _File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 293, in baudrate
    raise ValueError("Not a valid baudrate: {!r}".format(baudrate))_
_ValueError: Not a valid baudrate: None_

turbokongen, 2021-08-30:
I have added the missing code to enable YAML config, please try again. 🙏

Bo1jo, 2021-08-30:
Thanks! I've tried the new code. It doesn't matter if I add the integration under the integrations page in home assistant or in the yaml. The only thing I see in the log is:

2021-08-30 17:40:34 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-08-30 17:40:34 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-08-30 17:40:34 DEBUG (MainThread) [custom_components.ams] Finish init of AMS

So there are never any data read and no sensors added,

turbokongen, 2021-08-30:
I added some verbose debugging lines to try to figure out where it stops. Could you please try?
Logs might get big 😰

Bo1jo, 2021-08-30:
This is what I get in the log (attaching a long version of the log)
[home-assistant (2).log](https://github.com/turbokongen/hass-AMS/files/7079245/home-assistant.2.log)
:

2021-08-30 21:48:21 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] byte_counter and bytelist set
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer
2021-08-30 21:48:27 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buf=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buffer read=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] Adding to byte counter
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buf=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buffer read=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] Adding to byte counter
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buf=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buffer read=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] Adding to byte counter
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buf=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buffer read=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] Adding to byte counter
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buf=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] buffer read=b'\x00'
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] Adding to byte counter
2021-08-30 21:48:27 DEBUG (Thread-3) [custom_components.ams] self._running and starting to read buffer

turbokongen, 2021-08-31:
Thanks. Looking at the debug, there is only NULL bytes coming from the usb. 
I added a serial port setup data to the debug. Could you try that? It will show in the log how the port is set up.
I have created a similar one for the 1.3 version. https://github.com/turbokongen/hass-AMS/tree/1.3test

Bo1jo, 2021-09-01:
Thanks! I've tried the 1.3test and the baudrate and have attached the logs. I have two different mbus-readers and tried them both, to see if it makes any difference. Hence the two logs.

[home-assistant 1_3test.log](https://github.com/turbokongen/hass-AMS/files/7093654/home-assistant.1_3test.log)
[home-assistant baudrate.log](https://github.com/turbokongen/hass-AMS/files/7093658/home-assistant.baudrate.log)
[home-assistant baudrate_2nd.log](https://github.com/turbokongen/hass-AMS/files/7093660/home-assistant.baudrate_2nd.log)

turbokongen, 2021-09-01:
Thank you for your feedback.
To say the least, this is frustrating. 
Here is what I have found so far:(Just notes)
1: The data received into the buffer is corrupt or faulty on any version above 1.3
2: Looking throught the changes, between 1.3 and above, nothing has changed in the code leading up to the "stop" point (This is inside the read_data() method before sending anything to be crc checked or parsed.) except:
  - pyserial was bumped from 3.4 to 3.5
  - a few typing hints of some methods were removed
  - constants were defined in .const instead of in each file
 I edit the 1.3test branch and bumped pyserial to 3.5. Could you try that and see if that still works?
Then if that works ok, I will add in one change by one until we break it.
I cannot do this locally because I have a norwegian kamstrup meter running both a RPI and a Ubuntu 20.04LTS machine.
Thank you for your patience.

Bo1jo, 2021-09-01:
I tried the latest 1.3test branch. Works fine (see attached log)
I get the frustration! Thank YOU for your patience,

[home-assistant.log](https://github.com/turbokongen/hass-AMS/files/7093905/home-assistant.log)

turbokongen, 2021-09-02:
I've made seperate branches for changes and numbered them 1.3test 1,2,3,4
Please try one by one and see if that break anything. 
Next step is the constants conversion, but I hold on with that until these changes are tested. :)

Bo1jo, 2021-09-02:
Thanks All 4 tests works perfect (logs attached).
[home-assistant 1_3test-1.log](https://github.com/turbokongen/hass-AMS/files/7101315/home-assistant.1_3test-1.log)
[home-assistant 1_3test-2.log](https://github.com/turbokongen/hass-AMS/files/7101316/home-assistant.1_3test-2.log)
[home-assistant 1_3test-3.log](https://github.com/turbokongen/hass-AMS/files/7101318/home-assistant.1_3test-3.log)
[home-assistant 1_3test-4.log](https://github.com/turbokongen/hass-AMS/files/7101320/home-assistant.1_3test-4.log)

turbokongen, 2021-09-02:
Okidoki. So I reversed the constant centralisation I did, which should be the only thing left now. Please try the baudrate branch. 🙏

Bo1jo, 2021-09-05:
Ok, so now I tried the new baudrate branch. When I'm setting baudrate to 115200 it doesn't work (baudrate.log), but if i do a manual configuration in yaml and omit baudrate it works (no baudrate set.log).

config.yaml:

ams:
  serial_port: '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0' # Required. The serial port used to communicate through
  parity: 'N'  # Optional, defaults to 'N'
  meter_manufacturer: 'aidon_se' # Optional, defaults to 'auto' 'aidon_se'

[home-assistant baudrate.log](https://github.com/turbokongen/hass-AMS/files/7111895/home-assistant.baudrate.log)
[home-assistant no baudrate set.log](https://github.com/turbokongen/hass-AMS/files/7111896/home-assistant.no.baudrate.set.log)

Bo1jo, 2021-09-05:
Tried to change the meter setting to "auto" and then it doesn't work (se attached log).
[home-assistant auto.log](https://github.com/turbokongen/hass-AMS/files/7111917/home-assistant.auto.log)

ams:
serial_port: '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0' # Required. The serial port used to communicate through
parity: 'N' # Optional, defaults to 'N'
meter_manufacturer: 'auto' # Optional, defaults to 'auto' 'aidon_se'

turbokongen, 2021-09-05:
That is good news!
If you omit the baudrate, it will default to 2400.  Then the 115200 info from HANporten.se is wrong.
I found a typo in the autodetect for aidon_se. Should work now. Please try the latest in baudrate branch.

Bo1jo, 2021-09-05:
Tried the latest baudrate branch, and it's working straight away!

[home-assistant (2).log](https://github.com/turbokongen/hass-AMS/files/7112097/home-assistant.2.log)

turbokongen, 2021-09-05:
Nice! I will release the latest code. Thanks for debugging 💯 🥇

closed 2021-09-05 by turbokongen