turbokongen/hass-AMS #45: Aidon 6534 - All sensors unavailable with 1.4.0-beta

mariwing, 2021-04-03:
Hi,

I have Aidon 6534 and with the 1.4.0-beta all of the sensors shows unavailable. Rolling back to 1.3 it works as before.

This is the debug log when running 1.4.0-beta:

2021-04-03 11:12:06 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
2021-04-03 11:12:06 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-04-03 11:12:06 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams] data read from port=[126, 161, 30, 65, 8, 131, 19, 238, 238, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 13, 2, 2, 9, 6, 1, 1, 0, 2, 129, 255, 10, 11, 65, 73, 68, 79, 78, 95, 86, 48, 48, 48, 49, 2, 2, 9, 6, 0, 0, 96, 1, 0, 255, 10, 16, 55, 51, 53, 57, 57, 57, 50, 56, 57, 53, 57, 49, 51, 49, 57, 53, 2, 2, 9, 6, 0, 0, 96, 1, 7, 255, 10, 4, 54, 53, 51, 52, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 27, 30, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 2, 118, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 193, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 23, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 81, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 39, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 79, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 52, 2, 2, 15, 255, 22, 35, 27, 183, 126]
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] list_type is 13
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] obis_list_version, OBIS:[1, 1, 0, 2, 129, 255], Index:(24, 30), Type:10 Single OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value Single OBIS type 10: AIDON_V0001, Index:(32, 43)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] meter_serial, OBIS:[0, 0, 96, 1, 0, 255], Index:(47, 53), Type:10 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type 10: 7359992895913195, Index:(55, 71)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] meter_type, OBIS:[0, 0, 96, 1, 7, 255], Index:(75, 81), Type:10 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type 10: 6534 3-phase Meter with CB and Neutral Current Measurement, Index:(83, 87)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_active_power_import, OBIS:[1, 0, 1, 7, 0, 255], Index:(91, 97), Type:6 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  6: 6942, Index:(98, 102)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_active_power_export, OBIS:[1, 0, 2, 7, 0, 255], Index:(112, 118), Type:6 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  6: 0, Index:(119, 123)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_reactive_power_import, OBIS:[1, 0, 3, 7, 0, 255], Index:(133, 139), Type:6 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  6: 0, Index:(140, 144)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_reactive_power_export, OBIS:[1, 0, 4, 7, 0, 255], Index:(154, 160), Type:6 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  6: 630, Index:(161, 165)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_current_l1, OBIS:[1, 0, 31, 7, 0, 255], Index:(175, 181), Type:16 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 19.3, Index:(182, 184)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_current_l2, OBIS:[1, 0, 51, 7, 0, 255], Index:(194, 200), Type:16 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 2.3, Index:(201, 203)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_current_l3, OBIS:[1, 0, 71, 7, 0, 255], Index:(213, 219), Type:16 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 8.1, Index:(220, 222)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_voltage_l1, OBIS:[1, 0, 32, 7, 0, 255], Index:(232, 238), Type:18 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 234.3, Index:(239, 241)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_voltage_l2, OBIS:[1, 0, 52, 7, 0, 255], Index:(251, 257), Type:18 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 238.3, Index:(258, 260)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_voltage_l3, OBIS:[1, 0, 72, 7, 0, 255], Index:(270, 276), Type:18 Double OBIS
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  16/18: 235.6, Index:(277, 279)
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams] Got 10 new devices from the serial
2021-04-03 11:12:10 DEBUG (Thread-5) [custom_components.ams] DUMP {'ams_active_power_import': {'state': 6942, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.1.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}, 'ams_active_power_export': {'state': 0, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.2.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_import': {'state': 0, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.3.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_export': {'state': 630, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.4.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_current_l1': {'state': 19.3, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.31.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l2': {'state': 2.3, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.51.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l3': {'state': 8.1, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.71.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_voltage_l1': {'state': 234.3, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.32.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l2': {'state': 238.3, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.52.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l3': {'state': 235.6, 'attributes': {'meter_manufacturer': 'AIDON_V0001', 'meter_type': '6534 3-phase Meter with CB and Neutral Current Measurement', 'obis_code': '1.0.72.7.0.255', 'meter_serial': '7359992895913195', 'unit_of_measurement': 'V', 'icon': ' [... cut ...]
2021-04-03 11:12:11 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] list_type is 1
2021-04-03 11:12:11 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_active_power_import, OBIS:[1, 0, 1, 7, 0, 255], Index:(24, 30), Type:6 Double OBIS
2021-04-03 11:12:11 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] Value double OBIS type  6: 6640, Index:(31, 35)
2021-04-03 11:12:11 DEBUG (Thread-5) [custom_components.ams.parsers.aidon] ams_active_power_import, OBIS:[1, 0, 1, 7, 0, 255], Index:(24, 30), Type:6 Double OBIS

turbokongen, 2021-04-07:
Sorry for late reply.
The logs looks ok. Maybe the integration has create a new set of sensors. Check the entity list in developer tools for xxx_2 sensors. Also check the integrations page for two integration entries/double set of sensors.
In that case, delete the integration, restart hass and reinstall.

mariwing, 2021-04-07:
No need to apologize! I deleted the integration, updated again til 1.4.0-beta (i was running 1.3.0 that I installed manually), restarted Home Assistant and added the integration back again. Everything showed up, using the old entities names so everything is good.

Just a note; I have tried updating from 1.3.0 to 1.4.0-beta at two occasions, both with the same result (no entities available). So the key for me seems to have been removing the integration before updating.

turbokongen, 2021-04-07:
Perfect. I will add this to the releasenotes. Thank you for reporting. 💯

mariwing, 2021-04-07:
I am sorry, but it seems I was too quick on this. After a second rebot this happened:

2021-04-07 14:03:47 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
021-04-07 14:03:51 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2021-04-07 14:03:51 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-04-07 14:03:51 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2021-04-07 14:03:52 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-04-07 14:03:52 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-04-07 14:03:52 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-04-07 14:03:52 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-04-07 14:03:52 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-04-07 14:03:52 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-04-07 14:03:52 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-04-07 14:03:53 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-04-07 14:03:53 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-04-07 14:03:53 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-04-07 14:03:53 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-04-07 14:03:53 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-04-07 14:03:53 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-04-07 14:03:53 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-04-07 14:03:56 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-04-07 14:03:56 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-04-07 14:03:56 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-04-07 14:03:56 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-04-07 14:03:56 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-04-07 14:03:56 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-04-07 14:03:56 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-04-07 14:03:58 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-04-07 14:03:58 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 162, 67]]
2021-04-07 14:03:58 DEBUG (Thread-3) [custom_components.ams] Testing for [[126, 161, 79]]
2021-04-07 14:03:58 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 102, 109, 95]
2021-04-07 14:03:58 DEBUG (Thread-3) [custom_components.ams] Testing for [75, 97, 109, 115, 116, 114, 117, 112, 95]
2021-04-07 14:03:58 WARNING (Thread-3) [custom_components.ams] No parser detected
2021-04-07 14:03:58 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-04-07 14:04:03 DEBUG (Thread-3) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2021-04-07 14:04:03 INFO (Thread-3) [custom_components.ams] Detected Aidon meter
2021-04-07 14:04:03 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 17, 203, 2, 2, 15, 0, 22, 27, 240, 202, 126]
2021-04-07 14:04:03 DEBUG (Thread-3) [custom_components.ams.parsers.aidon] list_type is 1
2021-04-07 14:04:06 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 17, 205, 2, 2, 15, 0, 22, 27, 61, 146, 126]
2021-04-07 14:04:06 DEBUG (Thread-3) [custom_components.ams.parsers.aidon] list_type is 1
2021-04-07 14:04:08 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 160, 42, 65, 8, 131, 19, 4, 19, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 1, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 17, 220, 2, 2, 15, 0, 22, 27, 33, 184, 126]
2021-04-07 14:04:08 DEBUG (Thread-3) [custom_components.ams.parsers.aidon] list_type is 1
2021-04-07 14:04:12 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 161, 30, 65, 8, 131, 19, 238, 238, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 13, 2, 2, 9, 6, 1, 1, 0, 2, 129, 255, 10, 11, 65, 73, 68, 79, 78, 95, 86, 48, 48, 48, 49, 2, 2, 9, 6, 0, 0, 96, 1, 0, 255, 10, 16, 55, 51, 53, 57, 57, 57, 50, 56, 57, 53, 57, 49, 51, 49, 57, 53, 2, 2, 9, 6, 0, 0, 96, 1, 7, 255, 10, 4, 54, 53, 51, 52, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 17, 220, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 3, 151, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 31, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 86, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 83, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 50, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 44, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 54, 2, 2, 15, 255, 22, 35, 93, 30, 126]
[... cut ...]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 165, in connect
    self.sensor_data, _ = parser.parse_data(self.sensor_data,
  File "/config/custom_components/ams/parsers/aidon.py", line 213, in parse_data
    const.HAN_METER_MANUFACTURER: han_data[
KeyError: 'obis_list_version'

turbokongen, 2021-04-07:
I think I found the issue.
Please try the real_parser branch and see if that fixes it.

mariwing, 2021-04-07:
Now I understand what your Github user name is all about! The fix worked, thank you for the quick reply 👍

turbokongen, 2021-04-11:
Glad to help! I put up a new beta release 🥇 
Fixed in v1.4.0-beta1

closed 2021-04-11 by turbokongen