turbokongen/hass-AMS #53: Swedish Kaifa MA304H4PSE

runlar, 2021-09-20:
Hi

I got a swedish Kaifa MA304H4PSE I try to connect to via this MBUS module:

https://www.aliexpress.com/item/32719562958.html?spm=a2g0s.9042311.0.0.dcd44c4dufrCZ2

I got err as follow:
With autodetect I got err in the log:

This error originated from a custom integration.

Logger: custom_components.ams
Source: custom_components/ams/__init__.py:236
Integration: Norwegian AMS (documentation, issues)
First occurred: 19:13:01 (26 occurrences)
Last logged: 19:19:16

No parser detected

2021-09-20 19:12:49 INFO (MainThread) [homeassistant.setup] Setting up ams
2021-09-20 19:12:49 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
2021-09-20 19:12:49 INFO (MainThread) [homeassistant.setup] Setup of domain ams took 0.0 seconds
2021-09-20 19:12:49 INFO (Thread-9) [custom_components.ams] Autodetecting meter manufacturer
2021-09-20 19:12:49 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.ams
2021-09-20 19:13:01 WARNING (Thread-9) [custom_components.ams] No parser detected
2021-09-20 19:13:01 INFO (Thread-9) [custom_components.ams] Autodetecting meter manufacturer
2021-09-20 19:13:16 WARNING (Thread-9) [custom_components.ams] No parser detected
2021-09-20 19:13:16 INFO (Thread-9) [custom_components.ams] Autodetecting meter manufacturer

And when I manual choose Kaifa I got this err in the log:

This error originated from a custom integration.

Logger: custom_components.ams
Source: custom_components/ams/__init__.py:236
Integration: Norwegian AMS (documentation, issues)
First occurred: 19:13:01 (49 occurrences)
Last logged: 19:25:01

No parser detected

This error originated from a custom integration.

Logger: root
Source: custom_components/ams/__init__.py:215
Integration: Norwegian AMS (documentation, issues)
First occurred: 19:25:03 (1 occurrences)
Last logged: 19:25:03

Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 183, in connect
    self.meter_manufacturer = self._find_parser(pkg)
  File "/config/custom_components/ams/__init__.py", line 220, in _find_parser
    if _test_meter(pkg, AIDON_METER_SEQ):
  File "/config/custom_components/ams/__init__.py", line 215, in _test_meter
    for i in range(len(pkg)):
TypeError: object of type 'NoneType' has no len()

Is it possible to get it to work with hass-AMS?

Best regards
Rune

turbokongen, 2021-09-20:
Hi Rune.
Thanks for using (or trying to use 😊) this addon.
The Kaifa meter code that exists is for the Norwegian data. I need to implement it for the swedish kaifa meters. Could you enable debug logging, and attach the logs in a txt file? Then I can examine them and make a parser for the swedish kaifa data.

runlar, 2021-09-20:
Hi

Here comes the log.

[home-assistant.log](https://github.com/turbokongen/hass-AMS/files/7198491/home-assistant.log)

turbokongen, 2021-09-21:
Hi, I have made a draft for you to test. Please install [kaifa_SE](https://github.com/turbokongen/hass-AMS/tree/kaifa_SE) branch and test.

runlar, 2021-09-21:
Hi

I have now testet kaifa_SE, tested both autodetect and manual choose Kaifa I have added the logs for you to examinate.

[Autodetect_home-assistant.log](https://github.com/turbokongen/hass-AMS/files/7205189/Autodetect_home-assistant.log)
[Manual choose Kaifa_home-assistant.log](https://github.com/turbokongen/hass-AMS/files/7205190/Manual.choose.Kaifa_home-assistant.log)

turbokongen, 2021-09-21:
Those logs are still the "old" production code, not the new kaifa_SE branch.
Maybe you need to delete the _pycache_ directories after you have downloaded the kaifa_SE branch.
Best way is to totally delete all the old files, and download the kaifa_SE branch in the same directory.

runlar, 2021-09-21:
I see
When I goto the branch kaifa_SE click Code och download zip file I get the master files. That's why it was the old code.

Maybe a stupid question, how do I download the right kaifa_Se zip file?

runlar, 2021-09-21:
Thank you I give it a try to morrow.

https://github.com/turbokongen/hass-AMS/archive/refs/heads/kaifa_SE.zip

runlar, 2021-09-22:
Hi

Have tested now, it looks very good.

What I can see is that:

"active power import" has comma fault my meter displays example 4,340kW and the reading is 434W (should be 4340W) 

"current L1, L2 and L3" is to high, could be comma fault here as well but unfortunately my meter dont view the current in the display so I cant verify it.

"reactive energy/power" cant verify the meter dont display this either.

active energy import and voltage seems to be correct.

I have added my logg.

[home-assistant.log](https://github.com/turbokongen/hass-AMS/files/7212609/home-assistant.log)

runlar, 2021-09-22:
I looked a bit more att the current readings not shure if they are to high, there seems to be som peaks in the readings. I attached at dokument with screenshots.

[Current L1 L2 L3.odt](https://github.com/turbokongen/hass-AMS/files/7212752/Current.L1.L2.L3.odt)

runlar, 2021-09-23:
Hi
Testet the new one, looks realy good.

sensor.ams_active_energy_export_: Can't read from the display on my meter.
sensor.ams_active_energy_import_: The reading is correct.
sensor.ams_reactive_energy_export_: Can't read from the display on my meter.
sensor.ams_reactive_energy_import_: Can't read from the display on my meter.

sensor.ams_active_power_import_: The comma is correct now.

I need to figure a way to verify the 3 energy sensors, probably I have to call my grid supplier to get it verified. I will let you know asap.

turbokongen, 2021-09-23:
No need to verify the others. They are grouped with the same unit.
All energy sensors, voltage sensors, current sensors and power sensors.
I will make a release tomorrow.
Thank you for testing this!

Sendt fra min Commodore 64

runlar, 2021-09-23:
I don't think I see these sensors:
Cumulative hourly active import energy (A+) (Q1+Q4) kWh
Cumulative hourly active export energy (A-) (Q2+Q3) kWh
Cumulative hourly reactive import energy (R+) (Q1+Q2) kVArh
Cumulative hourly reactive export energy (R-) (Q3+Q4) kVArh

Was it them you meat when you asked about "The hourly meter reading (energy sensors) is that correct as well?" ?

turbokongen, 2021-09-23:
What sensors are shown. There should be 14 in total for a 3phase meter and
10 for a 1phase meter.

Sendt fra min Commodore 64

runlar, 2021-09-23:
I got 14, in the attached document you se them.

[AMS.odt](https://github.com/turbokongen/hass-AMS/files/7220267/AMS.odt)

turbokongen, 2021-09-23:
Perfect. Then all data is decoded 😊
The sensors for hourly (cummulative) is the ones named
..........._energy_......
Sendt fra min Commodore 64

runlar, 2021-09-23:
Thank you for the work you put in this projekt, great job.
Best regards.

turbokongen, 2021-09-24:
Fixed by #54

closed 2021-09-24 by turbokongen