turbokongen/hass-AMS #55: Tryed to add sensors under the new energy dashboard

dr-eamy, 2021-09-25:
Hi i installed the ams integrations and now i can se the values from the kamstrup ams hanport, but how do i get them into the new energy dash ?

the only to sensor I can add is : ams_active_energy_export_(serialnumber) and ams_active_energy_import_

her is a list of all sensors your integration finds from the hanport on the kamstrup meter:

ams_active_energy_export_(serialnumber)		value = Unknown
ams_active_energy_import_(serialnumber)		value = 100,051.49 kWh
ams_active_power_export_(serialnumber)		Value = 0 W
ams_active_power_import_(serialnumber)		Value = 3,207 W
ams_current_l1_(serialnumber)				Value = 13.75 A
ams_reactive_energy_export_(serialnumber)	Value = 2,525.68 kVAr
ams_reactive_energy_import_(serialnumber)	Value = 8,797.17 kVAr
ams_reactive_power_export_(serialnumber)	Value = 490 VAr
ams_reactive_power_import_(serialnumber)	Value = 0 VAr
ams_voltage_l1_(serialnumber)				Value = 241 V

I'v added all the grafs from the diferent sensors in image under
[image]

regards
Eivind

turbokongen, 2021-09-25:
Hi. HA energy dashboard only supports sensors with kWh unit. (Energy usage) So ams_active_energy_import_ (Your paid consumption) is the grid consumption one and ams_active_energy_export_ (Your produced energy that you sell back to the energy company.) is the one for return to grid.

dr-eamy, 2021-09-25:
Ok, then its me that has misunderstand :),
 else its a nice work  you have don whit the integration, love the possibility to watch the voltage, as it is fluctuating at my location. and the rapid update of consumption of Kw we do at our home!
 keep up the good work
regards
E

closed 2021-09-25 by dr-eamy