turbokongen/hass-AMS #63: svm elster f2  and Kamstrup Multical 302

hesselb, 2021-11-13:
I have 3 elstar  f2 meters and 1 kamstrup mltical 302
i used it befoor whit a raspberry pi and domotics
is it posible to get this working in homassist?

hesselb, 2021-11-13:
the php scripts i used before are availible

turbokongen, 2021-11-13:
I am pretty sure it can. As long as it communicates over serial. So, I would need some raw package and of course those php scripts will be of help as well. 😊

hesselb, 2021-11-13:
[kamstup4 php.txt](https://github.com/turbokongen/hass-AMS/files/7532210/kamstup4.php.txt
[ww.php.txt](https://github.com/turbokongen/hass-AMS/files/7532211/ww.php.txt)
[cv.php.txt](https://github.com/turbokongen/hass-AMS/files/7532212/cv.php.txt)
)
[bron.php.txt](https://github.com/turbokongen/hass-AMS/files/7532213/bron.php.txt)

hesselb, 2021-11-13:
kan i pul the raw package whit homassist?
els i need to hook mi raspberrypi on

turbokongen, 2021-11-13:
Just set up the AMS reader with the same settings as you had on the rpi.
Select auto for meter_manufacturer. Enable debug logging, and the logs will
give you the raw package.

Sendt fra min Commodore 64

hesselb, 2021-11-13:
this is wat i found :2021-11-13 16:10:24 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using port /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AC01XF1P-if00-port0
2021-11-13 16:10:24 INFO (Thread-3) [custom_components.ams] Autodetecting meter manufacturer
2021-11-13 16:10:24 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
 whit the raspberry this
[elstar2.txt](https://github.com/turbokongen/hass-AMS/files/7532302/elstar2.txt)

[kamstrup4.txt](https://github.com/turbokongen/hass-AMS/files/7532303/kamstrup4.txt)

turbokongen, 2021-11-13:
Ok, so from what I read from the code, and the logs you sent, there is first sent a request to get data, and then data is received.(This is polling)
The AMS meters work differently: they push data to Mbus at fixed intervals.
What you need sound like this: https://github.com/golles/ha-kamstrup_403

hesselb, 2021-11-14:
thank you for you time

closed 2021-11-14 by hesselb