turbokongen/hass-AMS #66: Alternative methods of reading data

yurgh, 2021-11-23:
Not an issue, but a suggestion:

I've modified my Tibber Pulse to send data to my MQTT-server (explained here: https://github.com/iotux/ElWiz#oppsett-av-pulse ). Unfortunately that project doesn't support my Kamstrup Direct 1-phase meter.
However; with my limited Python skills I've been unable to modify hass-AMS to use the binary data from the MQTT server instead of reading from serial. 

So; any plans on adding alternative input methods your component?
I vote for MQTT-support, but could be anything that delivers binary data! :-)

sandos, 2021-11-23:
This method should mayb work for this?

https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_1-install-ser2net

Hmm, ok, it does look like zigbee2mqtt has some specific support for a tcp-protocol for the serial port, so likely some work to get it to work.

Hellowlol, 2021-11-23:
Tried using https://github.com/toreamun/amshan-homeassistant

yurgh, 2021-11-24:
Thanks for suggestions!

I was able to hack the parserto match what my meter is sending. The meter suddenly changed the format, thus all my sensors just reported bogus values. Seems to have started sending more bytes, because the offset for the values was changed by four. I guess thats what you get for hardcoding offsets instead of looking for the OBIS-codes and their corresponding data... :P