turbokongen/hass-AMS #75: Feature request: Use serial.serial_for_url() instead of serial.Serial()

ruant, 2022-02-01:
I got a ESP32 set up with [StreamServer](https://github.com/oxan/esphome-stream-server) to expose the data from the UART connection to the powermeter over my network.

But unfortunately this hass-AMS only supports local ports..

That's a "easy fix" tho, just swap `serial.Serial()` with `serial.serial_for_url()`
`serial_for_url` has supports for it all.
The part where I got stuck was with the home assistant scheme UI setup stuff..

Another thing I've noticed, since this is over the network, the connection might drop out.
Haven't thought about how to do reconnections in the code, I solved it with a node-red flow that reloaded the component when no data was received for x amount of seconds.

I'll be happy to help out with this more.

Refs:
https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.serial_for_url
https://pyserial.readthedocs.io/en/latest/url_handlers.html#urls

turbokongen, 2022-02-07:
Seems to be doable. What settings are required to pass into serial_for_url() for your usecase. 
There needs to be a change in the config_flow as well.
When those things are in place we can look at reconnectiong logic.
I made the serial_for_url branch https://github.com/turbokongen/hass-AMS/tree/serial_for_url

turbokongen, 2022-02-24:
@ruant can you test the new code of testbranch?

ruant, 2022-02-27:
@turbokongen Sorry for the total lack of response here!
I'm pulling it down and taking it for a spin now 👍

ruant, 2022-02-27:
@turbokongen 
It's receiving data from the meter fine with these changes for me.

As I wrote in my initial post, I've had issues with the connection failing at random. Everything from 5 mins to 2-3 weeks.
Since it's a network connection, there might be some kinda reconnection logic put in some where.
I've turned on verbose logging for this component, so hopefully if it happens again I'll catch some useful stuff.

ruant, 2022-03-11:
@turbokongen A small updated from me.
It's still going strong. No connection dropouts or anything like that.
Tho I've done a lot in HA lately so I've restarted quite a few times a day, so that might have helped it keep things up and running.

I'll try my best to keep my fingeres away from HA the next 10 days.
But if it's still hasn't produced any errors, I'll see what happens when I disconnect the ESP32 which provides the data, and see if the connection comes back up when the ESP32 is back up again. Just to see if there should be some kinda specific reconnection logic somewhere, or if no valid data has been received the last X amount of seconds it should do a complete reconnect.

yada75, 2022-06-26:
Is this serial over TCP or M-bus TCP like [https://github.com/rscada/libmbus/tree/master/mbus](https://github.com/rscada/libmbus/tree/master/mbus)

ruant, 2022-06-26:
@yada75 Serial over TCP.  
I use a m-bus to serial adapter, that I've connected to the ESP32. (https://www.aliexpress.com/item/1005003771951731.html)
Then I use the StreamServer to expose a TCP server this AMS integration can connect to over WiFi.

ruant, 2022-06-26:
@turbokongen 
LGTM.
Haven't experienced any issues lately, so if directly connected serial also works maybe it should be merged in.

PR #76 (2022-07-10): Serial for url
Add support for TCP/IP
Fixes #75
closed 2022-07-10 by turbokongen