turbokongen/hass-AMS #77: [custom_components.ams] Timeout waiting for end of packet. Flush  current packet

eadrife, 2022-03-07:
Hello,

suddenly my reader stopped working and I dont know exactly why. 
I managed to get it working after some help here:

https://community.home-assistant.io/t/norwegian-han-sensors/147283/53

But after switching off my RBP3+ to change power supply, it sopped updating counters.

My config is like this:

>   serial_port: '/dev/ttyUSB0' # Required. The serial port used to communicate through
>   baudrate: 2400 # Optional, defaults to '2400'
>   parity: 'N'  # Optional, defaults to 'N'
>   meter_manufacturer: 'aidon_se' # Optional, defaults to 'auto'

I have an aidon 6534.

After enabling loging I see thins kind of errors:

> 2022-03-07 22:07:01 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=149, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 7, 1, 22, 7, 0, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 231, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 1, 192, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 29, 2, 2, 15, 255, 22, 33, 2, 255]
2022-03-07 22:07:11 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=149, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 7, 1, 22, 7, 10, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 230, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 1, 181, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 29, 2, 2, 15, 255, 22, 33, 210, 252]
2022-03-07 22:07:21 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=152, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 7, 1, 22, 7, 20, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 3, 173, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 183, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 29, 2, 2, 15, 255, 22, 33, 2, 131, 9, 222, 255]

Any idea what the problem could be?

Thanks

turbokongen, 2022-03-08:
Reading the output from your log, it seems that there is a pause in the datastream from the meter or the reader hangs receiving mid packet. Try increasing the timeout https://github.com/turbokongen/hass-AMS/blob/ad7e58e5773bb4336d5d1f89c479cf80011bc521/custom_components/ams/const.py#L70 and see if that helps.

eadrife, 2022-03-08:
Hi, thanks a lot for your quick answer and support!

I tried with 0.2, 0.5, 1 and 2 as DEFAULT_TIMEOUT with similar result.

> 2022-03-08 21:27:33 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=148, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 8, 2, 21, 27, 30, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 2, 102, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 198, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 11, 2, 2, 15, 255, 22, 161, 227]
2022-03-08 21:27:43 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=149, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 8, 2, 21, 27, 40, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 5, 29, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 3, 59, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 11, 2, 2, 15, 255, 22, 33, 241, 251]
2022-03-08 21:27:53 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=146, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 8, 2, 21, 27, 50, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 5, 21, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 3, 54, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 11, 2, 2, 15, 89, 253]
2022-03-08 21:28:03 DEBUG (Thread-3) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=145, frame_started=True, package_size=581, DUMP: [126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 3, 8, 2, 21, 28, 0, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 2, 91, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 185, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 10, 130, 249, 255, 252]

Maybe the reader is malfunctioning?
I have a different model of reader I have never tried yet, I should give it a try....

Best regars!

eadrife, 2022-03-10:
Hello, I think we can close this issue.

Today morning I tried another MBUS reader and it works fine.
My initial setup used this reader (+ USBtoTTL converter), that worked for 2 weeks:

https://www.aliexpress.com/item/32751482255.html?spm=a2g0o.order_list.0.0.21ef1802EoA4f9)

I assume either the reader or the USBtoTTL has some problem.'

Now I am using this module that works fine:

https://www.aliexpress.com/item/32934275072.html?spm=a2g0o.order_list.0.0.21ef1802EoA4f9

Hopefully it lasts more than 2 weeks.

closed 2022-03-10 by eadrife