turbokongen/hass-AMS #78: Kaifa MA105H2E confirmed working.

Futrax, 2022-04-11:
I have tested this on my Kaifa MA105H2E, can confirm this is working, wiki can be updateted.

turbokongen, 2022-04-11:
Thanks for reporting 💯

closed 2022-04-11 by turbokongen

Futrax, 2022-04-13:
It works, but the entities for imported and exported reactive energy reads way to high.
[image]
[image]
[image]

Futrax, 2022-04-25:
Hello, I am quite new to HA, can you help me on how to activate debug logging for this?

turbokongen, 2022-05-02:
```# Example configuration.yaml entry
logger:
  default: info
  logs:
    custom_components.ams: debug

Futrax, 2022-05-05:
`2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_current_l1
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_energy_import
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_energy_export
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_import
2022-05-05 19:05:01 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_export
2022-05-05 19:05:02 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 5, 5, 4, 19, 5, 2, 255, 128, 0, 0, 2, 1, 6, 0, 0, 9, 79, 61, 126, 126]`

Is this waht you need?

turbokongen, 2022-05-06:
Yes, but the long hourly package that is received once an hour. Normally at xx:xx:05

Futrax, 2022-05-18:
This should be it then!

`022-05-05 21:00:11 DEBUG (Thread-3) [custom_components.ams] data read from port=[126, 160, 135, 1, 2, 1, 16, 158, 109, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 5, 5, 4, 21, 0, 10, 255, 128, 0, 0, 2, 14, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 56, 50, 50, 53, 57, 50, 56, 9, 8, 77, 65, 49, 48, 53, 72, 50, 69, 6, 0, 0, 6, 119, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 12, 6, 0, 0, 27, 40, 6, 0, 0, 9, 120, 9, 12, 7, 230, 5, 5, 4, 21, 0, 10, 255, 128, 0, 0, 6, 1, 134, 127, 141, 6, 0, 0, 0, 0, 6, 0, 63, 93, 46, 6, 0, 6, 95, 183, 223, 214, 126]`

Futrax, 2022-05-24:
Can comfim. Thanks a lot 😄