turbokongen/hass-AMS #80: No read out for meter, and error in Home Assistant

Futrax, 2022-09-13:
The readouts form my AMS meter has stopped worknig, and i have an error in Home Asssitant logs.
`Logger: homeassistant.config_entries
Source: custom_components/ams/__init__.py:166
Integration: AMS Reader ([documentation](https://github.com/turbokongen/hass-AMS), [issues](https://github.com/turbokongen/hass-AMS/issues))
First occurred: 18:55:40 (5 occurrences)
Last logged: 19:22:27
Error calling entry remove callback AMS Reader for ams

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 536, in async_remove
    await component.async_remove_entry(hass, self)
  File "/config/custom_components/ams/__init__.py", line 114, in async_remove_entry
    await hass.async_add_executor_job(hass.data[DOMAIN].stop_serial_read)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/ams/__init__.py", line 166, in stop_serial_read
    self._ser.close()
AttributeError: 'AmsHub' object has no attribute '_ser'`

How do I fix this?

turbokongen, 2022-09-15:
The log output you attached tells me that a close/shutdown/remove event has been triggered.
Try to remove/uninstall the componet and reinstall.

Futrax, 2022-09-16:
I have no reinstalled, and is now receiving data. But no entities from the AMS.
```
2022-09-16 22:17:10.092 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kaifa_se] list_type is 7
2022-09-16 22:17:11.699 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kaifa_se] Invalid packet size 41
2022-09-16 22:17:11.700 DEBUG (Thread-2 (connect)) [custom_components.ams] failed package: [126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 9, 16, 5, 22, 17, 12, 255, 128, 0, 0, 2, 1, 6, 0, 0, 1, 208, 1, 159, 126]
```

This is the output from the log.

[image]

husky999, 2022-09-18:
Hi,

I have a similar issue. Looks like I'm receiving packets but no entities are created. See log below.

`2022-09-18 14:55:49.060 DEBUG (MainThread) [custom_components.ams.config_flow] ['/dev/ttyUSB0: FT232R USB UART - FT232R USB UART']
2022-09-18 14:55:56.542 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
2022-09-18 14:55:56.542 DEBUG (MainThread) [custom_components.ams] config entry = {'serial_port': '/dev/ttyUSB0', 'meter_manufacturer': 'auto', 'parity': 'O', 'baudrate': 2400, 'protocol': 'Serial port'}
2022-09-18 14:55:56.542 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using serialport /dev/ttyUSB0
2022-09-18 14:55:56.558 INFO (Thread-10 (connect)) [custom_components.ams] Autodetecting meter manufacturer
2022-09-18 14:55:56.558 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2022-09-18 14:55:56.659 DEBUG (Thread-10 (connect)) [custom_components.ams] Timeout waiting for end of packet. Flush  current packet. byte_counter=1, frame_started=True, package_size=-1, DUMP: [126]
2022-09-18 14:56:02.637 DEBUG (Thread-10 (connect)) [custom_components.ams] Testing for [65, 73, 68, 79, 78, 95]
2022-09-18 14:56:02.637 DEBUG (Thread-10 (connect)) [custom_components.ams] Testing for [126, 162, 67]
2022-09-18 14:56:02.637 DEBUG (Thread-10 (connect)) [custom_components.ams] Testing for [126, 161, 79]
2022-09-18 14:56:02.637 DEBUG (Thread-10 (connect)) [custom_components.ams] Testing for [75, 102, 109, 95]
2022-09-18 14:56:02.637 DEBUG (Thread-10 (connect)) [custom_components.ams] Testing for [75, 70, 77, 95]
2022-09-18 14:56:02.638 INFO (Thread-10 (connect)) [custom_components.ams] Detected Swedish Kaifa meter
2022-09-18 14:56:02.638 DEBUG (Thread-10 (connect)) [custom_components.ams] data read from port=[126, 160, 155, 1, 0, 1, 16, 86, 27, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 9, 18, 7, 14, 56, 0, 255, 128, 0, 0, 2, 18, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 55, 51, 52, 48, 49, 53, 55, 48, 49, 49, 50, 55, 52, 53, 51, 50, 9, 8, 77, 65, 51, 48, 52, 72, 52, 68, 6, 0, 0, 6, 73, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 1, 206, 6, 0, 0, 2, 175, 6, 0, 0, 17, 27, 6, 0, 0, 9, 236, 6, 0, 0, 8, 241, 6, 0, 0, 9, 2, 6, 0, 0, 8, 252, 9, 12, 7, 230, 9, 18, 7, 14, 56, 0, 255, 128, 0, 0, 6, 8, 166, 101, 178, 6, 0, 0, 0, 0, 6, 2, 217, 43, 105, 6, 0, 34, 14, 104, 126, 48, 126]
2022-09-18 14:56:02.638 DEBUG (Thread-10 (connect)) [custom_components.ams.parsers.kaifa_se] list_type is 7
2022-09-18 14:56:17.627 DEBUG (Thread-10 (connect)) [custom_components.ams] data read from port=[126, 160, 155, 1, 0, 1, 16, 86, 27, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 9, 18, 7, 14, 56, 15, 255, 128, 0, 0, 2, 18, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 55, 51, 52, 48, 49, 53, 55, 48, 49, 49, 50, 55, 52, 53, 51, 50, 9, 8, 77, 65, 51, 48, 52, 72, 52, 68, 6, 0, 0, 6, 54, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 1, 208, 6, 0, 0, 2, 212, 6, 0, 0, 16, 217, 6, 0, 0, 9, 187, 6, 0, 0, 8, 235, 6, 0, 0, 9, 2, 6, 0, 0, 8, 251, 9, 12, 7, 230, 9, 18, 7, 14, 56, 15, 255, 128, 0, 0, 6, 8, 166, 101, 185, 6, 0, 0, 0, 0, 6, 2, 217, 43, 105, 6, 0, 34, 14, 106, 197, 201, 126]
2022-09-18 14:56:17.628 DEBUG (Thread-10 (connect)) [custom_components.ams.parsers.kaifa_se] list_type is 7
2022-09-18 14:56:32.632 DEBUG (Thread-10 (connect)) [custom_components.ams] data read from port=[126, 160, 155, 1, 0, 1, 16, 86, 27, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 9, 18, 7, 14, 56, 30, 255, 128, 0, 0, 2, 18, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 55, 51, 52, 48, 49, 53, 55, 48, 49, 49, 50, 55, 52, 53, 51, 50, 9, 8, 77, 65, 51, 48, 52, 72, 52, 68, 6, 0, 0, 6, 21, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 1, 204, 6, 0, 0, 2, 189, 6, 0, 0, 16, 105, 6, 0, 0, 9, 145, 6, 0, 0, 8, 239, 6, 0, 0, 9, 6, 6, 0, 0, 8, 246, 9, 12, 7, 230, 9, 18, 7, 14, 56, 30, 255, 128, 0, 0, 6, 8, 166, 101, 191, 6, 0, 0, 0, 0, 6, 2, 217, 43, 105, 6, 0, 34, 14, 108, 43, 173, 126]
2022-09-18 14:56:32.633 DEBUG (Thread-10 (connect)) [custom_components.ams.parsers.kaifa_se] list_type is 7
`

turbokongen, 2022-09-18:
Looks like the kaifa_se packet format has changed. I will have to make a new parser. Can any of you send logs of different list types?
The log above shows list_type 7. I would also need a package at the hour, which contains energy readings.

husky999, 2022-09-18:
Thx, that would be great!

Unfortunately, I don’t see any other list_type than 7 in my logs. Looking several hours back…

turbokongen, 2022-09-18:
Actually, try to set meter_type to kaifa instead of kaifa_se to see if it works.

Futrax, 2022-09-18:
When setting it to kaifa, the redings stop, and i get this in the log:
```
2022-09-18 19:50:48.365 DEBUG (Thread-3 (connect)) [custom_components.ams] failed package: None
2022-09-18 19:50:53.786 DEBUG (MainThread) [custom_components.ams.config_flow] ['/dev/ttyUSB0: USB-Serial Controller', '/dev/ttyACM0: ttyACM0']
2022-09-18 19:50:58.099 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.ams
```

husky999, 2022-09-18:
Seems to work for me after a restart!

Futrax, 2022-09-18:
Yes, a restart helped here to. But why does it identify the meter as kaifa_se when it should be kaifa?

turbokongen, 2022-09-18:
Well, up until now, the swedish kaifa meters sent different content in the package than the Norwegian ones. It seems that they have changed that now, and transmits the same kind of packets.

husky999, 2022-09-19:
Been running now for 12 hours and everything seems to work ok except that I don't get the "ams_active_power_import..." sensor. I have 13 entities populated but I guess I should have 14? Any pointers to what could be wrong? Any logs you need?

turbokongen, 2022-09-20:
Please try https://github.com/turbokongen/hass-AMS/tree/new_swedish_kaifa

husky999, 2022-09-20:
Thx! Now I get all 14 entities but the new active power sensor reports incorrect numbers, see attached screenshot.

[image]

turbokongen, 2022-09-21:
Ouch! I need to inspect the packages to see where the data is stored. Not the same as the norwegian ones then.
The values that are off are:
active_power_import
active_energy_import
reactive_energy_import
reactive_energy_export

rogere66, 2022-09-22:
I have the same problem as described in the initial post. It used to work, but stopped working a few weeks back, probably after a core update (or downgrade).
Now it won't work on my original install or even on a completely fresh install with the latest versions of all code - always showing the error message below.
I also tried a flash card with an old install (december 21) and this did still work, even after updating OS and core to the latest versions. Kind of strange...
HW is rpi4, 4MB and USB com port.

2022-09-22 15:56:37.577 ERROR (Thread-2 (connect)) [root] Uncaught thread exception
Traceback (most recent call last):
File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
self.run()
File "/usr/local/lib/python3.10/threading.py", line 953, in run
self._target(*self._args, **self._kwargs)
File "/config/custom_components/ams/__init__.py", line 261, in connect
data = self.read_packet()
File "/config/custom_components/ams/__init__.py", line 175, in read_packet
buf = self._ser.read()
AttributeError: 'AmsHub' object has no attribute '_ser'

turbokongen, 2022-09-25:
@husky999 please try latest code. To see if all values are good.

husky999, 2022-09-25:
@turbokongen, everything looks ok now :-) Thx!

turbokongen, 2022-09-25:
@Futrax @rogere66 do you setup the component from Yaml or through integrations page?

rogere66, 2022-09-25:
I set it up in configuration.yaml

turbokongen, 2022-09-27:
@rogere66 @Futrax  I found that I had forgotten to update the manual yaml validation at setup when I added the tcp/ip feature.
Please test the latest code at https://github.com/turbokongen/hass-AMS/tree/new_swedish_kaifa to see if it resolves the problem.

rogere66, 2022-09-27:
@turbokongen Thanks, but I can still not get it working. Use latest code of HA and AMS (1.9.4) and get invalid config when rebooting:
  Invalid config for [ams]: expected a dictionary for dictionary value @ data['ams']['protocol']. Got 'serial'. (See /config/configuration.yaml, line 24). 
                   
yaml config:
ams:
  protocol: 'serial'           # Required. The protocol type for communications. Defaults to 'serial'
  serial_port: '/dev/ttyS0'    # Required. The serial port used to communicate through
  baudrate: 2400               # Optional, defaults to '2400'
  parity: 'E'                  # Optional, defaults to 'N'
  meter_manufacturer: 'aidon'  # Optional, defaults to 'auto'

rogere66, 2022-09-30:
I have tested the 4 latest hass-AMS versions and found this:
- Version 1.9.1: works using both yaml and integrations page setup.
- Version 1.9.2: works using both yaml and integrations page setup.
- Version 1.9.3: works only using integrations page setup.
- Version 1.9.4 (test version): can not get it working at all.

A major problem is that integrations page setup don't support serial port /dev/ttyS0, thus is not useful in my setup.

It is also somewhat confusing that the few latest hass-AMS versions downloadable trough HACS all show version 1.9.1 in manifest.json.

turbokongen, 2022-10-02:
@rogere66 
Please test latest code at https://github.com/turbokongen/hass-AMS/tree/new_swedish_kaifa
Like I wrote earlier, I have forgotten to update the yaml configuration validation when I added the TCP_IP option. That is the source of the error you get. Please refer to the https://github.com/turbokongen/hass-AMS/tree/new_swedish_kaifa/README.md to see the valid yaml setup.

Regarding the serial port, I have added a [debug line](https://github.com/turbokongen/hass-AMS/blob/new_swedish_kaifa/custom_components/ams/config_flow.py#L74) to the code that you can post here. It contains the available serial ports detected by the system when setting up via the integrations page. On my system /dev/ttyS0 appears in the list of selectable ports in the integrations setup.

rogere66, 2022-10-03:
I have tested this in depth and am still not able to run the latest code reliably. The code did run on a particular 32-bit HA image a few times so I kind of assumed it was a problem related to 64-bit HA image (used so far), but further testing did not show any difference between 32-bit or 64-bit. The flash card that had the working image was re-flashed with a fresh image so I was not able to test that further.

Testing was done on a rpi 4 with 4GB RAM. I used freshly installed code image "haos_rpi4-9.0.img.xz" as well as older versions and 64-bit versions. They all failed with the this error message:

Logger: root
Source: custom_components/ams/__init__.py:281
Integration: AMS Reader (documentation, issues)
First occurred: 10:48:21 PM (1 occurrences)
Last logged: 10:48:21 PM
Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 281, in connect
    self.sensor_data, _ = parser.parse_data(
TypeError: parse_data() takes 2 positional arguments but 3 were given

Regarding the ttyS0 serial port: Since the swedish_kaifa test code wouldn't run, I copied the debug line into version 1.9.2 and got this message (a USB serial port was connected):
2022-10-03 20:45:34.509 DEBUG (MainThread) [custom_components.ams.config_flow] ['/dev/ttyUSB0: TTL232R-3V3 - TTL232R-3V3', '/dev/ttyAMA0: ttyAMA0']

No ttyS0 there and I think maybe it could be related to how I enabled the port. The port is disabled by default and I enabled it by uncomment "enable_uart=1" in the config.txt file in the hassos-boot sector before booting (using a PC). The ttyS0 port is enabled and works fine when using yaml setup, but is somehow not included in the port list.

Is there maybe a better way of enabling the serial port?

turbokongen, 2022-10-04:
I have updated the code. I do not have a meter swedish meter to test with. 
Please test the latest code.
I am thinking maybe an optional field that can take text input in the integrations page could solve the undetected serial port problem. I will have a look.

rogere66, 2022-10-04:
@turbokongen 
The code works fine now when using a norwegian Aidon meter. A text input field should be ok for selecting port.

turbokongen, 2022-10-08:
@rogere66 @Futrax @husky999 
Can you try the latest code at https://github.com/turbokongen/hass-AMS/tree/new_swedish_kaifa to see if it all works ok?
Remember to delete the integration and install again. Because of the config change.

rogere66, 2022-10-09:
@turbokongen 
Works ok now on my setup, using serial port pick list and manual input, and also yaml config.

PR #81 (2022-10-15): Update swedish kaifa meter format.
Fixes #80
Fixes #79
closed 2022-10-15 by turbokongen