turbokongen/hass-AMS #83: negative current gettting corrupt values

karka292, 2022-10-12:
Hi, I got this to work yesterday on my RBpi4B and an "Aidon 6484 SE"-meter. It works great and was simple to install and get working even for me as a dummy! But I think the parsing results in corrupt values when the currents go negative? I have solar panels on my roof and today when the solar panels produce enough power to make us energy producers, the simple gauges I implemented to visualize current (L1, L2, L3) go bananas. On the physical Aidon meter display, I get correct values. What can I do to assist?

turbokongen, 2022-10-12:
Enable debug, and attach one packet with positive values, and one with negative values. Also a screenshot of the parser values would be Nice, but not needed. Is it only currents that go crazy?

karka292, 2022-10-16:
Hej, I'm not sure I did this correct, Please assist me on how to do it correctly. I added those 4 lines to my configuration.yaml

logger:
  default: info
  logs:
    ams: debug
    custom_components.ams: debug
    custom_components.ams.sensor: debug

And got this in my home-assistant.log. ams_current_l1 is displayed wrongly on line 37 (the current other two phases were positive): 

2022-10-16 17:14:12.379 DEBUG (Thread-2 (connect)) [custom_components.ams] data read from port=[126, 162, 67, 65, 8, 131, 19, 133, 235, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1, 27, 2, 2, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 10, 16, 0, 16, 14, 10, 255, 128, 0, 255, 2, 3, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 2, 248, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 4, 16, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 31, 7, 0, 255, 16, 255, 246, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 51, 7, 0, 255, 16, 0, 23, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 24, 2, 2, 15, 255, 22, 33, 2, 3, 9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 44, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 57, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 74, 2, 2, 15, 255, 22, 35, 2, 3, 9, 6, 1, 0, 21, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 22, 7, 0, 255, 6, 0, 0, 0, 39, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 23, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 24, 7, 0, 255, 6, 0, 0, 0, 242, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 41, 7, 0, 255, 6, 0, 0, 1, 123, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 42, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 43, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 44, 7, 0, 255, 6, 0, 0, 1, 132, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 61, 7, 0, 255, 6, 0, 0, 1, 165, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 62, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, 9, 6, 1, 0, 63, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 64, 7, 0, 255, 6, 0, 0, 1, 147, 2, 2, 15, 0, 22, 29, 2, 3, 9, 6, 1, 0, 1, 8, 0, 255, 6, 2, 71, 16, 87, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 151, 1, 103, 2, 2, 15, 0, 22, 30, 2, 3, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 1, 85, 202, 2, 2, 15, 0, 22, 32, 2, 3, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 143, 201, 175, 2, 2, 15, 0, 22, 32, 106, 221, 126]
2022-10-16 17:14:12.380 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] list_type is 27
2022-10-16 17:14:12.380 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] meter_date_time, OBIS:[0, 0, 1, 0, 0, 255], Index:(24, 30), Type:9 Double OBIS
2022-10-16 17:14:12.380 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] meter_date_time, [0, 0, 1, 0, 0, 255], 2022, 10, 16, None, 16, 14, 10, 2022-10-16-16-14-14-10
2022-10-16 17:14:12.383 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_active_power_import, OBIS:[1, 0, 1, 7, 0, 255], Index:(48, 54), Type:6 Double OBIS
2022-10-16 17:14:12.383 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 760, Index:(55, 59)
2022-10-16 17:14:12.390 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_active_power_export, OBIS:[1, 0, 2, 7, 0, 255], Index:(69, 75), Type:6 Double OBIS
2022-10-16 17:14:12.390 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 0, Index:(76, 80)
2022-10-16 17:14:12.391 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_reactive_power_import, OBIS:[1, 0, 3, 7, 0, 255], Index:(90, 96), Type:6 Double OBIS
2022-10-16 17:14:12.391 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 0, Index:(97, 101)
2022-10-16 17:14:12.393 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_reactive_power_export, OBIS:[1, 0, 4, 7, 0, 255], Index:(111, 117), Type:6 Double OBIS
2022-10-16 17:14:12.393 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value double OBIS type  6: 1040, Index:(118, 122)
2022-10-16 17:14:12.394 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_active_power_import_l1, OBIS:[1, 0, 21, 7, 0, 255], Index:(246, 252), Type:6 Single OBIS
2022-10-16 17:14:12.394 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value single OBIS type 6: 0 Index:(253, 257)
2022-10-16 17:14:12.395 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_active_power_export_l1, OBIS:[1, 0, 22, 7, 0, 255], Index:(267, 273), Type:6 Single OBIS
2022-10-16 17:14:12.395 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value single OBIS type 6: 39 Index:(274, 278)
2022-10-16 17:14:12.396 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_reactive_power_import_l1, OBIS:[1, 0, 23, 7, 0, 255], Index:(288, 294), Type:6 Single OBIS
2022-10-16 17:14:12.396 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] Value single OBIS type 6: 0 Index:(295, 299)
2022-10-16 17:14:12.396 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.aidon_se] ams_reactive_power_export_l1, OBIS:[1, 0, 24, 7, 0, 255], Index:(309, 315), Type:6 Single OBIS
[... cut ...]
2022-10-16 17:14:12.427 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l3
2022-10-16 17:14:12.427 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_energy_import
2022-10-16 17:14:12.427 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_energy_export
2022-10-16 17:14:12.428 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_import
2022-10-16 17:14:12.428 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_export

turbokongen, 2022-10-17:
@karka292 Can you test the [kaifa_se_dual](https://github.com/turbokongen/hass-AMS/tree/kaifa_se_dual) branch?
Issue was that I didn't cosider signed values for current measurments.

karka292, 2022-10-19:
It looks great and corresponds well to the figures shown in the meter.
[image]

PR #86 (2022-10-19): Allow two different types of swedish kaifa meters alongside norwegiam ones, fix signed currents.
Fixes #85 
Fixes #83