turbokongen/hass-AMS #85: After upgrading to 1.9.4 integration stopped working swedish Kaifa meter.

runlar, 2022-10-15:
Hej

After updating to version 1.9.4 i stopped working. Restart did not solve the problem. Removed and reinstalled hass-ams via hacs with restart between with the same result no entitys is created.

My meter is a swedish Kaifa MA304H4PSE 

When i install version 1.9.3 again it works.

From logg file version 1.9.4:

2022-10-15 15:07:52.654 DEBUG (MainThread) [custom_components.ams] config entry = {'serial_port': '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0', 'meter_manufacturer': 'kaifa_se', 'parity': 'E', 'baudrate': 2400, 'protocol': 'serial'}
2022-10-15 15:07:52.654 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using serialport /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2022-10-15 15:07:52.711 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2022-10-15 15:08:00.750 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kaifa] Invalid packet size 287
2022-10-15 15:08:00.751 DEBUG (Thread-2 (connect)) [custom_components.ams] failed package: [126, 161, 29, 1, 0, 1, 16, 176, 174, 230, 231, 0, 15, 64, 0, 0, 0, 0, 2, 36, 9, 6, 1, 0, 0, 2, 129, 255, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 6, 0, 0, 96, 1, 0, 255, 9, 16, 55, 51, 52, 48, 49, 53, 55, 48, 51, 48, 53, 52, 56, 51, 48, 48, 9, 6, 0, 0, 96, 1, 7, 255, 9, 7, 77, 65, 51, 48, 52, 72, 52, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 1, 77, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 120, 9, 6, 1, 0, 31, 7, 0, 255, 6, 0, 0, 2, 143, 9, 6, 1, 0, 51, 7, 0, 255, 6, 0, 0, 2, 12, 9, 6, 1, 0, 71, 7, 0, 255, 6, 0, 0, 2, 114, 9, 6, 1, 0, 32, 7, 0, 255, 6, 0, 0, 9, 49, 9, 6, 1, 0, 52, 7, 0, 255, 6, 0, 0, 9, 64, 9, 6, 1, 0, 72, 7, 0, 255, 6, 0, 0, 9, 51, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 10, 15, 6, 15, 8, 0, 255, 255, 196, 0, 9, 6, 1, 0, 1, 8, 0, 255, 6, 0, 148, 130, 98, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 1, 47, 198, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 19, 107, 43, 145, 61, 126]
2022-10-15 15:08:15.763 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kaifa] Invalid packet size 287
2022-10-15 15:08:15.763 DEBUG (Thread-2 (connect)) [custom_components.ams] failed package: [126, 161, 29, 1, 0, 1, 16, 176, 174, 230, 231, 0, 15, 64, 0, 0, 0, 0, 2, 36, 9, 6, 1, 0, 0, 2, 129, 255, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 6, 0, 0, 96, 1, 0, 255, 9, 16, 55, 51, 52, 48, 49, 53, 55, 48, 51, 48, 53, 52, 56, 51, 48, 48, 9, 6, 0, 0, 96, 1, 7, 255, 9, 7, 77, 65, 51, 48, 52, 72, 52, 9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 1, 41, 9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 107, 9, 6, 1, 0, 31, 7, 0, 255, 6, 0, 0, 2, 104, 9, 6, 1, 0, 51, 7, 0, 255, 6, 0, 0, 2, 17, 9, 6, 1, 0, 71, 7, 0, 255, 6, 0, 0, 2, 46, 9, 6, 1, 0, 32, 7, 0, 255, 6, 0, 0, 9, 50, 9, 6, 1, 0, 52, 7, 0, 255, 6, 0, 0, 9, 65, 9, 6, 1, 0, 72, 7, 0, 255, 6, 0, 0, 9, 48, 9, 6, 0, 0, 1, 0, 0, 255, 9, 12, 7, 230, 10, 15, 6, 15, 8, 15, 255, 255, 196, 0, 9, 6, 1, 0, 1, 8, 0, 255, 6, 0, 148, 130, 99, 9, 6, 1, 0, 2, 8, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 0, 3, 8, 0, 255, 6, 0, 1, 47, 198, 9, 6, 1, 0, 4, 8, 0, 255, 6, 0, 19, 107, 43, 188, 84, 126]
2022-10-15 15:08:28.249 DEBUG (SyncWorker_6) [custom_components.ams] stop_serial_read
2022-10-15 15:08:28.290 DEBUG (Thread-2 (connect)) [custom_components.ams] failed package: None

Best regards

turbokongen, 2022-10-15:
Oh crap... Back to the drawing table. I will draft up a solution. It appears that different meters produce different types of data.

turbokongen, 2022-10-15:
Please try this proposed code: https://github.com/turbokongen/hass-AMS/tree/kaifa_se_dual
@husky999 @futrax Can you also test, with your meters?
The two meters MA304H4 and MA304H4D sendes quite different type of packages.

runlar, 2022-10-15:
Hi

It seems to work fine.

I have attachad my log file.

[home-assistant_kaifa_se_dual.log](https://github.com/turbokongen/hass-AMS/files/9792968/home-assistant_kaifa_se_dual.log)

PR #86 (2022-10-19): Allow two different types of swedish kaifa meters alongside norwegiam ones, fix signed currents.
Fixes #85 
Fixes #83
closed 2022-10-19 by turbokongen