turbokongen/hass-AMS #88: Kaifa MA304H4 is working, but with several incorrect redouts

ohagenes2, 2022-10-29:
Thanks for making such a cool and useful piece of software.

I have a Kaifa MA304H4 meter in my house (400V TN). I’m able read the data and everything looks ok, except that most of the readouts are incorrect.

Active Power Input is the only readout I am been able to verify to be correct. 

Active Power Export and Active Energy Export might be correct enough. Zero in the meter display.

I’m not able to read off Reactive Power and Reactive Energy in my meter display. 

The other readouts are incorrect.  

I have added a log file and a screenshot of my AMS Energy Meter. I hope it is sufficient for you to figure out what is needed to correct the issue.

[home-assistant.log](https://github.com/turbokongen/hass-AMS/files/9894542/home-assistant.log)
[image]

turbokongen, 2022-10-29:
I see you have selected kaifa as parser. Have you tried auto or kaifa_se?

Sendt fra min Commodore 64

ohagenes2, 2022-10-30:
I have tried “auto” and “kaifa_se” and it doesn’t look right, no readouts (utilgjengelig). The “kaifa” option seems to be my best chois for me so far. What do you think? I have attached my “auto” and “kaifa_se” log. 

[home-assistant_auto.log](https://github.com/turbokongen/hass-AMS/files/9895682/home-assistant_auto.log)
[home-assistant_kaifa_se.log](https://github.com/turbokongen/hass-AMS/files/9895683/home-assistant_kaifa_se.log)

turbokongen, 2022-10-30:
Thank you for the data. Could you test https://github.com/turbokongen/hass-AMS/tree/no_meter_type to see if that fixes your problem

ohagenes2, 2022-10-31:
Thank you for the fix, it is perfect! :-)

All readouts are verified, except from the Active Power Export and the Active Power Import which is not available as readouts in my meter display.

Why the difference, a firmware upgrade of the meter maybe…?

closed 2022-11-01 by turbokongen