turbokongen/hass-AMS #90: Active power import jumping to 100 000 000 W

joernnilsson, 2022-11-04:
I'm struggling with what looks like a parsing error. `ams_active_power_import` is jumping from the actual value of around 1 kW to a fixed value of 100M.

Other measurements look ok, although I haven't checked the history of all of them.

This started yesterday when I upgraded to 1.9.7, but stayed when I tried downgrading to 1.9.6.

I'm quite capable of gathering debug data and testing changes, but wanted to check with the dev(s) first if you have any pointers.

Meter:
Kaifa MA304H3E (Norwegian)

[image]

turbokongen, 2022-11-05:
Can you enable debug, and attach the log. I may have an Idea of what is wrong, but I need the logs to confirm. 👍

joernnilsson, 2022-11-07:
`2022-11-07 09:34:40.448 INFO (MainThread) [custom_components.ams] No YAML config available, using config_entries
2022-11-07 09:34:40.469 DEBUG (MainThread) [custom_components.ams] config entry = {'serial_port': '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0', 'meter_manufacturer': 'kaifa', 'parity': 'N', 'baudrate': 2400, 'protocol': 'serial'}
2022-11-07 09:34:40.469 DEBUG (MainThread) [custom_components.ams] Connecting to HAN using serialport /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
2022-11-07 09:34:40.471 DEBUG (MainThread) [custom_components.ams] Finish init of AMS
2022-11-07 09:34:41.458 DEBUG (Thread-2 (connect)) [custom_components.ams] data read from port=[126, 160, 121, 1, 2, 1, 16, 128, 147, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 11, 7, 1, 9, 34, 40, 255, 128, 0, 0, 2, 13, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 52, 49, 50, 57, 57, 53, 52, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 2, 118, 6, 0, 0, 0, 0, 6, 0, 0, 0, 16, 6, 0, 0, 0, 0, 6, 0, 0, 3, 11, 6, 0, 0, 8, 167, 6, 0, 0, 8, 251, 6, 0, 0, 9, 22, 6, 0, 0, 0, 0, 6, 0, 0, 9, 45, 213, 78, 126]
2022-11-07 09:34:41.458 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kaifa] kaifa meter_type length is 8
2022-11-07 09:34:41.460 DEBUG (Thread-2 (connect)) [custom_components.ams] miss_attrs=[]
2022-11-07 09:34:41.460 DEBUG (Thread-2 (connect)) [custom_components.ams] Got 10 new devices from the serial
2022-11-07 09:34:41.460 DEBUG (Thread-2 (connect)) [custom_components.ams] DUMP {'ams_active_power_import': {'state': 100663298, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}, 'ams_active_power_export': {'state': 0.0, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_import': {'state': 16, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_reactive_power_export': {'state': 0, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'VAr', 'icon': 'mdi:gauge'}}, 'ams_current_l1': {'state': 0.779, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l2': {'state': 2.215, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_current_l3': {'state': 2.299, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'A', 'icon': 'mdi:current-ac'}}, 'ams_voltage_l1': {'state': 232.6, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l2': {'state': 0.0, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}, 'ams_voltage_l3': {'state': 234.9, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'V', 'icon': 'mdi:flash'}}}
2022-11-07 09:34:41.461 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_import
2022-11-07 09:34:41.461 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Init ams_active_power_import DUMP sensor_states {'name': 'ams_active_power_import', 'state': 100663298, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}
2022-11-07 09:34:41.461 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_export
2022-11-07 09:34:41.461 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Init ams_active_power_export DUMP sensor_states {'name': 'ams_active_power_export', 'state': 0.0, 'attributes': {'timestamp': '2022-11-7 09:34:40', 'meter_manufacturer': 'Kfm_001', 'meter_type': 'Domestic/Industrial 3 Phase 230V 3-Wire meter', 'meter_serial': '6970631404129954', 'unit_of_measurement': 'W', 'icon': 'mdi:gauge'}}
[... cut ...]
2022-11-07 09:37:06.866 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l2
2022-11-07 09:37:06.866 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l3
2022-11-07 09:37:06.866 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l1
2022-11-07 09:37:06.867 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l2
2022-11-07 09:37:06.867 DEBUG (Thread-2 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l3
`

joernnilsson, 2022-11-07:
Attached log, from boot to having seen the issue at least once

[ams.log](https://github.com/turbokongen/hass-AMS/files/9949566/ams.log)

joernnilsson, 2022-11-07:
I printed the parsed states as well, and it seems to happen when a larger chunk of data is read:

```
2022-11-07 09:44:38.838 DEBUG (Thread-3 (connect)) [custom_components.ams] data read from port=[126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 11, 7, 1, 9, 44, 38, 255, 128, 0, 0, 2, 1, 6, 0, 0, 2, 37, 242, 181, 126]
2022-11-07 09:44:38.839 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_import (state: 549)
2022-11-07 09:44:38.839 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_export (state: 0.0)
2022-11-07 09:44:38.840 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import (state: 0)
2022-11-07 09:44:38.840 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export (state: 81)
2022-11-07 09:44:38.840 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l1 (state: 0.787)
2022-11-07 09:44:38.840 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l2 (state: 1.874)
2022-11-07 09:44:38.841 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l3 (state: 2.108)
2022-11-07 09:44:38.841 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l1 (state: 231.7)
2022-11-07 09:44:38.841 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l2 (state: 0.0)
2022-11-07 09:44:38.841 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l3 (state: 234.8)
2022-11-07 09:44:41.291 DEBUG (Thread-3 (connect)) [custom_components.ams] data read from port=[126, 160, 121, 1, 2, 1, 16, 128, 147, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 11, 7, 1, 9, 44, 40, 255, 128, 0, 0, 2, 13, 9, 7, 75, 70, 77, 95, 48, 48, 49, 9, 16, 54, 57, 55, 48, 54, 51, 49, 52, 48, 52, 49, 50, 57, 57, 53, 52, 9, 8, 77, 65, 51, 48, 52, 72, 51, 69, 6, 0, 0, 2, 34, 6, 0, 0, 0, 0, 6, 0, 0, 0, 0, 6, 0, 0, 0, 81, 6, 0, 0, 3, 13, 6, 0, 0, 7, 37, 6, 0, 0, 8, 14, 6, 0, 0, 9, 16, 6, 0, 0, 0, 0, 6, 0, 0, 9, 44, 111, 222, 126]
2022-11-07 09:44:41.291 DEBUG (Thread-3 (connect)) [custom_components.ams.parsers.kaifa] kaifa meter_type length is 8
2022-11-07 09:44:41.292 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_import (state: 100663298)
2022-11-07 09:44:41.292 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_export (state: 0.0)
2022-11-07 09:44:41.293 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import (state: 0)
2022-11-07 09:44:41.293 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export (state: 81)
2022-11-07 09:44:41.294 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l1 (state: 0.781)
2022-11-07 09:44:41.294 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l2 (state: 1.829)
2022-11-07 09:44:41.295 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l3 (state: 2.062)
2022-11-07 09:44:41.295 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l1 (state: 232.0)
2022-11-07 09:44:41.296 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l2 (state: 0.0)
2022-11-07 09:44:41.296 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l3 (state: 234.8)
2022-11-07 09:44:42.899 DEBUG (Thread-3 (connect)) [custom_components.ams] data read from port=[126, 160, 39, 1, 2, 1, 16, 90, 135, 230, 231, 0, 15, 64, 0, 0, 0, 9, 12, 7, 230, 11, 7, 1, 9, 44, 42, 255, 128, 0, 0, 2, 1, 6, 0, 0, 2, 48, 68, 253, 126]
2022-11-07 09:44:42.900 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_import (state: 560)
2022-11-07 09:44:42.901 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_active_power_export (state: 0.0)
2022-11-07 09:44:42.901 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_import (state: 0)
2022-11-07 09:44:42.901 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_reactive_power_export (state: 81)
2022-11-07 09:44:42.901 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l1 (state: 0.781)
2022-11-07 09:44:42.902 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l2 (state: 1.829)
2022-11-07 09:44:42.902 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_current_l3 (state: 2.062)
2022-11-07 09:44:42.902 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l1 (state: 232.0)
2022-11-07 09:44:42.902 DEBUG (Thread-3 (connect)) [custom_components.ams.sensor] Updating sensor ams_voltage_l2 (state: 0.0)
```

joernnilsson, 2022-11-07:
https://github.com/turbokongen/hass-AMS/blob/1738118e8a727cd924ab255994b2b026795583a9/custom_components/ams/parsers/kaifa.py#L92-L112

Changing the call on line 100 to `byte_decode(fields=pkt[71:75])` seems to give correct results on my setup. I wouldn't presume to know the history of these parsers well enough to say it's a solution, but perhaps you do @turbokongen.

If this is merely a index issue, I can create a PR with the change. If this change brakes something else, we'll have to go a bit deeper.

turbokongen, 2022-11-08:
Fixed in #91 [v1.9.8](https://github.com/turbokongen/hass-AMS/tree/v1.9.8) Thank you for reporting

closed 2022-11-08 by turbokongen