turbokongen/hass-AMS #94: See only 5 entities after installation using aidon_se, see zero entities when using aidon

helgeha, 2022-12-30:
Hi
I am having trouble trying to complete the installation of hass-AMS.  I am in Norway with Raspberry pi 4 connected with USB cable to OSS-brikke which is connected to Aidon 6525
I do not get any errors during installation, and I am not sure if I am looking in the right logs for troubleshooting this.
In Home assistant I see zero devices and zero entities related to AMS Reader if I set meter_manufacturer: 'aidon' in /root/config/configuration.yaml 
If I change this to meter_manufacturer: 'aidon_se' then in Home assistant I see 1 device and 5 entities related to AMS Reader. 
HA > Settings > Devices & services > Integrations > AMS reader > now it says 1 device and 5 entities:
ams_active_energy_export_00
ams_active_energy_import_00
ams_active_power_import_00 4,788 W
ams_reactive_energy_export_00
ams_reactive_energy_import_00

Nothing appears in HA > Energy after installation.
So I have received 5, and it should be totally 14 entities according to https://github.com/turbokongen/hass-AMS/issues/80
Should I see anything inside Home assistant Energy after installation of AMS Reader?
Should I look in any other logs than HA > Settings > System > Logs > Home assistant core > Load full logs ?
I am a newbie trying to do my best in reading documentations, searching forums and watching youtube videos.

I have tested on Raspberry pi 4 with Home assistant operating system
I have tested on Raspberry pi 3b+ with RPI OS Lite 64-bit and Home assistant container installed in Docker
I experience this in both environments

I have tried restarting HA and restarting RPI
I have tried deleting cache inside Edge on my Windows 10 computer
I have installed via HACS followed by setup of AMS reader in HA > Settings > Devices & services > Integrations > Add integration > AMS reader.  This gave no devices and no entities.
I have tried adding ams into configuration.yaml for HA, and this seems to have taken it a step further, because now I receive data almost every 2 seconds inside this log in HA > Settings > System > Logs > Home assistant core > Load full logs. See example from logs further down.

Here comes more details of the steps that have been done so far on RPI4:
The OSS mobile app is receiving data from OSS-brikke
Downloaded RPI4 64bit Home assistant operating system
HA > Settings > About:	
	Home Assistant 2022.12.8
	Supervisor 2022.12.1
	Operating system 9.4
	Frontend 20221213.1 - latest

HA > Settings > Add-ons > Add-on store > Official add-ons > Terminal & SSH > Install
HA > Settings > Add-ons > Terminal & SSH > Start button > Open web ui > ha info
arch: aarch64
channel: stable
docker: 20.10.19
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- os_agent
- haos
- resolved
- journal
hassos: "9.4"
homeassistant: 2022.12.8
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 9.4
state: running
supervisor: 2022.12.1
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Oslo

Installed HACS according to https://hacs.xyz/docs/setup/download
HA > HACS > Integrations > Explore & download repositories > hass-ams > Download > version = v1.9.8 > Download
HA > Settings > System > Restart > Restart
[... 134 lines cut ...]
There is no other than list_type is 1

I tested with different manufacturer, and repeated the following steps to collect the following information:
HA > Settings > Devices & services > Integrations > AMS reader > Delete
HA > Settings > Add-ons > Terminal & SSH > Open web ui > nano /root/config/configuration.yaml 
HA > Settings > Devices & services > Integrations > add integration > AMS reader > select same manufacturer as in yaml
HA > Settings > System > Restart
HA > Settings > Devices & services > Integrations > AMS reader > now it says X device and X entities
HA > Settings > System > Logs > Home assistant core > Load full logs > CTRL + F to search for list_type is

Manufacturer   Entities  list_type is      numbers received
auto		        0	none		no entity to show anything
aidon		0      1		        no entity to show anything
aidon_se	        5	1		        ams_active_power_import_00 4,788 W
kaifa		0	none		no entity to show anything
kaifa_se	        0	1		        no entity to show anything
kamstrup	        0	none		no entity to show anything

Attached log from each of these tests
[aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10325754/aidon_se.txt)
[auto.txt](https://github.com/turbokongen/hass-AMS/files/10325755/auto.txt)
[kaifa.txt](https://github.com/turbokongen/hass-AMS/files/10325756/kaifa.txt)
[kaifa_se.txt](https://github.com/turbokongen/hass-AMS/files/10325757/kaifa_se.txt)
[kamstrup.txt](https://github.com/turbokongen/hass-AMS/files/10325758/kamstrup.txt)
[aidon.txt](https://github.com/turbokongen/hass-AMS/files/10325759/aidon.txt)

turbokongen, 2022-12-30:
Try using default values for baudrate and parity.  Only the small data
package from the meter is received correctly. The others are chopped at
44bytes. Everything else looks good.

Sendt fra min Commodore 64

helgeha, 2022-12-30:
Thanks for quick reply
ok, now I tried using baudrate 2400 parity N, but then I get no entities and no data.  
I get 5 entities and receive data only with baudrate 115200.  I followed my same steps as before, but in addition deleted cache, see steps at bottom.

I have been moving OSS-brikke from RPI4 to RPI3b+ and I guess that made it move to ttyUSB1.  I rebooted RPI4, and the reboot set it back to ttyUSB0.  Here are test results:

Serial port      Baud Parity Manuf.     entities
/dev/ttyUSB1  2400 N aidon_se          none
/dev/ttyUSB1  2400 N aidon                 none
/dev/ttyUSB1  2400 N auto                  none
/dev/ttyUSB1  115200 N auto                  5
/dev/ttyUSB1  4800 N auto                  none
/dev/ttyUSB1  2400 N aidon_se          none

HA > Settings > Add-ons > Terminal & SSH > Open web ui > reboot
/dev/ttyUSB0  2400 N aidon                 none
/dev/ttyUSB0  2400 N aidon_se            none

HA > Settings > Add-ons > Terminal & SSH > Open web ui > cat /dev/ttyUSB0 > I receive nothing, and the cursor moves to next command line where I can write next command
HA > Settings > Add-ons > Node-red > open web ui > serial-in > change to /dev/ttyUSB0 2400 8 None 1 > Update > Done > Deploy 
The node ‘serial-in’ says connected
Node-red > Debug > receive no data
HA > Settings > Add-ons > Terminal & SSH > Open web ui > cat /dev/ttyUSB0 > receive only one character each 10 seconds
/dev/ttyUSB0  115200 N aidon_se            5
I get the impression that Node-red somehow gives the connection some kind of refresh because cat /dev/ttyUSB0 start receiving some data after Node-red has been connected to ttyUSB0.

Here are my test steps:
HA > Settings > Devices & services > Integrations > AMS reader > Delete
HA > Settings > Add-ons > Terminal & SSH > Open web ui > nano /root/config/configuration.yaml
On my Windows 10 > Microsoft Edge > … > Settings > Privacy, search & services > Clear browsing data > Choose what to clear > "Cached images and files" + "Cookies and other site data" > Clear
HA > Settings > Devices & services > Integrations > add integration > AMS reader > select same manufacturer as in yaml
HA > Settings > System > Restart
HA > Settings > Devices & services > Integrations > AMS reader > now it says X device and X entities
HA > Settings > System > Logs > Home assistant core > Load full logs
[ttyUSB1_2400_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10326278/ttyUSB1_2400_N_aidon_se.txt)
[ttyUSB1_2400_N_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10326279/ttyUSB1_2400_N_aidon.txt)
[ttyUSB1_2400_N_auto.txt](https://github.com/turbokongen/hass-AMS/files/10326280/ttyUSB1_2400_N_auto.txt)
[ttyUSB1_115200_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10326281/ttyUSB1_115200_N_aidon_se.txt)
[ttyUSB1_4800_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10326282/ttyUSB1_4800_N_aidon_se.txt)
[ttyUSB0_2400_N_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10326283/ttyUSB0_2400_N_aidon.txt)
[ttyUSB0_2400_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10326284/ttyUSB0_2400_N_aidon_se.txt)
[ttyUSB0_115200_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10326285/ttyUSB0_115200_N_aidon_se.txt)

turbokongen, 2022-12-31:
There looks to be a problem with the communication.
From the logs, serial communication just randomly times out mid package. Ranging from 2 bytes to two hundred something bytes. 
Doing some reasearch, I found this: https://github.com/toreamun/amshan-homeassistant/issues/48
The raw data looks very similart to what you have provided in the logs.
I would try to play around with parity to try and see what happens. From research, the OSS seems to only work at 115200 baud.
All packages should start end end with 126 in the logs. That means a complete package has been received.  Partial packages willl not be decoded.
Does any other raw Mbus addon work properly?

helgeha, 2023-01-01:
I have repeated my test, now playing around with Parity E, O or N.  The best test result is 115200 N aidon_se which creates 5 entities.  The log still shows many packages not ending with 126.  Here are my test details:
Serial port      Baudrate  Parity  Manuf.     Entities     Log
/dev/ttyUSB0  115200  E  aidon_se          none		log receives data, but many not ending with 126
/dev/ttyUSB0  115200  O  aidon_se          none		log receives data, but many not ending with 126
/dev/ttyUSB0  115200  N  aidon_se          5 (1592 W)	log receives data, but many not ending with 126
/dev/ttyUSB0  115200  E  aidon                none		log receives data, but many not ending with 126
/dev/ttyUSB0  115200  O  aidon                none		log receives data, but many not ending with 126
/dev/ttyUSB0  115200  N  aidon                none		log receives data, but many not ending with 126
/dev/ttyUSB0  2400       E  aidon_se          none		log receives no data 
/dev/ttyUSB0  2400       O  aidon_se          none		log receives no data
/dev/ttyUSB0  2400       N  aidon_se          none		log receives no data
/dev/ttyUSB0  2400       E  aidon                 none		log receives no data
/dev/ttyUSB0  2400       O  aidon                 none		log receives no data
/dev/ttyUSB0  2400       N  aidon                 none		log receives no data

[ttyUSB0_115200_E_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329179/ttyUSB0_115200_E_aidon_se.txt)
[ttyUSB0_115200_O_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329180/ttyUSB0_115200_O_aidon_se.txt)
[ttyUSB0_115200_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329181/ttyUSB0_115200_N_aidon_se.txt)
[ttyUSB0_115200_E_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329182/ttyUSB0_115200_E_aidon.txt)
[ttyUSB0_115200_O_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329183/ttyUSB0_115200_O_aidon.txt)
[ttyUSB0_115200_N_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329184/ttyUSB0_115200_N_aidon.txt)
[ttyUSB0_2400_E_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329186/ttyUSB0_2400_E_aidon_se.txt)
[ttyUSB0_2400_O_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329187/ttyUSB0_2400_O_aidon_se.txt)
[ttyUSB0_2400_N_aidon_se.txt](https://github.com/turbokongen/hass-AMS/files/10329189/ttyUSB0_2400_N_aidon_se.txt)
[ttyUSB0_2400_E_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329190/ttyUSB0_2400_E_aidon.txt)
[ttyUSB0_2400_O_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329191/ttyUSB0_2400_O_aidon.txt)
[ttyUSB0_2400_N_aidon.txt](https://github.com/turbokongen/hass-AMS/files/10329192/ttyUSB0_2400_N_aidon.txt)

I get error ‘Timeout reading data’ when trying to connect toreamun AMSHAN to OSS-brikke, so it seems like I have not found the correct combination of parameters for setting up AMSHAN correctly.

So far I have succeeded with connecting your turbokongen AMS Reader to OSS-brikke by typing into configuration.yaml, but I get only 5 of 14 entities and receive only watt every 2 seconds.

In Node-red I only receive watt every 2 seconds, and this was done by using https://flows.nodered.org/node/node-red-contrib-ams-decoder

I am not sure what you mean with raw Mbus addons. Can you name some of them ?

turbokongen, 2023-01-06:
Hi.
I was referring to toreamun's AMSHAN regarding other mbus addons.
This looks to be an issue with the way OSS birkken sends data over serial interface.
Can you try to use this branch: https://github.com/turbokongen/hass-AMS/tree/timeout-test to do a test?
I increased the reading timeout on serial interface from 0.1sec to 1sec.
If that is a success, I will add in configurable timeout setting to the config.

helgeha, 2023-01-08:
Hi
I am not sure how to use branch.  I installed hass-AMS in HACS, and there I can only select version 1.9.8, 2.0.0 etc, but I cannot see any option for selecting branch inside HACS.  I tried in Terminal sudo git clone on the existing installation inside folder ams, but this seems to not understand git branch and git checkout.  Therefore, I manually changed const.py line 79 DEFAULT_TIMEOUT = 1.  
Are there any other changes that need to be done?  
Is there some other way I should install this to be able to use branch?  

On my current test where timeout is set to 1 I still receive the same result as before: 1 device, 5 entities, ams_active_power_import_00 shows 836 W and is updated every 2 to 25 seconds.

I manually changed const.py line 79 DEFAULT_TIMEOUT = 1 in both of my test environments RPI4 and RPI3b+, and both gave the same test result.  I had some extra issue on my RPI4 which was not showing any data in the logs despite that I was receiving watt every 2 seconds.  Therefore, I have no logs to show from RPI4.  But I have attached logs from RPI3b+ which has Home Assistant container in Docker.

Here are my test details from RPI3b+:
git version	
	2.30.2
git branch	
	fatal: not a git repository (or any of the parent directories): .git
git checkout	
	fatal: not a git repository (or any of the parent directories): .git
cd ./homeassistant/config/custom_components/ams
sudo git clone https://github.com/turbokongen/hass-AMS/tree/timeout-test	
	this succeeded
cat ./homeassistant/config/custom_components/ams/manifest.json: "version": "1.9.8"	
cat ./homeassistant/config/custom_components/ams/const.py > line 79 DEFAULT_TIMEOUT = 0.1	
sudo nano /root/config/custom_components/ams/const.py > line 79 DEFAULT_TIMEOUT = 1	
	
HA > Settings > Devices & services > Integrations > AMS reader > Delete	
HA > Settings > System > Restart	
On my Windows 10 > Microsoft Edge > … > Settings > Privacy, search & services > Clear browsing data > Choose what to clear > "Cached images and files" + "Cookies and other site data" > Clear	
HA > Settings > Devices & services > Integrations > add integration > AMS reader > select same manufacturer as in yaml	
HA > Settings > System > Restart	
On my Windows 10 > Microsoft Edge > … > Settings > Privacy, search & services > Clear browsing data > Choose what to clear > "Cached images and files" + "Cookies and other site data" > Clear	
HA > Settings > Devices & services > Integrations > AMS reader > now it says 1 device and 5 entities > ams_active_power_import... 836 W > and the watt is changing every 2 seconds to 25 seconds	
HA > Settings > System > Logs > Home assistant core > Load full logs > lots of data, but only "list_type is 1"	

[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout01.txt](https://github.com/turbokongen/hass-AMS/files/10368766/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout01.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout1.txt](https://github.com/turbokongen/hass-AMS/files/10368767/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout1.txt)

turbokongen, 2023-01-08:
Excellent! You can also use the method you are using.  I only changed the
default timeout setting.
Looks like we are on the right path to succes.
Now all failed packets are consistently 269bytes. Data inside looks good as
well. Still, the packet end byte is not received.
Try increasing the timeout until you get complete packages. The aidon meter
interval between packages is 2.5 so 2.4 should be the max timeout. I'm
guessing a timeout around 1.5 should be sufficient. Remember that the
hourly package is lager than the 10 second packages, so a larger timeout
than what works with the 10sec packages might be needed.

Sendt fra min Commodore 64

helgeha, 2023-01-09:
Hi
I removed ams folder, ran git clone, and now I succeeded in running sudo git checkout timeout-test.  Copied this to folder ams
cd ./homeassistant/config/custom_components/ams
This shows:
cat manifest.json > "version": "2.0.0"
cat const.py shows DEFAULT_TIMEOUT = 1

I tested with different timeout, but Devices & services still shows only 5 entities and there is only 'list_type is 1' on all tests
-----------------------------------------------------------
Here are my steps for changing timeout:
sudo nano const.py > DEFAULT_TIMEOUT = change this
HA > Settings > Devices & services > Integrations > AMS reader > Delete
On my Windows 10 > Microsoft Edge > … > Settings > Privacy, search & services > Clear browsing data > Choose what to clear > "Cached images and files" + "Cookies and other site data" > Clear
HA > Settings > Devices & services > Integrations > add integration > AMS reader > select same as in yaml /dev/ttyUSB0 115200 N aidon_se
HA > Settings > System > Restart
On my Windows 10 > Microsoft Edge > … > Settings > Privacy, search & services > Clear browsing data > Choose what to clear > "Cached images and files" + "Cookies and other site data" > Clear
HA > Settings > Devices & services > Integrations > AMS reader > on every timeout test it says 1 device and 5 entities > ams_active_power_import... 1821 W > and the watt is changing every 2 seconds to 25 seconds
HA > Settings > System > Logs > Home assistant core > Load full logs
-----------------------------------------------------------
Here is a summary of the log content on different timeout:
[summary_of_logs.txt](https://github.com/turbokongen/hass-AMS/files/10377136/summary_of_logs.txt)

Here are the logs:
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout1_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377138/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout1_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout15_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377144/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout15_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout01_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377146/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout01_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout100_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377149/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout100_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout25_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377152/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout25_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout24_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377153/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout24_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout20_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377156/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout20_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout18_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377157/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout18_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout16_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377160/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout16_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout15_on_branch_timeout_test_b.txt](https://github.com/turbokongen/hass-AMS/files/10377163/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout15_on_branch_timeout_test_b.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout14_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377165/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout14_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout13_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377166/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout13_on_branch_timeout_test.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout05_on_branch_timeout_test.txt](https://github.com/turbokongen/hass-AMS/files/10377176/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_timeout05_on_branch_timeout_test.txt)

helgeha, 2023-01-19:
Hi
I cloned your updated code
sudo git checkout timeout-test
copied from folder hass-AMS to ams
sudo nano const.py > DEFAULT_TIMEOUT = manually changed this
HA > Settings > Devices & services > Integrations > AMS reader > ... > Delete
HA > Settings > System > Restart
HA > Settings > Devices & services > Integrations > Add integration > AMS reader > select same as in yaml /dev/ttyUSB0 115200 N aidon_se
HA > Settings > System > Restart
Repeated this for different timeout settings.  Here is summary of log content:
[summary_of_logs.txt](https://github.com/turbokongen/hass-AMS/files/10459262/summary_of_logs.txt)

Here are the logs:
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout01.txt](https://github.com/turbokongen/hass-AMS/files/10459263/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout01.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout05.txt](https://github.com/turbokongen/hass-AMS/files/10459264/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout05.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout1.txt](https://github.com/turbokongen/hass-AMS/files/10459269/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout1.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout15.txt](https://github.com/turbokongen/hass-AMS/files/10459270/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout15.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout20.txt](https://github.com/turbokongen/hass-AMS/files/10459272/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout20.txt)
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout24.txt](https://github.com/turbokongen/hass-AMS/files/10459273/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout24.txt)

turbokongen, 2023-01-21:
Ok, this is a weird one....
The data received is perfectly fine, and I can decode it directly using the parser.
The package is missing the frame crc and end byte (7e / 126)
The package size is correctly set in the package without them.
When I visit the official code from OSS, they state that the data is base64 encoded, which clearly it isn't.

This is the data broken down:
```
126, "Frame delimiter"
161, 11, "size of package == 269" Which is correct for the data received.
65, 8, 131, 19, 250, 124, 230, 231, 0, 15, 64, 0, 0, 0, 0, 1,
12, "list_type" 12 which is also correct for the data received. 12 elements in list.
2, 2, "header-stuff"

9, 6, 1, 1, 0, 2, 129, 255, 13, 10, 11, 65, 73, 68, 79, 78, 95, 86, 48, 48, 48, 49, 2, 2, "OBIS_list_identifier"
9, 6, 0, 0, 96, 1, 0, 255, 13, 10, 16, 55, 51, 53, 57, 57, 57, 50, 56, 57, 56, 48, 51, 51, 55, 55, 55, 2, 2, "meter_serial"
9, 6, 0, 0, 96, 1, 7, 255, 13, 10, 4, 54, 53, 50, 53, 2, 3, "meter_type"
9, 6, 1, 0, 1, 7, 0, 255, 6, 0, 0, 17, 153, 2, 2, 15, 0, 22, 27, 2, 3, "active_power_import"
9, 6, 1, 0, 2, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 27, 2, 3, "active_power_export"
9, 6, 1, 0, 3, 7, 0, 255, 6, 0, 0, 0, 103, 2, 2, 15, 0, 22, 29, 2, 3, "reactive_power_import"
9, 6, 1, 0, 4, 7, 0, 255, 6, 0, 0, 0, 0, 2, 2, 15, 0, 22, 29, 2, 3, "reactive_power_export"
9, 6, 1, 0, 31, 7, 0, 255, 16, 0, 76, 2, 2, 15, 255, 22, 33, 2, 3, "current_l1"
Missing "current_l2", as this is obviously a three phase meter, give you have three voltages.???
9, 6, 1, 0, 71, 7, 0, 255, 16, 0, 155, 2, 2, 15, 255, 22, 33, 2, 3, "current_l3"
9, 6, 1, 0, 32, 7, 0, 255, 18, 9, 58, 2, 2, 15, 255, 22, 35, 2, 3, "voltage_l1"
9, 6, 1, 0, 52, 7, 0, 255, 18, 9, 75, 2, 2, 15, 255, 22, 35, 2, 3, "voltage_l2"
9, 6, 1, 0, 72, 7, 0, 255, 18, 9, 46, 2, 2, 15, 255, 22, 35 "voltage_l3"
This is end of package.
Missing Frame CRC and Frame delimiter.
```

So I have updated the branch code with some changes.
You should now get 9 entities, and hopefully at the hour 4 more.
Please give it a try.

helgeha, 2023-01-21:
Thanks !!!
I used the default setting in const.py > DEFAULT_TIMEOUT = 0.1
Now I see 13 entities instead of only 5.  There are values on 11 entities, and the remaining 2 entities are zero because I do not produce / export power.  In Home assistant > Energy > Add consumption I have added ams_active_energy_import_00.  I will probably see energy usage in Home assistant energy dashboard after some hours.  So far everything seems to be working.  Here is example of log.
[RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout01_log98minutes.txt](https://github.com/turbokongen/hass-AMS/files/10472690/RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout01_log98minutes.txt)

Thank you so much for your help !!!

helgeha, 2023-01-22:
Hi again
I notice that the following values need to be multiplied by 10 to be right according to the values displayed on the physical Aidon power meter itself:
ams_active_energy_import_00 (kWh)
ams_reactive_energy_export_00 (kVAr)
ams_reactive_energy_import_00 (kVAr)

My power meter is Aidon 6483SE (it was not Aidon 6525 as I wrote in first issue description)
Should I multiply by 10 myself somewhere in Home assistant or in configuration.yaml, or should this be done somewhere in your hass-AMS code ?

turbokongen, 2023-01-22:
Could you send me a capture from the log of the hourly packages? I have to look at the raw data.
I found some other interesting things in regard of meter_type, meter_serial and meter_manufacturer.

turbokongen, 2023-01-22:
I have updated the branch code with the new Config Flow to set that you have the OSS brikken.
Could you please test this? That it fails if you do not select OSS, and works if you select OSS?
As far as the meter type, the data says Aidon 6525, but you probably have a 6483SE RF-2 addon module mounted as well. This is not the actual meter, but a HAN port addon mounted to some Aidon meters.

helgeha, 2023-01-23:
Hi
Can you use the log I sent 2 days ago ?  It contains 2 hourly packages, see file
RPI3bPlus_OSSbrikke_ttyUSB0_115200_N_aidon_se_on_branch_timeout_test_timeout01_log98minutes.txt

Meter type: 
You are right.  When I look more closely on the physical power meter, it says Aidon 6525, and in bottom right it says Aidon 6483SE RF500

I tested your updated code on branch timeout-test, but it gives error before I see selection OSS=true/false.  Branch master shows 5 entities.  Here are details:
Test on branch timeout-test:
sudo rm -r hass-AMS				
sudo git clone https://github.com/turbokongen/hass-AMS.git				
cd hass-AMS				
sudo git checkout timeout-test				
cd ./homeassistant/config/custom_components/hass-AMS/custom_components				
sudo cp -R ams ../../				
cd ../../ams				
grep -A 5 -B 5 "OSS" const.py				
	CONF_OSS_BRIKKEN			
				
HA > Settings > Devices & services > Integrations > AMS reader > ... > Delete				
HA > Settings > System > Restart				
HA > Settings > Devices & services > Integrations > configuration.yaml AMS Reader > no devices, no entities				
HA > Settings > System > Logs > an error appear above / before click on Load full logs				
	Logger: root			
	Source: custom_components/ams/__init__.py:301			
	Integration: AMS Reader (documentation, issues)			
	First occurred: 10:44:02 AM (1 occurrences)			
	Last logged: 10:44:02 AM			
				
	Uncaught thread exception			
	Traceback (most recent call last):			
	  File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner			
	    self.run()			
	  File "/usr/local/lib/python3.10/threading.py", line 953, in run			
	    self._target(*self._args, **self._kwargs)			
	  File "/config/custom_components/ams/__init__.py", line 301, in connect			
	    if parser.test_valid_data(data, oss):			
	TypeError: test_valid_data() takes 1 positional argument but 2 were given			
				
HA > Settings > Devices & services > Integrations > configuration.yaml AMS Reader > Delete				
HA > Settings > Devices & services > Integrations > Add integration > AMS reader > Choose protocol for communications = there is no ttyUSB0 that I can select above Protocol type = serial > Submit > error appear below "Choose protocol for communications": Unknown error occurred				

--------------------------------------------------------------------------------------				
Test on branch master:				
sudo git checkout master				
cd ./homeassistant/config/custom_components/hass-AMS/custom_components				
sudo cp -R ams ../../				
cd ../../ams				
grep -A 5 -B 5 "OSS" const.py				
	did not find any OSS in that file			
HA > Settings > Devices & services > Integrations > there is no existing AMS reader				
HA > Settings > System > Restart				
HA > Settings > Devices & services > Integrations > configuration.yaml AMS Reader > 1 devices, 5 entities > see 2993 watt				
HA > Settings > System > Logs > no ams-errors appear above / before click on Load full logs				
HA > Settings > Devices & services > Integrations > configuration.yaml AMS Reader > Delete				
HA > Settings > Devices & services > Integrations > Add integration > AMS reader				
	AMS Reader			
	Choose protocol for communications			
	Protocol type = serial			
	Click on Submit			
Serial port = /dev/ttyUSB0: FT231X USB UART - FT231X USB UART				
Meter manufacturer = aidon_se				
Parity = N				
Baudrate = 115200				
Click on submit				
				
Success!				
Created configuration for AMS reader				
Finish				
				
[... 30 lines cut ...]
	Integration: AMS Reader (documentation, issues)			
	First occurred: 4:58:48 PM (1 occurrences)			
	Last logged: 4:58:48 PM			
				
	Uncaught thread exception			
	Traceback (most recent call last):			
	  File "/usr/local/lib/python3.10/threading.py", line 1016, in _bootstrap_inner			
	    self.run()			
	  File "/usr/local/lib/python3.10/threading.py", line 953, in run			
	    self._target(*self._args, **self._kwargs)			
	  File "/config/custom_components/ams/__init__.py", line 301, in connect			
	    if parser.test_valid_data(data, oss):			
	TypeError: test_valid_data() takes 1 positional argument but 2 were given			
HA > Settings > System > Logs > Load full logs > see attached file				
[Log_after_restart_HA.txt](https://github.com/turbokongen/hass-AMS/files/10481611/Log_after_restart_HA.txt)

HA > Settings > Devices & services > Integrations > configuration.yaml AMS Reader > Delete				
HA > Settings > Devices & services > Integrations > Add integration > AMS reader				
	AMS Reader			
	Choose protocol for communications			
	Protocol type = serial			
	Click on Submit			
Get error "Unknown error occurred"				
HA > Settings > System > Logs > Load full logs > see attached file				
[Log_when_Add_integration_AMS_reader_Protocol_type_serial_click_on_Submit.txt](https://github.com/turbokongen/hass-AMS/files/10481617/Log_when_Add_integration_AMS_reader_Protocol_type_serial_click_on_Submit.txt)

turbokongen, 2023-01-24:
My bad.... Values are described there in that file. :)
I have to fire up my testmachine to resolve that issue, but I will not have time till later this week. 
I will put changes out as pre-releases so you can just delete the integration you have on your machine, restart HA, and install the beta version directly from HACS.

turbokongen, 2023-01-29:
Please try:
https://github.com/turbokongen/hass-AMS/releases/tag/v2.0.1-beta11
You can install directly from HACS. Just enable betaversions.
See if you get the expected results.
You need to setup the integration from scratch, and enable OSS brikken option i the config flow.

helgeha, 2023-01-29:
Hi
I deleted the existing integration, removed existing folders ams and hass-AMS, and in HACS clicked on update to v2.0.1-beta11. The installation succeeded, but I see only 5 entities

Here are test details:
HA > Settings > Devices & services > Integration > AMS reader shows 0 devices and 0 entities > Delete	
cd ./homeassistant/config/custom_components	
sudo rm -r ams	
sudo rm -r hass-AMS	
HA > HACS > Updates > hass-AMS You are running version v2.0.1-beta7, version v2.0.1-beta11 is available > update	
I notice this added folder ./homeassistant/config/custom_components/ams	
HA > Settings > System > Restart	
HA > Settings > Devices & services > Integrations > Add integration > AMS reader	
	AMS Reader
	Choose protocol for communications
	Protocol type = serial
	Click on Submit
	Serial port = /dev/ttyUSB0: FT231X USB UART - FT231X USB UART
	Meter manufacturer = aidon_se
	Parity = N
	Baudrate = 115200
	OSS brikken = true
	Click on submit
	
	Success!
	Created configuration for AMS reader
	We found the following devices:
	AMS energy meter RF2-system module integrated HAN (aiden_se)
	Finish
	
HA > Settings > Devices & services > Integrations > AMS Reader > 1 devices, 5 entities (1650 watt)	
HA > Settings > System > Restart	
HA > Settings > Devices & services > Integrations > AMS Reader > 1 devices, 5 entities (1650 watt)	
HA > Settings > System > Logs > Load full logs > see attached log	
[Log202301291355_version201beta11.txt](https://github.com/turbokongen/hass-AMS/files/10529806/Log202301291355_version201beta11.txt)

turbokongen, 2023-01-29:
Hi there.
You have selected the wrong meter manufacturer. 
auto should work as well.
You need to use "adion" that is the norwegian parser. That is also the parser I have modified.
Please retry with aidon or auto as meter manufacturer. 👍

helgeha, 2023-01-29:
You are right.  Now I tested with aidon and oss_brikke true, and now I get 13 entities.  So this seems to be working.
[Log202301291515_version201beta11_aidon_oss_brikke.txt](https://github.com/turbokongen/hass-AMS/files/10529983/Log202301291515_version201beta11_aidon_oss_brikke.txt)

There are 4 entities showing state unknown (ams_active_energy_export, ams_active_energy_import, ams_reactive_energy_export, ams_reactive_energy_import), but I guess these will maybe show correct values after the next whole hour has passed.
Thank you very much for your help!

helgeha, 2023-01-29:
Hi again
There are still 4 entities showing state unknown after whole hour passed at 16:00.  The log at 16:00:12.312 says:
Packet size does not match read packet size: 377 : 380
[Log202301291602_version201beta11_aidon_oss_brikke.txt](https://github.com/turbokongen/hass-AMS/files/10530066/Log202301291602_version201beta11_aidon_oss_brikke.txt)

turbokongen, 2023-01-29:
My error. I had not got the hourly package data from your previous logs.
Corrected. Try v2.0.1-beta12

helgeha, 2023-01-30:
Hi and thank you very much!
I deleted existing integration, in HACS updated to v2.0.1-beta12, and then installed new integration.  The 4 entities that were showing state unknown are now showing values, so this solved the issue.  But the values now need to be devided by 10 for:
ams_active_energy_import_00 (kWh)
ams_reactive_energy_export_00 (kVAr)
ams_reactive_energy_import_00 (kVAr)
[Log202301301839_version201beta12_aidon_oss_brikke.txt](https://github.com/turbokongen/hass-AMS/files/10538447/Log202301301839_version201beta12_aidon_oss_brikke.txt)

turbokongen, 2023-01-30:
Modified in v2.0.1-beta14
Please retest :)

helgeha, 2023-01-30:
Hi
I updated to v2.0.1-beta14.  This gives correct values on these 3 entities.
Thank you so much for your help and time on this!
This is very much appreciated!

turbokongen, 2023-01-31:
Fixed by #97 
Thank you for your effort @helgeha 
New release v2.0.1

closed 2023-01-31 by turbokongen