turbokongen/hass-AMS #100: No parser detected

kristianwh, 2024-03-08:
Suddenly see that I stopped getting readings (I guess it has to be after updating to 2024.3.0)
This is the error I'm getting:

Logger: custom_components.ams
Source: custom_components/ams/__init__.py:360
integration: AMS Reader (documentation, issues)
First occurred: 21:04:07 (1 occurrences)
Last logged: 21:04:07

No parser detected

Tested with another AMS integration, and that worked..

motionthings, 2024-03-09:
This one looks unmaintained.

Which one did you switch to?
AMS HAN meter?

turbokongen, 2024-03-09:
Not unmaintained. I have not upgraded to 2024.3 yet. Still on 2024.2. Will look into it.

kristianwh, 2024-03-09:
Switched to amshan, but I prefer this one.

enpontus, 2024-03-09:
i see this problem as well and just tried to dig into it (but i am not very good at it, concluded though that myreader device seems to work...) eagerly awaiting a solution, thanks for the good work!

turbokongen, 2024-03-09:
I had a quick glance before I left home, and it seems like something has changed in the home assistant entity component. Data comes in, and get parsed fine, but fails to add entities to home assistant. They all are unknown.unknown=unknown
In the meantime, rverting back to 2024.2.4 fixes the problem until I can figure out.
I have to look at the changes in HA from 2024.2 to 2024.3 what could cause this.

turbokongen, 2024-03-13:
I am still trying to figure this out. I seem to have been away from python coding too long....
I found a related fault with error message, but I cannot find the offending place in my code!
It looks to be related to event_loop in homeassistant.

```
2024-03-13 06:11:22.178 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity_platform.py", line 625, in async_add_entities
    await add_func(coros, entities, timeout)
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity_platform.py", line 531, in _async_add_and_update_entities
    tasks = [create_eager_task(coro) for coro in coros]
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/util/async_.py", line 37, in create_eager_task
    loop=loop or get_running_loop(),
                 ^^^^^^^^^^^^^^^^^^
RuntimeError: no running event loop
2024-03-13 06:11:22.180 WARNING (MainThread) [py.warnings] /usr/lib/python3.12/asyncio/events.py:36: RuntimeWarning: coroutine 'EntityPlatform._async_add_entity' was never awaited
  def __init__(self, callback, args, loop, context=None):
```

closed 2024-03-17 by turbokongen