turbokongen/hass-AMS #104: Connection broken after 2024.5.x update

TEdvardS, 2024-05-03:
All devices got disconnected after updating to 2024.5.x
Connected to an 6525 3-phase Aidon meter via an serial connection to an OSS puch.

turbokongen, 2024-05-04:
Need more info. Please provide logs with debug enabled for the component.

TEdvardS, 2024-05-04:
I have downgraded back to 2024.4.4.
I can look at it when I am back home next weekend

Sendt fra min Galaxy

-------- Opprinnelig melding --------

Dato: 04.05.2024 10:10 (GMT+01:00)
Til: turbokongen/hass-AMS ***@***.***>
Ko: TEdvardS ***@***.***>, Author ***@***.***>

highzeth, 2024-05-04:
Can confirm the same happening here, asked a buddy to upgrade Core to confirm, same issue there. We both use USB serial adapter to Kaifa meter.

Debug log;

`2024-05-02 13:08:15.841 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration ams which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-02 13:08:23.973 WARNING (Thread-2 (connect)) [custom_components.ams] No parser detected
2024-05-02 13:08:26.009 WARNING (Thread-2 (connect)) [custom_components.ams] No parser detected
2024-05-02 13:08:27.979 WARNING (Thread-2 (connect)) [custom_components.ams] No parser detected
2024-05-02 13:08:30.469 WARNING (Thread-2 (connect)) [homeassistant.helpers.frame] Detected that custom integration 'ams' calls async_create_task from a thread at custom_components/ams/__init__.py, line 412: self._hass.async_create_task(self._signal_new_sensor()), please report it to the author of the 'ams' custom integration
2024-05-02 13:08:30.470 ERROR (Thread-2 (connect)) [root] Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 310, in connect
    self._check_for_new_sensors_and_update(self.sensor_data)
  File "/config/custom_components/ams/__init__.py", line 412, in _check_for_new_sensors_and_update
    self._hass.async_create_task(self._signal_new_sensor())
  File "/usr/src/homeassistant/homeassistant/core.py", line 816, in async_create_task
    self.verify_event_loop_thread("async_create_task")
  File "/usr/src/homeassistant/homeassistant/core.py", line 440, in verify_event_loop_thread
    frame.report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 162, in report
    _report_integration(what, integration_frame, level, error_if_integration)
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 203, in _report_integration
    raise RuntimeError(
RuntimeError: Detected that custom integration 'ams' calls async_create_task from a thread at custom_components/ams/__init__.py, line 412: self._hass.async_create_task(self._signal_new_sensor()). Please report it to the author of the 'ams' custom integration.
2024-05-02 13:08:30.545 WARNING (Thread-2 (connect)) [py.warnings] /usr/local/lib/python3.12/threading.py:1075: RuntimeWarning: coroutine 'AmsHub._signal_new_sensor' was never awaited
  self._invoke_excepthook(self)

2024-05-02 13:08:32.481 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-02 13:08:51.680 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.ams_active_energy_import_6970631408297680
2024-05-02 13:08:51.768 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 630: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-02 13:09:06.730 WARNING (MainThread) [homeassistant.core] Something is blocking Home Assistant from wrapping up the start up phase. We're going to continue anyway. Please report the following info at https://github.com/home-assistant/core/issues: lovelace, xiaomi_airfryer.switch, mobile_app.binary_sensor, xbox.binary_sensor, hardware, esphome.sensor, sql.sensor, frontend, zha.lock, zha.device_tracker, automation, api, wled.button, system_log, number, zha.fan, switch, assist_pipeline, google.calendar, dhcp, wled.select, zha, network, my, unifi.image, xiaomi_airfryer.sensor, deconz.button, wled, plex.button, update, scene, zha.siren, google, sensor, esphome.update, season, mobile_app.sensor, zha.sensor, media_source, xbox.remote, ams.sensor, stream, usb, xbox, deconz.fan, hassio.sensor, websocket_api, sensibo.button, wled.switch, light, sensibo.number, config, unifi.switch, webhook, conversation, ams, frigate.sensor, plex.media_player, harmony.select, zha.update, ssdp, input_datetime, tts, zha.select, systemmonitor.binary_sensor, wake_word, input_button, zone, http, sensibo.binary_sensor, zha.binary_sensor, squeezebox, deconz.number, deconz.sensor, blueprint, fan, backup, esphome.switch, deconz.switch, map, hassio.binary_sensor, zeroconf, image, script, sensibo.sensor, wled.light, google_translate.tts, system_health, unifi.update, xbox.sensor, frigate.camera, recorder, template.switch, cast.media_player, logbook, logger, influxdb, systemmonitor, mqtt, hacs, deconz.scene, cast, plex, deconz.lock, file_upload, esphome, deconz.select, harmony.remote, sensibo.select, auth, deconz.cover, group.media_player, sql, application_credentials, siren, wled.sensor, squeezebox.media_player, deconz.binary_sensor, mobile_app.de [... cut ...] input_text, sensibo.climate, bluetooth, tautulli.sensor, hassio.update, cover, stt, frigate.binary_sensor, sensibo.update, template.sensor, sensibo.switch, wled.binary_sensor, climate, season.sensor, frigate.number, harmony.switch, plex.update, timer, history The system is waiting for tasks: {<Task pending name='Task-2255' coro=<async_discover() running at /usr/local/lib/python3.12/site-packages/pysqueezebox/discovery.py:98> wait_for=<Future pending cb=[Task.task_wakeup()]> cb=[set.remove()]>}
`

commit a9e55871 (2024-05-05): Fix #104
PR #105 (2024-05-05): Turbokongen/issue104
Fixes #104
additional Thread checks are beeing made in core now.
commit d399131f (2024-05-05): Turbokongen/issue104 (#105)

* Fix #104

* Remember all instances
closed 2024-05-05 by turbokongen