turbokongen/hass-AMS #106: IndexError crash in byte_decode

Dead2, 2024-07-04:
Lately (past month or two), I have a problem where the ams readings stop updating very often. Sometimes it can last 3 days, other times 2 hours after a homeassistant restart.

The exception it throws is as follows:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/config/custom_components/ams/__init__.py", line 307, in connect
    self.sensor_data, _ = parser.parse_data(
                          ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ams/parsers/aidon.py", line 305, in parse_data
    (byte_decode(fields=pkt[v_start:v_stop],
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ams/parsers/__init__.py", line 20, in byte_decode
    data = fields[0] << 8 | fields[1]
           ~~~~~~^^^
IndexError: list index out of range
```

I use "oss-brikken" to connect to the meter, an Aidon 6525.

Dead2, 2024-07-04:
Btw, this might actually coincide with times when the OSS-brikken loses contact with the mobile network over a longer period of time (Coverage is really bad in our basement, and statistics have been missing a lot lately in the oss-app).

Unsure whether it is related or not, just an observation that possibly it sends some extra data to alert the user that it has lost uplink and the buffer is full, and the Aidon parser is not expecting that.

turbokongen, 2026-10-18:
Please test with https://github.com/turbokongen/hass-AMS/releases/tag/v2.07-beta3 to see if it solves the problem.