turbokongen/hass-AMS #107: Issue with Kamstrup 6841121 Meter Not Updating Properly in

Diddi-Dog, 2025-01-30:
I am using the hass-AMS integration with a Kamstrup 6841121BN243101040 meter, which reports OBIS codes as Kamstrup_V0001. However, I got  the some issues:

Power import values update only on large some changes

kamstrup_active_power_import only updates if there is a major difference (e.g., from 2 kW to 3 kW).
but it does not refresh to other values unless I manually reload the integration.

and these values are unknown

kamstrup_active_power_export_total
kamstrup_active_power_import_total
kamstrup_reactive_power_export_total
kamstrup_reactive_power_import_total
These always show as unknown, even after reloading the integration.
pls help and thx

turbokongen, 2025-02-01:
Please enable debug, and attach the logfile to this issue. 
I have the same meter model (6841121) And do not have those problems.

My first thought is that the sensor names you provide, does not match the default ones from the integration, and is actually the energy sensors. Those sensors are not populated until the first reading that occours once an hour. Until then there is no way of knowing the value of the energy used, simply because the data is not sent from the meter yet. 
So if you delete, and reinstall the integration, all the energy sensors will be unknown untill a value is received from the meter.  That occours only once an hour. Later, the sensors will be populated at a restart with the last known value.
The is no other updates to the sensor other than every hour (If the packet passes CRC check).

Diddi-Dog, 2025-02-01:
here is the logfile 
homeassistant.helpers.network.NoURLAvailableError
2025-02-01 11:06:42.731 DEBUG (MainThread) [custom_components.amshan.sensor] Decoded meter message: {'meter_manufacturer': 'Kamstrup', 'list_ver_id': 'Kamstrup_V0001', 'meter_id': '5706567277589390', 'meter_type': '6841121BN243101040', 'active_power_import': 4050, 'active_power_export': 0, 'reactive_power_import': 0, 'reactive_power_export': 141, 'current_l1': 15.77, 'current_l2': 5.92, 'current_l3': 12.57, 'voltage_l1': 233, 'voltage_l2': 230, 'voltage_l3': 233, 'meter_datetime': datetime.datetime(2025, 2, 1, 11, 6, 40)}
2025-02-01 11:06:42.731 DEBUG (MainThread) [custom_components.amshan.sensor] Received meter measures: {'meter_manufacturer': 'Kamstrup', 'list_ver_id': 'Kamstrup_V0001', 'meter_id': '5706567277589390', 'meter_type': '6841121BN243101040', 'active_power_import': 4050, 'active_power_export': 0, 'reactive_power_import': 0, 'reactive_power_export': 141, 'current_l1': 15.77, 'current_l2': 5.92, 'current_l3': 12.57, 'voltage_l1': 233, 'voltage_l2': 230, 'voltage_l3': 233, 'meter_datetime': datetime.datetime(2025, 2, 1, 11, 6, 40)}
2025-02-01 11:06:42.731 DEBUG (MainThread) [custom_components.amshan.sensor] Ignore unhandled measure_id meter_datetime
2025-02-01 11:06:42.731 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-list_ver_id with state Kamstrup_V0001
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-current_l1 with state 15.77
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-voltage_l2 with state 230
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-voltage_l1 with state 233
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-meter_type with state 6841121BN243101040
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-meter_id with state 5706567277589390
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-meter_manufacturer with state Kamstrup
2025-02-01 11:06:42.732 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-active_power_import with state 4050
2025-02-01 11:06:42.733 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-reactive_power_export with state 141
2025-02-01 11:06:42.733 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-active_power_export with state 0
2025-02-01 11:06:42.733 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-current_l3 with state 12.57
2025-02-01 11:06:42.733 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-voltage_l3 with state 233
2025-02-01 11:06:42.733 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-reactive_power_import with state 0
2025-02-01 11:06:42.734 DEBUG (MainThread) [custom_components.amshan.sensor] Update sensor Kamstrup-5706567277589390-current_l2 with state 5.92
2025-02-01 11:06:49.173 DEBUG (MainThread) [custom_components.amshan] homeassistant_stop received. Close down integration.
2025-02-01 11:06:49.218 DEBUG (MainThread) [custom_components.amshan.sensor] Received stop signal. Exit processing.
2025-02-01 11:06:50.313 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry DD:8B:CA:1F:7F:4A for melnor
Traceback (most recent call last):

turbokongen, 2025-02-01:
Ok. yes, exactly on point.
I beleive you have renamed your sensors. 
All sensors except energy sensors are updating regulary.
Energy sensors only update once an hour, and will be unknown untill first output from the meter. Be patient and let the meter send the data.
Your sensor names:
kamstrup_active_power_export_total is actually ams_active_energy_export_xxxxxxxxxx -> updated once every hour
kamstrup_active_power_import_total is actually ams_active_energy_import_xxxxxxxxxx -> updated once every hour
kamstrup_reactive_power_export_total is actually ams_reactive_energy_export_xxxxxxxxx -> updated once every hour
kamstrup_reactive_power_import_total is actually ams_reactive_energy_import_xxxxxxxxx -> updated once every hour
The meter transmits data for all other sensors ever 10seconds.

turbokongen, 2025-02-28:
Closing without feedback.

closed 2025-02-28 by turbokongen