turbokongen/hass-AMS #108: ams_active_energy_import showing values with factor 10

NilsFA, 2025-03-11:
My Kamstrup 6841138A suddenly jumped from 23 060 kWh -> 230 600 kWh, not sure why. Did anyone else notice this or is it just me? Not sure where to start looking for the source of this. Any tips?

[image]

turbokongen, 2025-03-24:
That look really wierd. No changes have bee made to the code regarding scaling.
Can you check all the energy sensor to see if they are the same, and also the Current sensors, because they all use the same scaling: 
https://github.com/turbokongen/hass-AMS/blob/d399131f19c57f1a6cd53f81604f1a556b9cc83d/custom_components/ams/parsers/kamstrup.py#L205
Enable debugging if you find nothing, and I'll have a look at the raw data.

NilsFA, 2025-04-15:
I think power, voltage etc looks okay, but the reactive energy seems too large as well. 

I think i got the large message in the log attached, I don't know what to look for though, perhaps you can find something :)

```
2025-04-15 22:00:56.837 DEBUG (Thread-2 (connect)) [custom_components.ams] Package complete
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams] data read from port=[126, 161, 44, 43, 33, 19, 252, 4, 230, 231, 0, 15, 0, 0, 0, 0, 12, 7, 233, 4, 15, 2, 21, 0, 55, 255, 128, 0, 0, 2, 35, 10, 14, 75, 97, 109, 115, 116, 114, 117, 112, 95, 86, 48, 48, 48, 49, 9, 6, 1, 1, 0, 0, 5, 255, 10, 16, 53, 55, 48, 54, 53, 54, 55, 51, 48, 55, 50, 49, 52, 49, 53, 53, 9, 6, 1, 1, 96, 1, 1, 255, 10, 18, 54, 56, 52, 49, 49, 51, 56, 65, 78, 50, 52, 53, 49, 48, 49, 48, 57, 48, 9, 6, 1, 1, 1, 7, 0, 255, 6, 0, 0, 2, 6, 9, 6, 1, 1, 2, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 1, 3, 7, 0, 255, 6, 0, 0, 0, 0, 9, 6, 1, 1, 4, 7, 0, 255, 6, 0, 0, 1, 11, 9, 6, 1, 1, 31, 7, 0, 255, 6, 0, 0, 0, 10, 9, 6, 1, 1, 51, 7, 0, 255, 6, 0, 0, 0, 201, 9, 6, 1, 1, 71, 7, 0, 255, 6, 0, 0, 0, 50, 9, 6, 1, 1, 32, 7, 0, 255, 18, 0, 236, 9, 6, 1, 1, 52, 7, 0, 255, 18, 0, 237, 9, 6, 1, 1, 72, 7, 0, 255, 18, 0, 236, 9, 6, 0, 1, 1, 0, 0, 255, 9, 12, 7, 233, 4, 15, 2, 21, 0, 55, 255, 128, 0, 0, 9, 6, 1, 1, 1, 8, 0, 255, 6, 1, 126, 8, 133, 9, 6, 1, 1, 2, 8, 0, 255, 6, 0, 0, 0, 249, 9, 6, 1, 1, 3, 8, 0, 255, 6, 0, 35, 166, 194, 9, 6, 1, 1, 4, 8, 0, 255, 6, 0, 151, 72, 82, 125, 191, 126]
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] list_type is 35
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] meter_serial, OBIS:[1, 1, 0, 0, 5, 255], Index:(49, 55), Type:10 Double OBIS
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type 10: 5706567307214155, Index:(57, 73)
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] meter_type, OBIS:[1, 1, 96, 1, 1, 255], Index:(75, 81), Type:10 Double OBIS
2025-04-15 22:00:56.838 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type 10: Omnipower 3 Phase Direct meter, Index:(83, 101)
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] meter_date_time, OBIS:[0, 1, 1, 0, 0, 255], Index:(227, 233), Type:9 Double OBIS
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] meter_date_time, [0, 1, 1, 0, 0, 255], 2025, 4, 15, Tuesday, 21, 00, 55, 2025-4-15-21-00-00-55
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_active_power_import, OBIS:[1, 1, 1, 7, 0, 255], Index:(103, 109), Type:6 Double OBIS
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 518, Index:(110, 114)
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_active_power_export, OBIS:[1, 1, 2, 7, 0, 255], Index:(116, 122), Type:6 Double OBIS
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 0, Index:(123, 127)
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_reactive_power_import, OBIS:[1, 1, 3, 7, 0, 255], Index:(129, 135), Type:6 Double OBIS
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 0, Index:(136, 140)
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_reactive_power_export, OBIS:[1, 1, 4, 7, 0, 255], Index:(142, 148), Type:6 Double OBIS
2025-04-15 22:00:56.839 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 267, Index:(149, 153)
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_current_l1, OBIS:[1, 1, 31, 7, 0, 255], Index:(155, 161), Type:6 Double OBIS
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 0.1, Index:(162, 166)
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_current_l2, OBIS:[1, 1, 51, 7, 0, 255], Index:(168, 174), Type:6 Double OBIS
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 2.01, Index:(175, 179)
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_current_l3, OBIS:[1, 1, 71, 7, 0, 255], Index:(181, 187), Type:6 Double OBIS
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  6: 0.5, Index:(188, 192)
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_voltage_l1, OBIS:[1, 1, 32, 7, 0, 255], Index:(194, 200), Type:18 Double OBIS
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  16/18: 236, Index:(201, 203)
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] ams_voltage_l2, OBIS:[1, 1, 52, 7, 0, 255], Index:(205, 211), Type:18 Double OBIS
2025-04-15 22:00:56.840 DEBUG (Thread-2 (connect)) [custom_components.ams.parsers.kamstrup] Value double OBIS type  16/18: 237, Index:(212, 214)
[... cut ...]
2025-04-15 22:01:01.529 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_voltage_l3
2025-04-15 22:01:01.529 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_energy_import
2025-04-15 22:01:01.529 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_active_energy_export
2025-04-15 22:01:01.529 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_import
2025-04-15 22:01:01.530 DEBUG (MainThread) [custom_components.ams.sensor] Updating sensor ams_reactive_energy_export
```

turbokongen, 2025-04-28:
From the logs, everything checks out. The correct value from the packet is 250369,33Kwh. Could you read the value of the display of the meter, for comparison?

NilsFA, 2025-05-03:
[image]

turbokongen, 2025-05-03:
[image]

So I did some analyzing of the data, without using my code, and still, the data in the raw datapacket aligns with the values you get, which no doubt is off by a factor of 10. So can anybody else watching this repository and have the same meter, check their values. I myself have 6841121B and do not have this issue.

NilsFA, 2025-06-12:
I'll try to get some info from the power company as well, perhaps they can change those settings OTA or something. Can't really see what should have changed on our side.